dotnet-bluetooth-le/dotnet-bluetooth-le #761: The platform IDevice.Name is not the same

yureng, 2023-10-24:
My Bluetooth device,  its 
CompleteLocalName: WaveCup,  
ShortenedLocalName: Cup.

## Steps to reproduce

1. Call the CrossBluetoothLE.Current.Adapter.StartScanningForDevicesAsync(..)  to search for devices

2. On Android,  IDevice.Name is: WaveCup, 

3. On IOS,  IDevice.Name is: Cup

## Expected behavior
The IDevice.Name obtained by both platforms should be the same,  both:  WaveCup

## Actual behavior
No！ 
Android is CompleteLocalName
iOS is ShortenedLocalName 

### Crashlog

none

## Configuration

**Version of the Plugin:**   v3.0.0

**Platform:**  Android 8 / Android 13 / iOS 14.7.1 / iOS 15.4

**Device:**  Samsung S8 / Samsung S22 / iPhone 8 / iPhone 8 Plus

on iOS, how can I get CompleteLocalName ? 
Thank you!

yureng, 2023-10-25:
Debug trace, I found:

CoreBluetooth.CBPeripheral.Name  is  "CompleteLocalName"
CoreBluetooth.CBDiscoveredPeripheralEventArgs.AdvertisementData[CBAdvertisement.DataLocalNameKey]  is  "ShortenedLocalName"

Hence the line of code， Apple.Adapter.cs  35 

var name = e.Peripheral.Name;
if (e.AdvertisementData.ContainsKey(CBAdvertisement.DataLocalNameKey))
{
    // iOS caches the peripheral name, so it can become stale (if changing)
    // keep track of the local name key manually
    name = ((NSString)e.AdvertisementData.ValueForKey(CBAdvertisement.DataLocalNameKey)).ToString();
}

"CompleteLocalName" is replaced with "ShortenedLocalName"

maybe， Apple.Adapter.cs  345 ：

records.Add(new AdvertisementRecord(AdvertisementRecordType.CompleteLocalName,
    NSData.FromString(advertisementData.ObjectForKey(key) as NSString).ToArray()));

Should be   "ShortenedLocalName"

janusw, 2023-11-18:
This might possibly be related to #703. There it was claimed that `Device.NativeDevice` has the full name. Could you check that?

closed 2024-01-18 by yureng