dotnet-bluetooth-le/dotnet-bluetooth-le #762: Can't find devices.

mthnzbk, 2023-10-24:
I don't get any error in the code below, but the StartScanningForDevicesAsync() method does not find open bluetooth devices. So Adapter.DeviceDiscovered += async (s, a) => {} I don't get any output in this part. Nor error.

## Configuration

**Version of the Plugin:** 3.0.0 

**Platform:**Android 10-11 

**Device:** Xiaomi Note 7
`using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using System.Collections.ObjectModel;
using System.Diagnostics;

namespace Kumanda
{
    public partial class MainPage : ContentPage
    {
        readonly IBluetoothLE Ble = CrossBluetoothLE.Current;
        readonly IAdapter Adapter = CrossBluetoothLE.Current.Adapter;

        public ObservableCollection<String> Devices;
        public ObservableCollection<IDevice> Device;

        public MainPage()
        {
            InitializeComponent();

            Ble.StateChanged += (s, e) =>
            {
                Debug.WriteLine("State: " + e.NewState);
            };

            Adapter.DeviceDiscovered += async (s, a) =>
            {
                Debug.WriteLine("Device: " + a.Device.ToString());
                try
                {
                    Devices.Add(a.Device.ToString());
                    Device.Add(a.Device);
                    await Adapter.ConnectToDeviceAsync(a.Device);
                    var service = await a.Device.GetServicesAsync();
                    Debug.WriteLine(service.ToString());
                    Debug.WriteLine(Devices + " " + Device);
                }
                catch (Exception ex)
                {
                    Debug.WriteLine(ex);
                }
                
            };

            Adapter.DeviceBondStateChanged += (s, e) =>
            {
                Debug.WriteLine(e.State + " " + e.Address);
            };

            Adapter.DeviceConnected += (s, e) =>
            {
                Debug.WriteLine(e.Device.ToString());
            };
        }

        protected override void OnAppearing()
        {
            base.OnAppearing();
            
        }

        private async void OnCounterClicked(object sender, EventArgs e)
        {
            Debug.WriteLine(Ble.State);
            try
            {
                await Adapter.StartScanningForDevicesAsync();
            }
            catch (Exception ex)
            {
                Debug.WriteLine(ex);
            }
        }
    }
}`

janusw, 2023-10-25:
> I don't get any error in the code below, but the StartScanningForDevicesAsync() method does not find open bluetooth devices. So Adapter.DeviceDiscovered += async (s, a) => {} I don't get any output in this part. Nor error.

Possibly you're failing to request permissions properly? See https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le#permissions

You need to add the relevant permissions to your manifest and also request them at runtime.

Which SDK level are you targeting? Do you use Xamarin or MAUI?

mthnzbk, 2023-10-26:
`<uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
<uses-permission android:name="android.permission.INTERNET" />
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
<uses-permission android:name="android.permission.READ_USER_DICTIONARY" />
<uses-permission android:name="android.permission.VIBRATE" />
<uses-permission android:name="android.permission.WRITE_USER_DICTIONARY" />
<uses-sdk android:minSdkVersion="26" android:targetSdkVersion="33" />`

`protected override void OnCreate(Bundle savedInstanceState)
{
    base.OnCreate(savedInstanceState);
    if (ContextCompat.CheckSelfPermission(this, Manifest.Permission.BluetoothScan) != Permission.Granted)
    {
        ActivityCompat.RequestPermissions(this, new[] {Manifest.Permission.BluetoothScan }, 0);
    }

    if (ContextCompat.CheckSelfPermission(this, Manifest.Permission.BluetoothAdvertise) != Permission.Granted)
    {
        ActivityCompat.RequestPermissions(this, new[] { Manifest.Permission.BluetoothAdvertise }, 0);
    }
    if (ContextCompat.CheckSelfPermission(this, Manifest.Permission.BluetoothConnect) != Permission.Granted)
    {
        ActivityCompat.RequestPermissions(this, new[] { Manifest.Permission.BluetoothConnect }, 0);
    }
    if (ContextCompat.CheckSelfPermission(this, Manifest.Permission.BluetoothAdmin) != Permission.Granted)
    {
        ActivityCompat.RequestPermissions(this, new[] { Manifest.Permission.BluetoothAdmin }, 0);
    }
}`

.net 7 Maui

janusw, 2023-10-27:
From https://developer.android.com/develop/connectivity/bluetooth/bt-permissions:

> If your app uses Bluetooth scan results to derive physical location, declare the [ACCESS_FINE_LOCATION](https://developer.android.com/reference/android/Manifest.permission#ACCESS_FINE_LOCATION) permission. Otherwise, you can [strongly assert that your app doesn't derive physical location](https://developer.android.com/develop/connectivity/bluetooth/bt-permissions#assert-never-for-location).