dotnet-bluetooth-le/dotnet-bluetooth-le #771: Plugin.BLE causes ambiguous invocation error with Community.ToolkitWinUI

NickSchurgers, 2023-11-06:
Since upgrading CommunityToolkit from v7 to v8, we have an ambiguous invocation error on the CommunityToolkit.WinUI namespace. After removing all our dependencies and reinstalling them one by one, we determined that the conflict is with the Plugin.BLE library. It appears it references the old v7 version, which causes the conflict with the CommunityToolkit.WinUI.Extensions  v8 package. 
We created a sample to illustrate: [Sample.zip](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/files/13267704/Sample.zip)

## Steps to reproduce

1. Create a new WinUI 3 project

2. Add CommunityToolkit.WinUI.Extension 8.0.230907 as reference

3. Add Plugin.BLE 3.0.0

4. Use one of the extensions methods, for example:
```csharp
var newWindow = new Window();
await newWindow.DispatcherQueue.EnqueueAsync(() => { });
```

## Expected behavior
No compilation errors.

## Actual behavior
The compiler throws the follow message: 
```
error CS0121: The call is ambiguous between the following methods or properties: 'CommunityToolkit.WinUI.DispatcherQueueExtensions.EnqueueAsync(Microsoft.UI.Dispatching.DispatcherQueue, 
 System.Action, Microsoft.UI.Dispatching.DispatcherQueuePriority)' and 'CommunityToolkit.WinUI.DispatcherQueueExtensions.EnqueueAsync(Microsoft.UI.Dispatching.DispatcherQueue, System.Action, Microsoft.UI.Dispatching.DispatcherQ 
ueuePriority)' 
```

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** Win11/WinUI 3

janusw, 2023-11-06:
As a matter of fact, the dependency on CommunityToolkit has recently been removed on the master branch. I assume this should fix your problem, which you can check by using a CI-generated nupkg of the latest commit on master: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/suites/17915662869/artifacts/1029254759

NickSchurgers, 2023-11-07:
I just tried the CI-generated package and it indeed fixes the conflict we have. 
We will wait for the next release. Any indication when that could be?

janusw, 2023-11-07:
> I just tried the CI-generated package and it indeed fixes the conflict we have.

Great to hear that! 😃 

> We will wait for the next release. Any indication when that could be?

I'll probably publish a 3.1.0-beta within a week. For the final 3.1.0 I want to include support for the .NET 8 target frameworks, therefore I'd like to wait until .NET 8 is released (which should happen soon). I hope that release 3.1.0 can be finalized by the end of November.