dotnet-bluetooth-le/dotnet-bluetooth-le #772: Advertise data parsing

janmarektelc, 2023-11-08:
If the BLE device broadcasts Broadcast_Name(0x30) in the Advertising data, this information is not correctly parsed into the AdvertisementRecord object.
There si an easy fix - add Broadcast_Name flag to AdvertisementRecordType. It's shared type used by all other platforms.

## Steps to reproduce

1. Scan BLE device which advertise Broadcast_Name in data

2. Check if data is correctly parsed

## Expected behavior
Correct behavior is to provide all data to List of AdvertisementRecord on ParseScanRecord method output.

## Actual behavior
If device broadcast this Broadcast_Name in advertisement data, parsing fails and data isn't complete.

### Crashlog
There is no exception, just Trace message "Advertisment record type not defined"

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** Android 13

**Device:** Samsung S21

janmarektelc, 2023-11-08:
Hi,

I have a working solution for this issue. It's just small change in AdvertisementRecordType - just add this 

        /// <summary>
        /// «Broadcast Name»   Public Broadcast Profile v1.0 or later
        /// </summary>
        BroadcastName = 0x30,

Unfortunately, I don't have permission to create a branch and pull request.

janusw, 2023-11-08:
> I have a working solution for this issue.

🥳 

> Unfortunately, I don't have permission to create a branch and pull request.

You can simply fork this repo, create a branch in your fork, and open a PR from there.

janusw, 2023-11-09:
Fixed via #773.

closed 2023-11-09 by janusw