dotnet-bluetooth-le/dotnet-bluetooth-le #775: Support to calculate advertising interval

midoragh, 2023-11-09:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Get advertising info from DeviceAdvertised event

2.

3.

## Expected behavior
Device info in DeviceAdvertised event should contain property like DateTime AdvertisementReceived
this would allow a more precise calculation of the advertisement interval (instead of using the time when the event reaches the app)

## Actual behavior
AdvertisementReceived is not available

### Crashlog

No crash

## Configuration

**Version of the Plugin:** 3.0

**Platform:** Android 13

**Device:** Nokia C32

janusw, 2023-11-09:
> Device info in DeviceAdvertised event should contain property like DateTime AdvertisementReceived

One can certainly add such a property ...

> this would allow a more precise calculation of the advertisement interval (instead of using the time when the event reaches the app)

... but I doubt it will make much of a difference. I would expect that the time when the event reaches the app is reasonably close to the time when the advertisement is received by the library. (Of course one can just try it, in order to see if there is a significant difference.)

axa88, 2024-05-08:
looking at Nordic device implementation, imho the gold standard
minimum possible advertising interval is 20ms.  recommended fast interval 30ms
https://devzone.nordicsemi.com/cfs-file/__key/communityserver-discussions-components-files/4/6708.Core_5F00_v4.1.pdf

Typical employ slower interval of 62.5ms + small random value, to not overwhelm devices.
https://stackoverflow.com/q/76337421/5122089

Its all but inconceivable that a callback contained in the same executable will contribute any significant overhead to this calculation.