dotnet-bluetooth-le/dotnet-bluetooth-le #780: characteristic.ValueUpdated never triggeed

manudev-1, 2023-11-19:
## Steps to reproduce

1. [image]
2. [image]
3. [image]
4. [image]
5. [image]
6.  Arduino's Code:

- 6.1
```CPP
#include <ArduinoBLE.h>
#include <Arduino.h>

const char * deviceServiceUuid = "ffe0ecd2-3d16-4f8d-90de-e89e7fc396a5";
const char * deviceServiceReqUuid = "d8de624e-140f-4a22-8594-e2216b84a5f2";
const char * deviceServiceResUuid = "DF2D5AFD-1CDE-4CAC-92F1-0452992D7692";

const int eventListener = 2;

BLEService appService(deviceServiceUuid);
BLECharacteristic appReqChar(deviceServiceReqUuid, BLEWrite | BLERead, 4);
BLECharacteristic appResChar(deviceServiceResUuid, BLERead | BLENotify, 4);

String inputString = "CALL";
byte byteArray[4];

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

  BLE.setDeviceName("Arduino UNO R4");
  BLE.setLocalName("Arduino UNO R4");

  if(!BLE.begin()){
    Serial.println("Starting BLE Module FAILED");
    while(1);
  }

  BLE.setAdvertisedService(appService);
  appService.addCharacteristic(appReqChar);
  appService.addCharacteristic(appResChar);
  BLE.addService(appService);

  BLE.advertise();
  
  pinMode(eventListener, INPUT);

  inputString.getBytes(byteArray, 4);

}

const uint8_t* currentValue = appReqChar.value();
uint8_t newValue;

void loop() {
  BLEDevice central = BLE.central();
  if(central){
    Serial.println("CONNECTED");
    if(digitalRead(eventListener) == HIGH){
      appResChar.writeValue(byteArray, 4);
    }
    else{
      appResChar.writeValue((int8_t)0);
    }
  }
  delay(1000);
}
```

- 6.2 [image]

## Expected behavior
I would like my application to scan the bluetooth devices next to it that show them on a list and that when the user presses on one of the devices in the list it will start a connection. I’m trying to use my arduino uno r4, and it when the app is connected changes the value of the feature shown in the code, and I would like inside the event to show me the message sent by the arduino.

## Actual behavior
Everything starts correctly but the value change listening event never runs.

### Crashlog
No exception

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** Android 13.0 - API 33

**Device:** Xiaomi 2201117TY

marcusebm, 2024-02-13:
Hi, any news to this topic? I have the same problem with .net maui 8. And different Android devices.

The same code is running on windows.

Not tested on ios.

janusw, 2024-02-13:
I think there are some API changes in API level 33 which might cause this. Can someone confirm that the problem only occurs with API 33 and above (Android 13/14), but works with API 32 and below (Android 12 and earlier)?

marcusebm, 2024-02-15:
It does not work with Android 6.0 (API23) on.

janusw, 2024-02-15:
> It does not work with Android 6.0 (API23).

That's surprising. Maybe you're doing something wrong? (Android 6 is pretty ancient, anyway.)

> Can I support in any other way?

Do you see characteristics updates working with *any* Plugin.BLE version (on *any* Android version)?

marcusebm, 2024-02-15:
Could be that I do something wrong. I use the following lines of code which should work.

`    this.receiveCharacteristic =
        await service.GetCharacteristicAsync(GattIdentifiers.UartGattCharacteristicReceiveId);
    this.receiveCharacteristic.ValueUpdated    += (o, args) =>
    {
        var receivedBytes = args.Characteristic.Value;
    };
`

The same code is working if I run the App on windows.

I tried all Plugin.BLE versions since version 3.x (including prereleases)

marcusebm, 2024-02-16:
Hi all,

it seems that my "ReadCharacter_ValueUpdated" is called (Android 12 API 31).

But only 2-3 times at the beginning. After the connected give the information that no att channel is available. and no buffer for sending the nottification.

Windows is still running.

marcusebm, 2024-02-19:
Hi all,

I could test it today and it is also  running on Android 6.

But values seems not to be valid and complete. Also receiving stop working after a while.

Are there any limitations on Android devices or rules that must be followed?