dotnet-bluetooth-le/dotnet-bluetooth-le #785: [iOS] Characteristic.WriteAsync() hangs

softeip, 2023-12-05:
I have a MAUI app which needs to monitor realtime data from BLE device.
Usually I send ~20 bytes to the device and expect to receive ~300-500 bytes back. Each packet gets sliced by 155 bytes per notification, so smartphone sends one notification in this case and receives multiple. What I noticed is that the app stops sending and receiving notifications after some random time (usually after 30-45 mins, but sometimes it hangs after 5-10 mins). 
I've tried to add more logging and spent hours on debugging and now I see that Characteristic.WriteAsync() call hangs for unknown reasons. I tried to pass cancellation token to its params and cancel it after ~200 ms - no reaction. Also I ensured that I call the method on the main thread.

About characteristic - WriteType is set to WithoutResponse. Trying to write data WithResponse throws an error saying that it is not supported by device or sth like that.

Sometimes I also gets an exception saying that my device disconnected but I don't know if it is related to my issue:
Device *** disconnected while writing characteristic with ***
**at** Plugin.BLE.Abstractions.Utils.TaskBuilder.<SafeEnqueueAndExecute>d__11`1[[System.Int32, System.Private.CoreLib, Version=7.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
**at** Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__9`3[[System.Int32, System.Private.CoreLib, Version=7.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler, System.Private.CoreLib, Version=7.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler`1[[CoreBluetooth.CBPeripheralErrorEventArgs, Microsoft.iOS, Version=16.4.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065]], System.Private.CoreLib, Version=7.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
**at** Plugin.BLE.Abstractions.CharacteristicBase`1.<WriteAsync>d__34[[CoreBluetooth.CBCharacteristic, Microsoft.iOS, Version=16.4.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065]].MoveNext()

## Configuration

**Version of the Plugin:** 3.0.0 but also used all -pre versions and had the same issue.
**Platform & devices:** 
Only iOS affected. Android app works without issues.
iPhone XR / iOS 16.6
iPhone 14 Pro / iOS 17.1.2
iPad Air 3 (Wi-Fi) / iPadOS 17.1.1

I would love to share an example or a project to reproduce the issue, but I'm not sure how to create one without exposing implementation details due to NDA.

What can be the reason of such behaviour on iOS?
What could I try to do to get more information on what's going on and how to fix it?

softeip, 2023-12-11:
Since passing CancellationToken to WriteAsync() does not return control to my code, I applied a temp hack:

```
var writeTask = _characteristic.WriteAsync(data);
var waitTask = Task.Delay(1000);
await Task.WhenAny(writeTask, waitTask);

if (writeTask.Status != TaskStatus.RanToCompletion)
{
  throw new MyException("Write timeouted");
}
```

Moreover, when WriteAsync() hangs, adapter events like DeviceConnectionError, DeviceConnectionLost, DeviceDisconnected seem to be not fired. Added more logging on this - will see how it goes for others

sschoeb, 2024-02-01:
I've had the same issue. Currently I've inlined the library and changed the following:
In the BleImplementation.cs for iOS i've changed on line 102 the DispatchQueue from current to DefaultGlobalQueue. 

`_centralManager = new CBCentralManager(cmDelegate, DispatchQueue.DefaultGlobalQueue, options);`

Not 100% sure what the downsides are of this change, but from her on I'm able to use the library on iOS.

softeip, 2024-02-16:
Thank you for the reply!
Tried it but it started to hang more often and more quickly🤔
The line you refer to is different in the latest version. Maybe it worked for one of the older versions