dotnet-bluetooth-le/dotnet-bluetooth-le #788: BLE.Client.WinConsole demo app hangs on GetServicesAsync call

melihercan, 2023-12-21:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Get main branch

2. Customize the program.cs so that it scans passively 3 times and then calls ShowNumberOfServices on [WGX-iBeacon](https://www.lucadentella.it/en/2018/03/19/wgx-ibeacons/) device:
```
using BLE.Client.WinConsole;
using Plugin.BLE;
using Plugin.BLE.Abstractions.Contracts;
using System;

Console.WriteLine("Hello, BLE World!");
using (var ct = new ConsoleTracer())
{
    ////Plugin.BLE.Abstractions.Trace.TraceImplementation = ct.GetPrefixedTrace("Plugin.BLE");
    var demo = new BleDemo(ct.GetPrefixedTrace("      DEMO"));

    for (int i=0; i<3; i++)
        await demo.DoTheScanning(ScanMode.Passive, 5000);
    await demo.ShowNumberOfServices("EB408B203E81");    // iBeacon
}
```
Also change the DoTheScanning debug messages a bit:
```
            ////Write("Bluetooth is on");
            Write("Scanning now for " + time_ms + " ms..." + " with mode:" + scanMode);            
```

## Expected behavior
1. It should scan and report the found BLE devices
2. Connect to WGX-iBeacon device
3. Get services and display them
4. Repeat 2 and 3

## Actual behavior
1. Scan is OK
2. Connection to the device takes a long time, ~30sec (please see the log below)!!!
3. Calls dev.GetServicesAsync and it hangs!!!

I have tried with different BLE devices and the issue is reproducible

### Crashlog
No crash just hanging. Here is the log:
```
Hello, BLE World!
16:31:40.687       DEMO - Scanning now for 5000 ms... with mode:Passive
16:31:41.226       DEMO - DeviceDiscovered: 5C4076784064 with Name = Bluetooth 5c:40:76:78:40:64
16:31:41.276       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
16:31:43.243       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
16:31:45.730       DEMO - Scanning now for 5000 ms... with mode:Passive
16:31:46.489       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
16:31:46.489       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
16:31:46.513       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
16:31:46.513       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
16:31:46.538       DEMO - DeviceDiscovered: 5C4076784064 with Name = Bluetooth 5c:40:76:78:40:64
16:31:46.538       DEMO - DeviceDiscovered: 5C4076784064 with Name = Bluetooth 5c:40:76:78:40:64
16:31:47.199       DEMO - DeviceDiscovered: E0FD80FCB5B7 with Name = Bluetooth e0:fd:80:fc:b5:b7
16:31:47.199       DEMO - DeviceDiscovered: E0FD80FCB5B7 with Name = Bluetooth e0:fd:80:fc:b5:b7
16:31:50.755       DEMO - Scanning now for 5000 ms... with mode:Passive
16:31:51.047       DEMO - DeviceDiscovered: 7F0B22C537DB with Name = Bluetooth 7f:0b:22:c5:37:db
16:31:51.047       DEMO - DeviceDiscovered: 7F0B22C537DB with Name = Bluetooth 7f:0b:22:c5:37:db
16:31:51.047       DEMO - DeviceDiscovered: 7F0B22C537DB with Name = Bluetooth 7f:0b:22:c5:37:db
16:31:51.521       DEMO - DeviceDiscovered: 5C4076784064 with Name = Bluetooth 5c:40:76:78:40:64
16:31:51.521       DEMO - DeviceDiscovered: 5C4076784064 with Name = Bluetooth 5c:40:76:78:40:64
16:31:51.521       DEMO - DeviceDiscovered: 5C4076784064 with Name = Bluetooth 5c:40:76:78:40:64
16:31:52.463       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
16:31:52.463       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
16:31:52.463       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
16:31:53.395       DEMO - DeviceDiscovered: EB408B203E81 with Name = WGX_iBeacon
16:31:53.395       DEMO - DeviceDiscovered: EB408B203E81 with Name = WGX_iBeacon
16:31:53.395       DEMO - DeviceDiscovered: EB408B203E81 with Name = WGX_iBeacon
16:31:54.588       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
16:31:54.588       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
16:31:54.588       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
16:31:55.787       DEMO - Connecting to device with address = EB408B203E81
16:32:33.591       DEMO - Connected to WGX_iBeacon EB408B203E81 Connected
16:32:33.591       DEMO - Calling dev.GetServicesAsync()...
```

## Configuration

**Version of the Plugin:** Repo main brach

**Platform:** Windows 11

**Device:** [WGX-iBeacon](https://www.lucadentella.it/en/2018/03/19/wgx-ibeacons/)

melihercan, 2023-12-21:
On the following run it works ok:

```
Hello, BLE World!
17:17:56.999       DEMO - Scanning now for 5000 ms... with mode:Passive
17:18:02.039       DEMO - Scanning now for 5000 ms... with mode:Passive
17:18:07.058       DEMO - Scanning now for 5000 ms... with mode:Passive
17:18:08.691       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:18:08.691       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:18:08.691       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:18:12.122       DEMO - Connecting to device with address = EB408B203E81
17:18:12.232       DEMO - Connected to WGX_iBeacon EB408B203E81 Connected
17:18:12.232       DEMO - Calling dev.GetServicesAsync()...
17:18:12.276       DEMO - Found 7 services
17:18:13.288       DEMO - Disconnecting from WGX_iBeacon EB408B203E81
17:18:14.300       DEMO - ReConnecting to device WGX_iBeacon EB408B203E81...
17:18:14.300       DEMO - Connect Done.
17:18:15.300       DEMO - Calling dev.GetServicesAsync()...
17:18:15.301       DEMO - Found 7 services
Console Tracer is Finished.
```

melihercan, 2023-12-21:
On the third run, it hung on the Disconnect call: It looks like there is a race condition somewhere on the native Windows path so that the native calls do not complete...
``
17:21:43.681       DEMO - Scanning now for 5000 ms... with mode:Passive
17:21:45.405       DEMO - DeviceDiscovered: EB408B203E81 with Name = WGX_iBeacon
17:21:45.874       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:21:48.577       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
17:21:48.726       DEMO - Scanning now for 5000 ms... with mode:Passive
17:21:48.897       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
17:21:48.897       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
17:21:51.025       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:21:51.025       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:21:52.210       DEMO - DeviceDiscovered: 4A344A6746ED with Name = Bluetooth 4a:34:4a:67:46:ed
17:21:52.210       DEMO - DeviceDiscovered: 4A344A6746ED with Name = Bluetooth 4a:34:4a:67:46:ed
17:21:53.757       DEMO - Scanning now for 5000 ms... with mode:Passive
17:21:54.275       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:21:54.275       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:21:54.275       DEMO - DeviceDiscovered: 43B1DA015144 with Name = Bluetooth 43:b1:da:01:51:44
17:21:55.703       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
17:21:55.703       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
17:21:55.703       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
17:21:56.771       DEMO - DeviceDiscovered: 4246CE57C1BC with Name = Bluetooth 42:46:ce:57:c1:bc
17:21:56.771       DEMO - DeviceDiscovered: 4246CE57C1BC with Name = Bluetooth 42:46:ce:57:c1:bc
17:21:56.771       DEMO - DeviceDiscovered: 4246CE57C1BC with Name = Bluetooth 42:46:ce:57:c1:bc
17:21:58.791       DEMO - Connecting to device with address = EB408B203E81
17:22:01.311       DEMO - Connected to WGX_iBeacon EB408B203E81 Connected
17:22:01.311       DEMO - Calling dev.GetServicesAsync()...
17:22:02.567       DEMO - Found 7 services
17:22:03.582       DEMO - Disconnecting from WGX_iBeacon EB408B203E81`
```

janusw, 2023-12-27:
> I have tried with different BLE devices and the issue is reproducible
> 
> [..]
> 
> **Platform:** Windows 11
> 
> **Device:** [WGX-iBeacon](https://www.lucadentella.it/en/2018/03/19/wgx-ibeacons/)

 So you are saying the same problem also occurs with other devices? Or is it specific to this WGX-iBeacon device?

janusw, 2023-12-27:
@AskBojesen You're the one who contributed the WinConsole demo app (and refactored the Windows BLE implementation). Can you actually reproduce this problem?

melihercan, 2023-12-27:
> > I have tried with different BLE devices and the issue is reproducible
> > [..]
> > **Platform:** Windows 11
> > **Device:** [WGX-iBeacon](https://www.lucadentella.it/en/2018/03/19/wgx-ibeacons/)
> 
> So you are saying the same problem also occurs with other devices? Or is it specific to this WGX-iBeacon device?

Yes, the same issue occurs on other devices. For example, I used [SimCPR](https://simcpr.com/en/#trainer)

GetServicesAsync hanging:
```
Hello, BLE World!
14:55:51.696       DEMO - Scanning now for 5000 ms... with mode:Passive
14:55:52.848       DEMO - DeviceDiscovered: E0189F0AEAC0 with Name = SimCPR sensor
14:55:54.058       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:55:56.736       DEMO - Scanning now for 5000 ms... with mode:Passive
14:55:57.258       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:55:57.258       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:56:00.682       DEMO - DeviceDiscovered: E0189F0AEAC0 with Name = SimCPR sensor
14:56:00.682       DEMO - DeviceDiscovered: E0189F0AEAC0 with Name = SimCPR sensor
14:56:01.757       DEMO - Scanning now for 5000 ms... with mode:Passive
14:56:02.807       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:56:02.807       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:56:02.807       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:56:03.584       DEMO - DeviceDiscovered: E0189F0AEAC0 with Name = SimCPR sensor
14:56:03.584       DEMO - DeviceDiscovered: E0189F0AEAC0 with Name = SimCPR sensor
14:56:03.584       DEMO - DeviceDiscovered: E0189F0AEAC0 with Name = SimCPR sensor
14:56:06.780       DEMO - Connecting to device with address = e0189f0a7401
14:56:08.225       DEMO - Connected to SimCPR sensor E0189F0A7401 Connected
14:56:08.225       DEMO - Calling dev.GetServicesAsync()...
```

Disconnect hanging:
=============
```
Hello, BLE World!
14:43:28.984       DEMO - Scanning now for 5000 ms... with mode:Passive
14:43:34.022       DEMO - Scanning now for 5000 ms... with mode:Passive
14:43:37.152       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:43:37.152       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:43:39.045       DEMO - Scanning now for 5000 ms... with mode:Passive
14:43:39.092       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:43:39.092       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:43:39.092       DEMO - DeviceDiscovered: E0189F0A7401 with Name = SimCPR sensor
14:43:44.083       DEMO - Connecting to device with address = e0189f0a7401
14:43:45.850       DEMO - Connected to SimCPR sensor E0189F0A7401 Connected
14:43:45.850       DEMO - Calling dev.GetServicesAsync()...
14:43:47.675       DEMO - Found 6 services
14:43:48.676       DEMO - Disconnecting from SimCPR sensor E0189F0A7401
14:45:43.556       DEMO - ReConnecting to device SimCPR sensor E0189F0A7401...
```

AskBojesen, 2024-02-05:
> @AskBojesen You're the one who contributed the WinConsole demo app (and refactored the Windows BLE implementation). Can you actually reproduce this problem?

I will try to reproduce and fix now

AskBojesen, 2024-02-05:
I cannot reproduce - @melihercan : Will you give PR #808 a test ?

melihercan, 2024-02-05:
@AskBojesen Sure, I will try it tomorrow and let you know.

melihercan, 2024-02-06:
@AskBojesen Unfortunately, I wasn't able to test it today due to an interruption from another project. I will check again tomorrow.

melihercan, 2024-02-07:
Hi @AskBojesen First I added an extra debug message to mark the end of ShowNumberOfServicesWithDisconnectConnect call.

```
        public async Task ShowNumberOfServicesWithDisconnectConnect()
        {
            string bleaddress = BleAddressSelector.GetBleAddress();
            Write("Connecting to device with address = {0}", bleaddress);
            IDevice dev = await Adapter.ConnectToKnownDeviceAsync(bleaddress.ToBleDeviceGuid()) ?? throw new Exception("null");
            string name = dev.Name;
            Write($"Connected to {name} {dev.Id.ToHexBleAddress()} {dev.State}");
            Write("Calling dev.GetServicesAsync()...");
            var services = await dev.GetServicesAsync();
            Write("Found {0} services", services.Count);
            await Task.Delay(1000);
            Write("Disconnecting from {0} {1}", name, dev.Id.ToHexBleAddress());
            await Adapter.DisconnectDeviceAsync(dev);
            await Task.Delay(1000);
            Write("ReConnecting to device {0} {1}...", name, dev.Id.ToHexBleAddress());
            await Adapter.ConnectToDeviceAsync(dev);
            Write("Connect Done.");
            await Task.Delay(1000);
            Write("Calling dev.GetServicesAsync()...");
            services = await dev.GetServicesAsync();
            Write("Found {0} services", services.Count);
            await Adapter.DisconnectDeviceAsync(dev);
            await Task.Delay(100);
            Write("------------ END ShowNumberOfServicesWithDisconnectConnect");
        }
```

then I ran the code 3 times:
Run1: Disconnect hangs
Run2: Connect takes 21 sec! and Disconnect hangs
Run3: Disconnect hangs
[Logs808.txt](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/files/14191620/Logs808.txt)

I attach the log file with 3 runs.

melihercan, 2024-02-07:
@AskBojesen FYI, I quickly replaced the  Window API layer with the following code (implemented only the APIs that I am calling - quick and dirty), and then each time it connects, gets services, and disconnects successfully.  No delays, no hanging.

```
namespace Plugin.BLE.Abstractions
{
    public class ScanFilterOptions
    {
    }
    public struct ConnectParameters
    {
    }

}

namespace Plugin.BLE.Abstractions.Contracts
{
    public interface IAdapter
    {
        int ScanTimeout { get; set; }
        ScanMode ScanMode { get; set; }

        event EventHandler<DeviceEventArgs> DeviceAdvertised;

        event EventHandler<DeviceEventArgs> DeviceDiscovered;
        event EventHandler<DeviceEventArgs> DeviceConnected;
        event EventHandler<DeviceEventArgs> DeviceDisconnected;
        event EventHandler<DeviceErrorEventArgs> DeviceConnectionLost;
        event EventHandler<DeviceErrorEventArgs> DeviceConnectionError;
        Task StartScanningForDevicesAsync(ScanFilterOptions scanFilterOptions = null, Func<IDevice, bool> deviceFilter = null, bool allowDuplicatesKey = false, CancellationToken cancellationToken = default);

        Task StopScanningForDevicesAsync();

        Task<IDevice> ConnectToKnownDeviceAsync(Guid deviceGuid, ConnectParameters connectParameters = default, CancellationToken cancellationToken = default);
        Task DisconnectDeviceAsync(IDevice device);

    }

    public interface IDevice
    {
        Guid Id { get; }
        string Name { get; }
        int Rssi { get; set; }

        Task<IReadOnlyList<IService>> GetServicesAsync(CancellationToken cancellationToken = default);
        Task<IService> GetServiceAsync(Guid id, CancellationToken cancellationToken = default);

    }

    public interface IService
    {
        Guid Id { get; }
        string Name { get; }

        Task<IReadOnlyList<ICharacteristic>> GetCharacteristicsAsync();
        Task<ICharacteristic> GetCharacteristicAsync(Guid id);

    }

    public interface IDescriptor
    {
    }

    public interface ICharacteristic
    {
        Guid Id { get; }
        string Name { get; }

        byte[] Value { get; }

        event EventHandler<CharacteristicUpdatedEventArgs> ValueUpdated;
[... 419 lines cut ...]
        public ICharacteristic Characteristic { get; set; }
        public CharacteristicUpdatedEventArgs(ICharacteristic characteristic)
        {
            Characteristic = characteristic;
        }
    }

}

namespace Plugin.BLE.Extensions
{

}

namespace Plugin.BLE.Abstractions.Extensions
{

}

namespace Plugin.BLE.Windows
{

}

```

AskBojesen, 2024-02-07:
@melihercan : But above is not related to the WinConsole client itself.
When you write that disconnect hangs - have you then been patient ? I have seen it takes 5 secs to disconnect in windows

melihercan, 2024-02-07:
> @melihercan : But above is not related to the WinConsole client itself. 
_**The issue is not coming from the WinConsole app, but the BLE library windows portion. The code I provided replaces the BLE library windows portion. Then it is working fine. I have seen the same hanging issue on MAUI app too.**_

When you write that disconnect hangs - have you then been patient ? I have seen it takes 5 secs to disconnect in windows
> 
_**Yes I did (minutes), but it is not completing.**_

> Do you have a better implementation ? then we need that in a new branch!
_**No I don't. I was not able to use the Windows portion of the BLE library due to the hanging issue. Hence I had the quick and dirty replacement and my app is running fine now.**_
> 
> In the WinConsole program I have also added some tests not using the plugin - to troubleshoot errors

AskBojesen, 2024-02-07:
My idea is to make "demos' in the WinConsole project to reproduce errors, and if I suspect that it is the implementation in the plugin which causes errors, then I make a Pure Windows demo to investigate in parallel.

I have just found another error, which I can reproduce every time... I will make a fix for that first.
The error is: Connect -> ConnecttionLost -> Connect

AskBojesen, 2024-02-12:
I have just reproduced with the latest'n'greatest from master including my last merged PR
We should rather rename this issue to: Disconnect hangs sometimes on Windows

AskBojesen, 2024-02-12:
@melihercan : Will you rename this issue to: "Disconnect hangs sometimes on Windows"
Just for sake of good order...

melihercan, 2024-02-12:
Renamed

MarkKananen, 2024-08-28:
Has there been any movement on this?   I am running a MAUI app on Windows using nuget Plugin.BLE version 3.1.0, and sometimes it hangs in DisconnectAsync.  Any work around anyone can  think of?

AskBojesen, 2024-08-29:
**Status:**
No movement

**Known Workarounds:**
1. Bluetooth on/off
2. Have device disconnect Bluetooth on/off
3. Wait 1 minute

**Recap**
Windows does NOT have a disconnect method  - and I tried all the other methods in the API - nothing disconnects.

AskBojesen, 2025-10-21:
Yes @sinatrocious - above solution would make the disconnect call return, but the device will in fact still be connected to the Windows. Plugin.BLE would report it as not connected - but in the underlaying OS it is in fact connected.

My scenario: Two computers=A, B  and one BLE device=X accepting one connection

1. A connects to X
2. A disconnects from X   <- This is the problem
3. B connects to X <- Cannot since A is still connected

The above solution will not solve my problem, but it might be better for other use cases to have the call at least return or throw an exception.

My problem is also within the same computer and same application
1. App discovers X
2. App connects to X 
3. App disconnects from X   <- This is the problem
4. App closes - process gone.
5. App starts again
6. App cannot discover X - since X is already/still connected to Windows!

My only solution has been to turn off bluetooth whenever I exit the application
I will look into this again - I will give it one more shot

BTW: In Android and iOS using same device and same libraries - I have never seen this problem