dotnet-bluetooth-le/dotnet-bluetooth-le #794: ScanTimeoutElapsed method not getting called

PrachiKarapurkar, 2024-01-03:
Once scan timeouts ScanTimeoutElapsed event not getting triggered.
Any solution to resolve this issue.
 
 ```
adapter.ScanTimeoutElapsed += ScanTimeoutElapsed;
 
 adapter.ScanTimeout = 20000;
 adapter.DeviceDiscovered += (s, a) =>
 {
       await adapter.StartScanningForDevicesAsync();
  }
```
        
  
  
  Thanks in advance

janusw, 2024-03-16:
I cannot confirm this on current master (close to v3.1.0-beta.2) ...

When running the MAUI sample client (BLE.Client.Maui), one can see in the logs that the `ScanTimeoutElapsed` event is being called correctly.

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b179852efef3402a74c3c513dfce004c9c158077/Source/BLE.Client/BLE.Client.Maui/ViewModels/BLEScannerViewModel.cs#L191-L198

Does the scanning itself work well for you? Are you seeing any advertisements? Can you try the sample client?

garrypeterson, 2025-04-15:
This order of calls is incorrect:

> adapter.DeviceDiscovered += (s, a) =>
> {
>       await adapter.StartScanningForDevicesAsync();
>  }

You need to scan then discover will occur:

adapter.DeviceDiscovered += (s, a) =>
{
     //do something
}

adapter.StartScanningForDevicesAsync();

janusw, 2025-04-15:
> This order of calls is incorrect:

Good catch! I completely missed that.

So: No bug, but misuse of the API.

closed 2025-04-15 by janusw