dotnet-bluetooth-le/dotnet-bluetooth-le #800: ISSUE with GetSystemConnectedOrPairedDevices() on Windows and iOS

Auto72, 2024-01-19:
GetSystemConnectedOrPairedDevices()  works for me on **Android**.

On **Windows** doesn't return the list of my paired BLE devices.
To Fix the problem on Windows I need to use the following code:

```
    public override IReadOnlyList<IDevice> GetSystemConnectedOrPairedDevices(Guid[] services = null)
    {
        string pairedSelector = BluetoothLEDevice.GetDeviceSelectorFromPairingState(true);
        DeviceInformationCollection pairedDevices = DeviceInformation.FindAllAsync(pairedSelector).GetAwaiter().GetResult();
        List<IDevice> devlist = ConnectedDevices.ToList();
        List<Guid> ids = ConnectedDevices.Select(d => d.Id).ToList();
        foreach (var dev in pairedDevices)
        {
            Guid id = dev.Id.ToBleDeviceGuidFromId();
            ulong bleaddress = id.ToBleAddress();
            if (!ids.Contains(id))
            {
                var bluetoothLeDevice = BluetoothLEDevice.FromBluetoothAddressAsync(bleaddress).AsTask().Result;
                if (bluetoothLeDevice != null)
                {
                    var device = new Device(this, bluetoothLeDevice, 0, id);
                    devlist.Add(device);
                    ids.Add(id);
                    Trace.Message("GetSystemConnectedOrPairedDevices: {0}: {1}", dev.Id, dev.Name);
                }
                else
                {
                    Trace.Message("GetSystemConnectedOrPairedDevices: {0}: {1}, BluetoothLEDevice == null", dev.Id, dev.Name);
                }
            }
        }
        return devlist;
    }

```
NOTE: we need to use BluetoothLEDevice, instead of the current BluetoothDevice.

On **iOS** if you call GetSystemConnectedOrPairedDevices()  you get the following error:

Value cannot be null. (Parameter 'serviceUUIDs')

```
    public override IReadOnlyList<IDevice> GetSystemConnectedOrPairedDevices(Guid[] services = null)
    {
        // NOTE: to avoid the null error, instead of using 'null' here, we need to use 'new CBUUID[1]'
        CBUUID[] serviceUuids = null;
        if (services != null)
        {
            serviceUuids = services.Select(guid => CBUUID.FromString(guid.ToString())).ToArray();
        }

        var nativeDevices = _centralManager.RetrieveConnectedPeripherals(serviceUuids);

        return nativeDevices.Select(d => new Device(this, d, _bleCentralManagerDelegate)).Cast<IDevice>().ToList();
    }
```
Anyway, even if I fix the error in iOS that according to the above note, it doesn't return my paired BLE device, that I can see listed in 'My Devices' under the Bluetooth Settings.

I need help in iOS, in order to get my paired BLE device listed by the method .GetSystemConnectedOrPairedDevices()

## Steps to reproduce
Call this method on Windows and iOS after you have already connected and paired a BLE device (BLE Just Works).

## Expected behavior
On **Windows** and **iOS** it should return the list of the paired BLE devices.

## Actual behavior
On **Windows** the list of the paired BLE devices is not returned, so you get an empty list.

On **iOS** if you call GetSystemConnectedOrPairedDevices()  you get the following error:

Value cannot be null. (Parameter 'serviceUUIDs')

### Crashlog

Value cannot be null. (Parameter 'serviceUUIDs')

## Configuration

**Version of the Plugin:** BLE.Plugin ver. 3.1.0-beta

**Platform:** iOS 17.2 / Windows 11

**Device:** iPad Air

Auto72, 2024-01-19:
[How get the list of paired bluetooth devices in swift?](https://stackoverflow.com/questions/34592179/how-get-the-list-of-paired-bluetooth-devices-in-swift)
https://stackoverflow.com/questions/34592179/how-get-the-list-of-paired-bluetooth-devices-in-swift

AskBojesen, 2024-01-22:
I have reproduced in Windows - the problem is that BluetoothDevice is different than BluetoothLEDevice in WIndows... 
I can fix the Windows implementation - I am currently the Windows dude in the code

AskBojesen, 2024-01-22:
I just opened a PR today https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/801
When that is merged I will make a new one for this...

Auto72, 2024-01-23:
**About iOS**
If my BLE device is connected and I specify the service GUID in the call **GetSystemConnectedOrPairedDevices(services)** I get in return my device. If no service GUID is specified I get the error: Value cannot be null. (Parameter 'serviceUUIDs').

What I want to achieve is getting the list of paired BLE devices (not connected), that I can see under "My Devices" in the Bluetooth settings. Reading around it seems it is not possible because **RetrieveConnectedPeripherals()** returns only connected devices.

https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518924-retrieveconnectedperipheralswith

AskBojesen, 2024-02-05:
**About Windows**
I have created this PR #807

closed 2024-05-06 by Auto72