dotnet-bluetooth-le/dotnet-bluetooth-le #803: Device Name is not showing in Android .net MAUI

Ganesh-210, 2024-01-25:
I'm working on a Bluetooth project for **Android**. As now I can discovered list of devices but I'm not getting the Name. Please anyone help me to resolve this issue. It would be very helpful for me. 

FYI : Attached image below

**Note : I'm getting Rssi , ID and Native Device address**

` public ObservableCollection<IDevice> ScannedDevices { get; } = new ObservableCollection<IDevice>();`
  ```
  var ble = CrossBluetoothLE.Current;
    var adapter = CrossBluetoothLE.Current.Adapter;
    adapter.DeviceDiscovered += (s, e) =>
    {
        if (!ScannedDevices.Contains(e.Device))
        {
            ScannedDevices.Add(e.Device);
            string sss = e.Device.Name;
        }
    };

    adapter.StartScanningForDevicesAsync();
}
```

Steps to reproduce
Call this method on **Android** after you have already connected and paired a BLE device (BLE Just Works).

## Crashlog
Name cannot be null. 

## Expected behavior
On **Android**, it should return the list of the paired BLE devices. Device Name is null and its not showing
On **Android** the list of the paired BLE devices is returned . So I can pair the device.

## Actual behavior
[image]

## Configuration

**Version of the Plugin:** 3.1.0 

**Platform:** Android 13

**Device:** Moto E13

axa88, 2024-01-27:
There seems to be a couple of concerns with your report.
Firstly, what certainty do you have that all the devices around you even have a name?
Many don't.

Secondly, what is the expected influence on your outcome if any particular device is paired/connected, why note it?
A paired device may still advertise and be found by a scan.  And afaik a connected device most likely will not advertise and thus not be discovered.  Neither which influences if a device even has a name. 
At best Android can allow you to change the device `Ailias`, but you mention none of that here.

Ganesh-210, 2024-01-29:
> There seems to be a couple of concerns with your report. Firstly, what certainty do you have that all the devices around you being scanned/discovered even have a name? Many don't.
> 
> Secondly, what is the expected influence on your outcome if any particular device is paired/connected, why note it? A paired device may still advertise and be found by a scan. And afaik a connected device most likely will not advertise and thus not be discovered. Neither which influences if a device even has a name. At best Android can allow you to change the device `Ailias`, but you mention none of that here.
> 
> Lastly, you already reported your "issue" #802

Thanks for the reply. I can get the name of paired Bluetooth devices.

I need to get all the Bluetooth Device Names which is nearby which is not paired.

` string _devname = e.Device.Name;` - Not getting all the device names and showing empty
` string _devid = e.Device.ID;` - Getting  all the device UUID's
` string _devrssi = e.Device.Rssi;` - Getting all the device Rssi's

axa88, 2024-01-29:
> Thanks for the reply. I can get the name of paired Bluetooth devices.

What is your results with an diagnostic tool such as nRF Connect?

Ganesh-210, 2024-01-29:
> > Thanks for the reply. I can get the name of paired Bluetooth devices.
> 
> What is your results with an diagnostic tool such as nRF Connect?

Check the attached image.

[image]

axa88, 2024-01-29:
What I have been attempting to explain to you is that not all devices have a Name.
So to convince yourself that the devices that you claim to have a name, actually have a name, I suggest you download a tool like nRF Connect for your mobile device and compare those results with yours.

Ganesh-210, 2024-01-30:
@axa88 Thanks ..it helped me a lot :D

axa88, 2024-01-30:
> @axa88 Yeah I have downloaded the nRF Connect app and checked .My App results and nRF Connect app results are same.it helped me a lot :D

Great. Why not close the issue...

Ganesh-210, 2024-01-30:
Done

closed 2024-01-30 by Ganesh-210