dotnet-bluetooth-le/dotnet-bluetooth-le #804: Missing Characteristics on CitySports Treadmill

hughesjs, 2024-01-31:
## Context

I'm trying to send requests to every discovered characteristic whilst doing a packet capture so that I can work out what UUIDs correspond with these handles:

[image]

## Demonstration Code

https://github.com/hughesjs/OpenEQiSports/tree/6bd2ec243acc91637b96031829c55f5e1087440b/proof-of-concept/EqiPoc

(This isn't a standalone repo, but the only thing in it so far is a PoC displaying this issue so it may as well be as long as you look at the linked commit specifically)

## Steps to reproduce

1. Launch app with debugging
2. Either break on TreadmillController:46 and look at the characteristics found (specifically for `service.Name == "Fitness Machine"`
3. Or press debug and look at the console output

## Expected behavior
According to nRF Connect, the "Fitness Machine" service has 14 characteristics.

[image]
[image]

## Actual behavior
Only 9 characteristics are discovered.

### Crashlog

N/A

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** iOS 17.3

**Device:** iPhone 15 Pro

janusw, 2024-01-31:
> ## Expected behavior
> 
> According to nRF Connect, the "Fitness Machine" service has 14 characteristics.

It seems your screenshot contains the service twice (with only 7 characteristics each)?

> ## Actual behavior
> 
> Only 9 characteristics are discovered.

Which ones do you see? Which ones are missing?

hughesjs, 2024-01-31:
@janusw - My bad, I accidentally pasted the same screenshot twice... I always forget on macOS you have to manually copy the screenshot... I've editted it.

That being said, I think I've worked out that the three lines means some of those are actually descriptors. So, I've been trying to write to them with this:

```cs
    private async Task DoHandshake()
    {
        if (_device == null) return;
        
        string[] characteristicUuids = ["2acd", "2ad3", "2ad9", "2ada", "c4208999-8d92-bee1-4456-2068528eccf6"];
        byte[][] values = [[0x01, 0x00], [0x01, 0x00], [0x02, 0x00], [0x01, 0x00], [0x01, 0x00]];
  
        for (int i = 0; i < characteristicUuids.Length; i++)
        {
            ICharacteristic? characteristic = _characteristics[characteristicUuids[i]];
            byte[] payload = values[i];
            IDescriptor? descriptor = (await characteristic.GetDescriptorsAsync()).Single();
            
            await descriptor.WriteAsync(payload);
        }
    }
```

With the UUIDs taken from hitting things through nRF Connect and looking at a packet capture to link the handles I need and it's all of the Client Characteristic Configurations

Now I'm getting this exception:

> ObjCRuntime.ObjCException: Objective-C exception thrown.  Name: NSInternalInconsistencyException Reason: Client Characteristic Configuration descriptors must be configured using setNotifyValue:forCharacteristic:

So I guess I'm writing to them wrong?

hughesjs, 2024-01-31:
Y'know, I might have actually misunderstood what's going on here... That might just be setting up notifications rather than being a handshake. Sorry if I've wasted your time here.

janusw, 2024-02-01:
> @janusw - My bad, I accidentally pasted the same screenshot twice... I always forget on macOS you have to manually copy the screenshot... I've editted it.
> 
> That being said, I think I've worked out that the three lines means some of those are actually descriptors. Which would explain the "missing" ones.

Ah, right. Those descriptors are probably the only thing that the plugin will recognize properly at present. All the other characteristics and the service itself are pretty much "unknown" AFAICS, but maybe you could open a PR to add them to these files?

* https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/Plugin.BLE/Shared/KnownServices.cs
* https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/master/Source/Plugin.BLE/Shared/KnownCharacteristics.cs

hughesjs, 2024-02-01:
Yeah sure, I can do that at some point this weekend probably

hughesjs, 2024-02-01:
There's a PR for you :)

janusw, 2024-02-02:
> There's a PR for you :) 

Awesome, thanks :+1:

I'll look into it soon ...

hughesjs, 2024-02-02:
Let me know if you need anything

PR #805 (2024-02-06): Added services for citysports treadmill
Fixes #804
closed 2024-02-06 by janusw