dotnet-bluetooth-le/dotnet-bluetooth-le #810: WinUI UpdateRssi method shouldnt return true if the function isnt (currently) supported

axa88, 2024-02-08:
It strikes me that for now this should return false on WinUI

 public override Task<bool> UpdateRssiAsync()
 {
     //No current method to update the Rssi of a device
     //In future implementations, maybe listen for device's advertisements

     Trace.Message("Request RSSI not supported in Windows");

     return Task.FromResult(true);
 }

janusw, 2024-02-21:
> It strikes me that until there is support on WinUI, this should return false

Yes, I think you're right. It's easy to fix that. PR welcome ;)

axa88, 2024-02-22:
> > It strikes me that until there is support on WinUI, this should return false
> 
> Yes, I think you're right. It's easy to fix that. PR welcome ;)

LoL. Alright... I'll see to finally making a fork and push this one out.

axa88, 2024-03-21:
> > It strikes me that until there is support on WinUI, this should return false
> 
> Yes, I think you're right. It's easy to fix that. PR welcome ;)

Ive now tried to pull and build several times, I just cannot and dont have the time to figure out what is wrong with my environment.
Problems from missing versions of out of support frameworks (.netcore 5?) to other ambiguous errors i cant resolve...

@janusw 
Please feel free to make this change if possible

PR #844 (2024-05-05): fix Device.UpdateRssiAsync on Windows
* updating the RSSI is currently not supported on Windows, therefore the method should return a false value
* also: update the documentation and fix some typos
* this fixes #810
closed 2024-05-05 by janusw