dotnet-bluetooth-le/dotnet-bluetooth-le #815: WinUI AdvertisementRecords only list a partial distribution of all advertised device records

axa88, 2024-02-21:
It appears to me there is an issue with collection of `IDevice.AdvertisementRecords` while scanning on `WinUI`.

I have found when a device advertises, WinUI the `AdvertisementRecords` will tend to intermittently lose/be overwritten by other `AdvertisementRecords` upon each subsequent `Adapter.DeviceAdvertised` events received.  And rarely if ever do all advertised data appear in the `AdvertisementRecords` at the same time.

Im unsure if this is a OS/hardware limitation but here are my observations.

Here is my esp32 device advertising 4 separate `AdvertisementRecord`s
[image]

And Here is my MAUI app running on Android will list all 4 devices as they are found in `IDevice.AdvertisementRecords` upon `Adapter.DeviceDiscovered` event, as well as every subsequent `Adapter.DeviceAdvertised` event raised.

[image]

While the same application running on WinUI will only list part of the advertised Records upon `Adapter.DeviceDiscovered` and each subsequent `Adapter.DeviceAdvertised` event raised.
Here, the 4 Records were consistently split into 2 different distributions, 3 records, and 1 record:

[image]

I couldn't observe any pattern to the inconsistency, any event might return either the 1 record or 3 records.

Note the advertisement pack over 31 bytes; could this be getting broken up by the underlying BLE stack causing inconsistency on the different platforms?
So I wonder if this issue relates to #814 where Ive found `ScanFilterOptions` to also not work on `WinUI`

## Expected behavior
`IDevice.AdvertisementRecords` should consistently list all records provided by a device after initial `Adapter.DeviceDiscovered` and each subsequent `Adapter.DeviceAdvertised` event raised.

## Actual behavior
`IDevice.AdvertisementRecords` somewhat randomly list a partial distribution of all available records, seemingly changing upon `Adapter.DeviceDiscovered` and each subsequent `Adapter.DeviceAdvertised` event raised.

### Crashlog

No crashes

## Configuration

**Version of the Plugin:**
v3.0.0

**Platform:** 
Not working .NET8 Win10.0.19041.0 
Working on .NET8 Android 12/13 target 34

**Device:** IBM X1 carbon 1, Samsung A11, Samsung A51

axa88, 2024-02-22:
@janusw 

Ill dig out the firmware and change the Advertising PDU payload to 31 bits or less and see what changes.
As far as what i know for sure, the BLE peripheral that is advertising,  ESP32 (ESP32-D0WD-V3 (revision 3)) => BLE 4..2 => so it doesn't support extended Advertising.  Just as reported on my A51 Android ble 4.2 device, but that platform does collect all in `AdvertisementRecords`

My windows device Windows X1 carbon gen 6 => BLE 4.1 fwiw, so its at least a windows issue in this case.

The work around is to collect all the records starting with the initial device on and `DeviceDiscovered` and add to it with each `DeviceAdvertised` event

This seems to work well but dont know if that's an appropriate thing to do overall if not positive of the underlying issue.

axa88, 2024-02-23:
@janusw 

> If extended advertising is supported at some Windows API level, this method needs to be properly implemented on Windows.

What about [System.DeviceInterface.Bluetooth.ProductVersion](https://learn.microsoft.com/en-us/windows/win32/properties/props-system-deviceinterface-bluetooth-productversion)  Then testing for version >= 5 ? 

I guess there might be more to it than just testing the version, otherwise Android and iOS might just return the version and not have a dedicated calls for extended advertising support

axa88, 2024-02-25:
Sure enough shortening the advertisement to non extended limitations this packet fragmentation no longer seem to manifest on windows.

Ive no real idea how a device that doesn't support extended advertisements, ESP32 BLE 4.2, manages to send _all_ of data over spec limit of 31 byte payload, but it seems Android gathers the data correctly while windows doesn't.

janusw, 2024-05-05:
> What about [System.DeviceInterface.Bluetooth.ProductVersion](https://learn.microsoft.com/en-us/windows/win32/properties/props-system-deviceinterface-bluetooth-productversion) Then testing for [...]

Right. Testing the supported BLE version is not enough here. An adapter might only support specifc features of BLE 5, but not all of them.

The necessary API endpoint for checking support for extended advertising on Windows seems to be this one:

https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothadapter.isextendedadvertisingsupported

axa88, 2024-07-14:
Closing this. It seems that it may be inherent to Windows that Discovered devices are returned to the API quickly and may not contain all associated Advertisement data.  There seems to be a delay or at least the option to delay (using the MATCH_MODE parameter) before most if not all associated advertisements had been received. 

If i a similar option on Windows is found it might be revisited.

closed 2024-07-14 by axa88