dotnet-bluetooth-le/dotnet-bluetooth-le #816: GetCharacteristicAsync returns null when characteristic is present

nathenxbrewer, 2024-02-21:
I am developing an app that utilizes BLE. I have a characteristic with UUID da6d0fee that resides in the da6d0fe1 service. When I 
call service.GetCharacteristicAsync, it is returning null. This is not happening on all of our products so I'm not sure what is going on here. The characteristic is present in nrfConnect.

[image]

Plugin.BLE reports 7 characteristics in this service, and the 7th ends at da6d0fed. 
[image]

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
3.1.0-beta.2

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS

**Device:** e.g. HTC Sensation /i Phone 7 ...
iPhone 15 Pro Max

axa88, 2024-02-23:
> The characteristic is present in nrfConnect.

> nrfConnect for iOS shows it's missing as well

This is confusing.  Are you trying to say the characteristic is present in nRF on (an)other platform(s), but not iOS?

interesting. though nRF is the gold standard, if its also missing the characteristic, i wouldn't at all expect its an issue of this library.

nathenxbrewer, 2024-02-24:
I ended up rebooting the phone a few times and it came back. Must have been the OS caching the values. I had just finished a Firmware update on the device and that characteristic was not present in earlier firmware versions.

closed 2024-02-24 by nathenxbrewer