dotnet-bluetooth-le/dotnet-bluetooth-le #817: Grammatical and formatting opportunities

abergquist, 2024-02-22:
Please see the attached document for various grammatical and formatting opportunities.
[Bluetooth LE plugin for Xamarin and MAUI.docx](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/files/14368780/Bluetooth.LE.plugin.for.Xamarin.and.MAUI.docx)

All edits are in red.

The only thing that might need explanation is the red right arrows that end in a red vertical line; they indicate a request to indent all the lines that are immediately to the right of the entire vertical line (in all cases, it's some type of code).

Note also that I marked a few edits with a dark (black) checkmark; that indicates that that was an edit that was resolved in an earlier issue (Issue #813).

janusw, 2024-02-22:
> Please see the attached document for various grammatical and formatting opportunities.

Wow, thanks for taking such a careful look at our README, but actually hand-written notes on a printed-out version of the document are a **very** old-school way of reporting such issues 😆

It would be much more useful (read: easier to handle) if you could open a pull request that fixes the formatting problems you found.

Also, I'm afraid that comments like "move to the next page" are not very suitable, since the README file is not actually optimized for being printed out in A4 (or any other format), and this is nothing we will be aiming for.

abergquist, 2024-02-22:
No problem, @janusw; I'll be gone for a few days. I'll work on this next week. Thanks.