dotnet-bluetooth-le/dotnet-bluetooth-le #821: Enable/disable Bluetooth on Android

janusw, 2024-02-29:
This is a follow-up to PR #784, which introduced a method `IBluetoothLE.TrySetStateAsync` for enable or disabling the Bluetooth radio (and only implemented in on Windows for the start).

It seems that this should also be possible on Android, via the method mentioned here:
https://stackoverflow.com/questions/76858093/how-to-activate-deactivate-bluetooth-in-net-maui

Would be great to have this implemented. 😃

janusw, 2024-02-29:
@axa88 Would you like to look into this? Otherwise I might take a stab ...

axa88, 2024-02-29:
> @axa88 Would you like to look into this? Otherwise I might take a stab ...

It's certainly on my radar.

midoragh, 2024-03-02:
JFI: I implemented this on App level because starting with Android 13 (API 33) you you are no longer allowed to call the method to enable/disable BLE (var setStateOnOff = bleCurrent.GetType().GetMethod("SetStateOnOff");).
Instead you have to call something like
StartActivityForResult(intentBtEnabled, REQUEST_ENABLE_BT);
in an Activity. This call will popup a request to the user if he likes to switch on BLE.

janusw, 2024-03-03:
> JFI: I implemented this on App level because starting with Android 13 (API 33) you you are no longer allowed to call the method to enable/disable BLE (var setStateOnOff = bleCurrent.GetType().GetMethod("SetStateOnOff");).

Yes, the enable/disable methods of the BluetoothAdapter are deprecated in API 33:

https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#enable()

> Instead you have to call something like StartActivityForResult(intentBtEnabled, REQUEST_ENABLE_BT); in an Activity. This call will popup a request to the user if he likes to switch on BLE.

I assume you mean `BluetoothAdapter.ACTION_REQUEST_ENABLE`, which was suggested in the Stackoverflow thread mentioned above.

https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#ACTION_REQUEST_ENABLE

axa88, 2024-03-03:
> @axa88 Would you like to look into this? Otherwise I might take a stab ...

@janusw I'm jammed with work for the foreseeable future and not up to speed on the workings of this library.  Please feel free to implement at your convenience. 

And while at it perhaps confirm the `IBluetooth` `State` properties and event on `WinUI`.  I'm pretty confident there is issue

janusw, 2024-03-03:
> @janusw I'm jammed with work for the foreseeable future and not up to speed on the workings of this library. Please feel free to implement at your convenience.

Ok, thanks for letting me know. I have just opened PR #823 for it (and would be grateful for some testing and feedback) ....

janusw, 2024-03-04:
Fixed via #823.

closed 2024-03-04 by janusw