dotnet-bluetooth-le/dotnet-bluetooth-le #822: [Android] Bluetooth radio off is not reported

axa88, 2024-03-01:
Upon switching an Android Bluetooth radio off, report IDevice.State remains On and IBluetoothLE.StateChanged event should trigger.

In contrast this works properly on WinUI

## Steps to reproduce

1. switch radio off

2. observe IDevice.State remains On

3. observe UponIBluetoothLE.StateChanged event doesnt trigger.

## Expected behavior
Upon switching an Android Bluetooth radio off, IDevice.State should propagate to Off and IBluetoothLE.StateChanged event should trigger.

## Actual behavior
Upon switching an Android Bluetooth radio off, report IDevice.State remains On and IBluetoothLE.StateChanged event should trigger.

### Crashlog

none

## Configuration
Version of the Plugin:
v3.0.0

Platform:
Not working .NET8 Win10.0.19041.0
Working on .NET8 Android 12/13 target 34

Device: IBM X1 carbon 1, Samsung A11, Samsung A51

janusw, 2024-03-01:
I cannot confirm this with v3.1.0-beta.2 and .NET 8 (on a Samsung Galaxy S21+ with Android 14).

When Bluetooth is disabled, I even see two `StateChanged` events:
* On -> TurningOff
* TurningOff -> Off

axa88, 2024-03-18:
@janusw 

To simplify my concern here, perhaps you could explain what you believe should happen when the Radio is turned off.

Right now as I (eventually) reported, on `Android` there is no update about the lost connection.  By that i mean the `IsConnected` property doesn't change, and the DeviceDisconnected event doesn't trigger.

While `WinUI` the `IsConnected` property does change and a `DeviceConnected` event does trigger.

One might argue that since the radio is off, the devices are no longer valid as Android will require a requisition of devices to get a new Gatt object...  But i think this is a weak argument and Id argue this should be consistent across platforms.

Again sorry to confuse this with the `State` property and event of `IBluetoothLE`
So if you justify the different behavior, ill close the issue.

axa88, 2024-12-18:
Looking into this, it seems whether this is a bug or not is subjective.
  
Simply put `IDevice.State` is not stateful.  Rather the `IDevice.State` property wraps a method to read the current `DeviceState` from each native device.  And there is no simple nor obvious mechanism that a non familiar user can use to know they must manually keep in sync.

Meaning as long as the radio is on, the Device Connection events can be used to notify the application that the `State` must be read.  But when the radio is disabled, none of that happens.  Never mind that there should've just been a single event, but there are **4!** Connection related events that don't do anything when the device disconnects/lost due to disabled radio.

But ill concede what happens after radio disable is undefined.