dotnet-bluetooth-le/dotnet-bluetooth-le #824: WriteAsync does not use the CancellationToken that is given to it

minno726, 2024-03-06:
## Steps to reproduce

1. Call `WriteAsync` by passing a `CancellationToken` to it.

2. Before the call completes, cancel the token. This is easiest to do by passing a very small timeout in the constructor for `CancellationTokenSource`.

## Expected behavior
The operation should be canceled.

## Actual behavior
The operation ignores the cancellation.

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** Observed on iOS due to #785, but the problem is in shared code.

The source of the problem is here:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b179852efef3402a74c3c513dfce004c9c158077/Source/Plugin.BLE/Shared/DescriptorBase.cs#L68-L76

The code does not use the cancellation token that was passed in as a function parameter. See the documentation on how to use cancellation tokens here: https://learn.microsoft.com/en-us/dotnet/standard/threading/cancellation-in-managed-threads, or remove the parameter if cancellation isn't practical to implement.

This is related to #785, since it prevents one workaround for that issue from working.

ygl-rg, 2024-03-21:
```c#
public static async Task<int> SendBytes(this ICharacteristic handle, byte[] data, CancellationToken token)
    {
        var cs = new TaskCompletionSource();
        await using(token.Register(() =>
        {
            cs.TrySetResult();
        }));
        var t1 = handle.WriteAsync(data, token);
        var t = await Task.WhenAny(cs.Task, t1);
        if (t == cs.Task)
            throw new TimeoutException("Send bytes timeout");
        return await t1;
    }
```

I use the above method for cancellation.

bcaceiro, 2024-05-21:
Hi @janusw , are you planning on addressing this issue? We have a temp workaround, however would be best if the cancellation would respect the given CancellationToken

janusw, 2024-05-25:
> Hi @janusw , are you planning on addressing this issue?

To be honest, I'm currently not planning to look into this myself (due to time constraints), but if someone is willing to contribute a PR, I'd be happy to review it and provide support.

In general, I'm not actully understanding my maintainer role here as "someone who is responsible for fixing all the bugs", but rather as "someone who is helping others to help themselves". 😄 

> And great work on this package btw :)

Thanks, but this compliment certainly does not go to me alone. A lot of people have contributed to this library over the years, and I'm only one of them. It's not "my" package in any way, it's a shared community effort.

My advice: If you need something fixed quickly, the best idea is usually to get involved yourself. 😉
(Or, in case you don't have the time or the skills for contributing a fix, but have some budget to spend: Find someone who can fix it, and pay him for it.)

janusw, 2024-10-22:
This should be fixed by #880, so I'll close the issue. If anything does not work, please open a new one.

closed 2024-10-22 by janusw