dotnet-bluetooth-le/dotnet-bluetooth-le #825: App .NET MAUI Crash on iOS

acaliaro, 2024-03-08:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

I am debugging a MAUI app on a real iOS device. Frequently the app crashes with the below stack trace. I can't say the precise instruction on which it crashes

## Steps to reproduce

1.

2.

3.

## Expected behavior
It should not crash

## Actual behavior
It crashes

### Crashlog

If something causes an exception paste full stack trace + Exception here

[0:] An error occurred: 'Object reference not set to an instance of an object.'. Callstack: '   at Plugin.BLE.iOS.Characteristic.NSErrorToGattStatus(NSError error)
   at Plugin.BLE.iOS.Characteristic.<>c__DisplayClass17_1.<StartUpdatesNativeAsync>b__7(Object sender, CBCharacteristicEventArgs args)
   at CoreBluetooth.CBPeripheral._CBPeripheralDelegate.UpdatedNotificationState(CBPeripheral peripheral, CBCharacteristic characteristic, NSError error) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/build/dotnet/ios/generated-sources/CoreBluetooth/CBPeripheral.g.cs:line 528
   at UIKit.UIApplication.UIApplicationMain(Int32 argc, String[] argv, IntPtr principalClassName, IntPtr delegateClassName) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:line 58
   at UIKit.UIApplication.Main(String[] args, Type principalClass, Type delegateClass) in /Users/builder/azdo/_work/1/s/xamarin-macios/src/UIKit/UIApplication.cs:line 94
   at daze.dazeapp.Program.Main(String[] args) in C:\Daze\daze.app.maui\daze.dazeapp\daze.dazeapp\Platforms\iOS\Program.cs:line 13
   at System.Reflection.MethodBaseInvoker.InterpretedInvoke_Method(Object obj, IntPtr* args)
   at System.Reflection.MethodBaseInvoker.InvokeDirectByRefWithFewArgs(Object obj, Span`1 copyOfArgs, BindingFlags invokeAttr)'

## Configuration

**Version of the Plugin:** e.g. 3.0.0

**Platform:** iOS 17.3

**Device:** iPhone 12

acaliaro, 2024-03-15:
Hi. there are some news about this issue? Thanks

janusw, 2024-03-15:
Unfortunately it's not clear how to reproduce this.

Your call stack seems to point to this method:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b179852efef3402a74c3c513dfce004c9c158077/Source/Plugin.BLE/Apple/Characteristic.cs#L289-L299

The only way I can see for an exception to occur is if the `error` parameter is null. Out of the three use cases of the method, two are guarded for this condition, but one is not (quite the contrary, it always seems to be called with null parameter here):

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/b179852efef3402a74c3c513dfce004c9c158077/Source/Plugin.BLE/Apple/Characteristic.cs#L226-L234

The method `NSErrorToGattStatus` was introduced in commit https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/f5a12b28fac1a30bce1be9cf3840487a93c66f72 by @chrimaka.

janusw, 2024-03-18:
The problem was introduced in version 3.0.0-beta.4, and affects all versions released since then, including 3.0.0 and up to 3.1.0-beta.2.

Is it fixed on master via #827. The fix will be included in the next release.

closed 2024-03-18 by janusw

acaliaro, 2024-03-19:
Hi @janusw thanks for your work. When will be released this patch on nuget?

janusw, 2024-03-19:
> Hi @janusw thanks for your work. When will be released this patch on nuget?

I'd like to tag a new beta release this week, but there are still two pending PRs that I want to include.

janusw, 2024-03-25:
I have published version 3.1.0-beta.3 today, which includes the fix for this issue.