dotnet-bluetooth-le/dotnet-bluetooth-le #826: Windows: More than one lost connection does not work correctly

CodeMusher, 2024-03-08:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. I run the WinConsole application. Press D1 to listen for devices, Select my Device and press D9 to run Connect -> Loop: ConnectionLost -> Connect 
2. The first time I power cycle my device it clearly state : DisconnectedPeripheral by lost signal:

3. Power up the device => Normal behavior

4. Power cycle the device again. Now it states: DisconnectedPeripheral by user: 

## Expected behavior
It should not matter now many times I power cycle the device. It should always be DisconnectedPeripheral by lost signal

## Actual behavior
It works one time! The first power cycle operation causes the device to be put in the disconnectingRegistry

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:**
3..0-beta-2.

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
Windows
**Device:** e.g. HTC Sensation /i Phone 7 ...
Laptop

janusw, 2024-03-16:
> **Version of the Plugin:** 3..0-beta-2.

This is supposed to be 3.1.0-beta.2, I assume?

CodeMusher, 2024-03-17:
Sorry, my bad! 3.1.0-beta.2 is correct

CodeMusher, 2024-04-07:
This could be solved by adding this line
disconnectingRegistry.Remove(id);
in the Device_ConnectionStatusChanged method of the Adapter class

janusw, 2024-05-04:
> This could be solved by adding this line disconnectingRegistry.Remove(id); in the Device_ConnectionStatusChanged method of the Adapter class

Hm, looks like that line is there already ...

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/04a5d273324941ebc9069b179939e4b9b43dad40/Source/Plugin.BLE/Windows/Adapter.cs#L148-L151

... but maybe it's not always called?

@AskBojesen This seems to be your code (see #809). Care to have a look?

AskBojesen, 2024-05-06:
I will try to reproduce with latest later today or tomorrow

AskBojesen, 2024-05-06:
> @janusw, @AskBojesen Sounds good. One more thing I need is the possibility to register for updates on a characteristic that is only INDICATE, not NOTIFY. That would be nice.

@CodeMusher: Please register a new issue, so we don't get more confused...
I will ignore your comment in this issue

CodeMusher, 2024-05-06:
@AskBojesen Sorry, I'll do that of course.

AskBojesen, 2024-05-06:
IMO this is not a bug in the plugin, but a wrong use of the API in the example code I have made to test something....
Take a look here: https://github.com/AskBojesen/dotnet-bluetooth-le/blob/master/Source/BLE.Client/BLE.Client.WinConsole/PluginDemos.cs line 166

This method does in fact not test connection lost > reconnect...

In our application we listen for connection lost and handles that with a reconnect,..

If this is to be "fixed" the WinConsole demo method must be fixed to show how it can work, but it is not related to the API and functionality it self - what do you say @janusw : Should I use the effort to show how this can work with the existing API (fix this integration test method) ?
 I can do that within a week - but it is only the winconsole demo... Which is now used - and I appreciate that, but it has consequences: All test methods must be valid

[image]

CodeMusher, 2024-05-06:
@AskBojesen Does the WinConsole demo work the same when you tried it or have I missed something?
I think it would be a great thing to describe how the connection lost/reconnect should be implemented for this to work as designed.

CodeMusher, 2024-05-07:
@AskBojesen, @janusw In my application I also listen for lost connection event and start a reconnection process. Since I only managed to get the lost connection event once I tried to find any errors. 
I used your WinConsole Demo try to locate the error. I came to the conclusion that the first lost connection invoked the lost connection event but it also put the device in the disconnectingRegistry. That caused the next lost connection to be handled as if the user disconnected and no lost connection event was invoked. Added a line in the source code, removing the device from the disconnectingRegistry every time we got connected kind of solved it. 

If this is not a bug and it works for you on all platforms, then something must be missing in my use of the API. 
Maybe you could provide some pseudocode to to do this properly?

AskBojesen, 2024-05-08:
Well now I have improved the demo in the winconsole.
PR: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/849

AskBojesen, 2024-05-08:
I found the bug and fixed the issue - sorry for my stubbornness from believing   there was a bug.
Fixed now in PR: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/849

CodeMusher, 2024-05-09:
@AskBojesen good to hear that I wasn’t wrong about this😉
I checked your log file and noticed that the second lost connection differs from the others. There is no gatt closed messages. Should the disconnect not be identical?

AskBojesen, 2024-05-09:
> @AskBojesen good to hear that I wasn’t wrong about this😉 I checked your log file and noticed that the second lost connection differs from the others. There is no gatt closed messages. Should the disconnect not be identical?

No I don't see it as an error with the  GattSession_SessionStatusChanged - we could leave it out of the log.
I guess it is some timeng when the GattSession is disposed

closed 2024-05-15 by janusw