dotnet-bluetooth-le/dotnet-bluetooth-le #829: [BUG] NET MAUI on iOS, I read a wrong name connectiong to a Linux device

acaliaro, 2024-03-21:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

On a Linux machine, they change the name of the bluetooth device, the "Name" property is not changed but an "Alias" is created. Using the library on an iOS device (iPhone), the library continues to read "Name" and it is not possible to recover the value of "Alias", which is instead recovered if an Android device is used.

2.

3.

## Expected behavior
Tell us what should happen
The

## Actual behavior
Tell us what happens instead

The library should read the Alias instead of the Name, or return the Alias at least in the native part. By downloading some third-party iOS apps, the Linux device is found and the Alias content is displayed

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 3.1.0-beta.2-10-gf6b048a

**Platform:** iOS 17.3

**Device:** iPhone 12

janusw, 2024-03-23:
Note: We have a couple of open issues concerning device names on iOS, e.g. #541, #698, #703, #749. Could you check if any of them is similar to your situation?

There are even two experimental PRs in that direction, but they did not seem to fix anything: #711, #792

janusw, 2024-03-23:
> On a Linux machine, they change the name of the bluetooth device, the "Name" property is not changed but an "Alias" is created. Using the library on an iOS device (iPhone), the library continues to [...]

Are you only looking at the advertisements, or are you connected to the device while the name changes?

I'm not sure which effect such an alias has on the advertisements. Could you check what kind of data the device is advertising before and after the change?

acaliaro, 2024-03-25:
Hi, 

on first discovery we tried to cast IDevice to CBPeripheral, which has the name property containing the name, with the unexpected value. Inspecting the istance of IDevice we can see that there's a property called AdvertisementRecors, but unfortunately is empty.

We are not connected to the device, until it is found by his name during the scan.

CC @Gaburiere-Remira

acaliaro, 2024-04-04:
Hi @janusw I have tried 3.1.0-beta.3 the problem seems to be present, although it seems to me that sometimes the name has been returned correctly..

acaliaro, 2024-04-04:
@janusw I think the problem is here

    
```
   var name = e.Peripheral.Name;
    if (e.AdvertisementData.ContainsKey(CBAdvertisement.DataLocalNameKey))
    {
        // iOS caches the peripheral name, so it can become stale (if changing)
        // keep track of the local name key manually
        name = ((NSString)e.AdvertisementData.ValueForKey(CBAdvertisement.DataLocalNameKey)).ToString();
    }
```
I think you should have a separate property (LocalName ???) to store the advertisementdata local name so that the developer can decide which one to use

acaliaro, 2024-04-10:
I think it is been solved using _adapter.DeviceAdvertised

closed 2024-04-10 by acaliaro