dotnet-bluetooth-le/dotnet-bluetooth-le #830: Can't detect device

vfiksdal, 2024-03-21:
Hi, and thanks for a very nice library :-)

I'm trying to use the BLE.Client.MAUI example to detect a custom hardware with a fairly standard BLE compliant chipset. I'm pretty sure the scanner itself works as intended, because it's picking up my headset, tv etc. Unfortunately it fails to detect the intended hardware. Said hardware is detected by Windows itself, a legacy UWP app, as well as a couple of third party BT scanners I downloaded (With bluetooth address 00:13:43:BE:43:CB), but not by the MAUI app.

This problem would be difficult to reproduce as it relies on custom hardware, but maybe you would be able to give me some pointers as to how I could address the problem? I've tried changing the ScanMode and ScanMatchMode parameters, but I don't really see any other alternatives.

janusw, 2024-03-22:
Two questions:
* Could you try `BLE.Client.MAUI` on a different operating system, in order to check if you see the device there? Android might be the easiest, but iOS and MacOS are also supported (if you have access to an Apple device).
* How does the device advertise itself? What kind of data does the advertisement include?

vfiksdal, 2024-03-22:
Hi, and thanks for a swift reply to a rather unreasonable report :-)

Good idea, I'll try to get the project and/or your sample project loaded onto my tablet over the weekend. It's a couple of years old though, so it may have some hoops of it own to overcome.

As for the advertisement I don't really know much about it, maybe you can recommend a third party app which provides more in-depth details? I just randomly downloaded Medeival Bluetooth Network Scanner (Which in retrospect doesn't sound too promising)  and it provided the following data:
[image]
[image]

Thanks again, and I'll get back to you ASAP on how it works on Android.

janusw, 2024-03-22:
> As for the advertisement I don't really know much about it, maybe you can recommend a third party app which provides more in-depth details?

nRF Connect is a pretty good tool. I mostly used it on Android, but it seems there is also a Desktop variant that runs on Windows:

https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-desktop

> I just randomly downloaded Medeival Bluetooth Network Scanner (Which in retrospect doesn't sound too promising) and it provided the following data: 

It seems this doesn't show any details on the actual advertisement data. nRF Connect will be able to reveal that.

vfiksdal, 2024-03-23:
Unfortunately, I can't get this tool to scan in Windows. Might be because I have several systems running a newer version of JLink.

It did work in Android though, and interestingly it doesn't detect the device either. Android itself do find the device though, so the radio itself is not the problem.

Can this be some kind of compliance issue? The datasheet for the chipset says it's 4.2 compliant. That's kind of old now?

janusw, 2024-03-23:
> It did work in Android though, and interestingly it doesn't detect the device either. Android itself do find the device though, so the radio itself is not the problem.

Just to clarify what you are talking about: Did you try nRF Connect or BLE.Client.Maui on Android? If the Android OS sees the device, then both of these tools should in theory see it as well (unless it's a Bluetooth Classic device, which does not support BLE) ...

vfiksdal, 2024-03-23:
It's nRF Connect that failed to detect it. I'll need to set permissions etc to get my own project running on android, and haven't gotten around to it yet.

The chipset supports both LE and Classic, and the firmware does specify that it uses LE, but it may be using Classic due to a bug. Do you know a simple way to identify this?

Thanks a lot for your help again.

vfiksdal, 2024-03-23:
Looking at the firmware source, it seems like it is indeed BLE. All of these are enabled:

 * Bit 0 - Enable / Disable LE Secure connection.
 * Bit 1 - Enable / Disable LL Connection parameter request.
 * Bit 2 - Enable / Disable LL ping procedure (Supported only in TC35680/81).
 * Bit 3 - Enable / Disable Support for 2M PHY (Supported only in TC35680/81).
 * Bit 4 - Enable / Disable Support for LE Coded PHY (Supported only in TC35680/81).
 * Bit 5 - Enable / Disable Support for LE Extended Advertising (Supported only in TC35680/81).
 * Bit 6 - Enable / Disable Support for LE Data length change event (Supported only in TC35680/81).

I assume "secure connection" means encrypted, but this should not apply to the discovery?

axa88, 2024-03-27:
> It's nRF Connect that failed to detect it.

fwiw, nRf on android is the gold standard imho.
If your devices isnt detectable with that, it either doesn't want to be found or more likely something about the device is incompatible

vfiksdal, 2024-03-27:
Apparantly outside the scope of the library. Thank you for your input on the issue.

closed 2024-03-27 by vfiksdal