dotnet-bluetooth-le/dotnet-bluetooth-le #831: when "OnConnectionStateChange: GattStatus: InsufficientAuthorization" is shown, the ble will hang forever

ygl-rg, 2024-03-25:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. connect to a ble device

2. walk away until it disconnects, and go back reconnect it again

3. after several times, the trace will print "OnConnectionStateChange: GattStatus: InsufficientAuthorization"

## Expected behavior
dispose the device and scan again, everything should be fine.

## Actual behavior
the ble is mailfunctional, the only way to bring it back is to restart the app.

### Crashlog

Adapter: Already cancelled scan.
2024-03-25 16:31:07.285 +08:00 [INF] connect needs 0.0044007 secs
2024-03-25 16:31:07.305 +08:00 [INF] XXX connected
2024-03-25 16:31:07.307 +08:00 [INF] Characteristic.StartUpdates
2024-03-25 16:31:08.806 +08:00 [INF] OnDescriptorWrite: 00-00
2024-03-25 16:31:08.808 +08:00 [INF] References of parent device and gatt callback device equal? TRUE
2024-03-25 16:31:08.808 +08:00 [INF] OnConnectionStateChange: GattStatus: InsufficientAuthorization
2024-03-25 16:31:08.808 +08:00 [INF] References of parent device gatt and callback gatt equal? TRUE
2024-03-25 16:31:08.811 +08:00 [ERR] disconnect dongle error
System.Exception: GattStatus: 133
   at Plugin.BLE.Abstractions.Utils.TaskBuilder.<SafeEnqueueAndExecute>d__11`1[[System.Boolean, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
   at Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__9`3[[System.Boolean, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler`1[[Plugin.BLE.Android.CallbackEventArgs.DescriptorCallbackEventArgs, Plugin.BLE, Version=3.1.0.0, Culture=neutral, PublicKeyToken=null]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
   at Plugin.BLE.Android.Characteristic.StopUpdatesNativeAsync(CancellationToken cancellationToken)
2024-03-25 16:31:08.813 +08:00 [INF] Disconnected 'XXX' by lost connection
2024-03-25 16:31:08.813 +08:00 [INF] DisconnectedPeripheral by lost signal: XXX
2024-03-25 16:31:08.813 +08:00 [INF] Removed device from discovered devices list: XXX
2024-03-25 16:31:08.831 +08:00 [INF] [Warning]: Can't disconnect XXX. Gatt is null.
2024-03-25 16:33:05.203 +08:00 [INF] Adapter: Already cancelled scan.

## Configuration

**Version of the Plugin:** e.g. 3.0.0/3.1.0-beta

**Platform:** e.g. Android 11, 12 and 13

**Device:** e.g. Samsung S21, Realme 9i, Redmi 9T

ygl-rg, 2024-03-26:
I noticed when "[Warning]: Can't disconnect XXX. Gatt is null." is thrown,  the BLE won't work anymore.

Do you have an idea what cause the gatt is null?

axa88, 2024-04-08:
> 2024-03-25 16:31:08.808 +08:00 [INF] OnConnectionStateChange: GattStatus: InsufficientAuthorization

what happens if you provide sufficient credentials?

ygl-rg, 2024-04-11:
> > 2024-03-25 16:31:08.808 +08:00 [INF] OnConnectionStateChange: GattStatus: InsufficientAuthorization
> 
> what happens if you provide sufficient credentials?

I have no idea. It looks like this issue is related to some old samsung phones.

axa88, 2024-04-12:
> I have no idea. It looks like this issue is related to some old samsung phones.

How so?
If so how does that make it a library issue?

It appears to me that as soon as you attempt to Write Characteristic/Descriptor, it is telling you this/these have security enabled and require proper credentials set.  To which you have not established resulting in a the disconnect as reported in the error.

Not sure this is an issue whatsoever

closed 2024-05-27 by ygl-rg