dotnet-bluetooth-le/dotnet-bluetooth-le #832: Manufacturer data packet is missing from device on iOS

ckrutsinger, 2024-03-29:
## Steps to reproduce

1. Using an iOS device, scan a BLE device that advertises with manufacturer specific data packet

2. Observe that the manufacturer data packet is missing from the advertisement records of the device

## Expected behavior
As you can see from this screen capture on an Android debugging session, the device is advertising 5 records:
[image]

## Actual behavior
As you can see from this screen capture on an iOS debugging session, the same device is only showing 3 of the 5 records:
[image]

### Crashlog
N/A

## Configuration

**Version of the Plugin:** 3.0.0

**Platform:** e.g. iOS 17.4

**Device:** e.g. iPhone XR

ckrutsinger, 2024-03-29:
This post on Stack Overflow [appears](https://stackoverflow.com/a/24199063/1256015) to discuss how to get manufacturer data to pass through.

janusw, 2024-04-16:
> RESOLVED: This issue is that iOS passes the advertised device twice. The first time does not contain the manufacturer specific data advertisement, but the second time it does. So the key to being able to access the manufacture specific data advertisement is to listen for DeviceAdvertised rather than DeviceDiscovered, since DeviceDiscovered will only send along the first device announcement and will return without announcing the second, more complete device information.

Yes, no matter if you are dealing with extended advertisments or not: If you want to obtain all advertisement data, and not just 'discover' it, you need to look at all advertisements, not just the first one. A device can choose to send several differing advertisements after each other or change the advertised data over time, etc.

closed 2024-04-16 by janusw