dotnet-bluetooth-le/dotnet-bluetooth-le #838: limit of bluetooth device can be connected simultaneous

Sujiyama, 2024-04-22:
Hello Team, i tried to connect heart rate monitor with your library that display the bpm in android tablet or UWP

When Device Discovered is up to seven devices in android then the eighth can't connect.
In UWP i had the same problem but the problem is up to eight device and can't connect to nineth

I made a reasearch in internet how can connect up to 15 devices ,but everyone said that reasons make that problem.

1) The library of bluetooth has limit.
2) The Bluetooth adapter has limit.

First of all i want to know if we have some limit in your library that is my first question.
My second question is how can i search for a bluetooth adapter that can connect up to 15 devices. Which is the search condition that i must search for it. 
Can you suggest me some adapters that can connect up to 15 devices simultaneous and is compatible with your library

Thank you

Sujiyama, 2024-04-22:
Hello AskBojesen,

Thanks for your response.

AskBojesen, 2024-04-22:
Great then! I suggest to close this issue!?

axa88, 2024-04-23:
Android connection limit is even more confusing and even depends on Manufacturer

Sujiyama, 2024-04-23:
Thank you all for your responses, i try to understand how other developers can connect more than 7 devices. 
Some of them present that can connect up to 10 and more with bluetooth. 

I think the limit is not coming from OS but from adapter
If you research for Chileaf BLE Data Receiving Portable Long Range Wireless USB Stick Dongle they said that you can connect up to 60 members’movement data can be collected via Bluetooth.

I don't want to send you link because i don't want to to promote this bluetooth adapter. Only i try to find a solution of my problem. 
If you allow me i will send you some links.

Also many companies develop softwares that work with bluetooth heart rate monitors and receive data with more than 7 devices. 
I try to find a solution because your library is so great and i don't want to work with ANT+ solution. 
Our project is for customer that have GYM and we develop a software that show the heart rate of BLE devices while training in class. 
The maximum people in class will be 10-14 people, with that limit we discovered, we stucked and we don't know how to find a solution.

That is the reason that i open ticket to you. 
Do you know some trick with your library that can we connect 4 more and avoid the limit?

Thank you

janusw, 2024-05-25:
I think this is more a question regarding OS or adapter limits, rather than a bug report. I'm transferring it from an 'issue' into a 'discussion'.