dotnet-bluetooth-le/dotnet-bluetooth-le #841: Wiindows Adapter.DeviceConnected event is triggered n+1 times each time the device is subsequently connected and disconnected

axa88, 2024-05-03:
Ive come to realize on Windows, with each subsequent connection and disconnection to a single Device, an additional trigger of the Adapter.DeviceConnected event will occur.
Doesnt occur under Android, no other platforms tested.

Meaning on the first connection, the event is triggered as expected.
Disconnect and Connect again, upon connection the event will trigger twice...
Disconnect and Connect again, upon connection the event will trigger three times....
So on and so fourth

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

Any programming should replicate this in windows

## Steps to reproduce

1. Set breakpoint on Adapter.DeviceConnected event

2. Connect, Disconnect, Connect again

3. Observe multiple events triggered.
Repeat as many times as you would like

## Expected behavior
On each subsequent Connection, an additional trigger of the event occurs

## Actual behavior
On each subsequent Connection, a one one trigger of the event should occur

### Crashlog

no crash

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
3.0.0

**Platform:**
Windows 10.0.19041

**Device:** 
Device name	Portable
Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz   1.80 GHz
Installed RAM	8.00 GB (7.86 GB usable)
Device ID	2F7EBCF5-2024-4778-9E85-988D5FD862D6
Product ID	00330-51185-11714-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

axa88, 2024-05-04:
Attempted to mitigate this issue by unsubscribing to the `Adapter.DeviceConnected` `event` within the event handler, and then resubscribing during the `Adapter.DeviceDisconnected` event handler.

But it seems not possible to resubscribe to the event a second time... or at least cannot do it during the `DeviceDisconnected` `event`.

Not sure where to go with this

janusw, 2024-05-04:
So you are seeing this with version 3.0.0? Does 3.1.0-rc.1 behave the same way?

janusw, 2024-05-04:
It sounds like the event is not properly unsubscribed when disconnecting.

axa88, 2024-05-04:
> So you are seeing this with version 3.0.0? Does 3.1.0-rc.1 behave the same way?

figured it would be asked... ill have to try asap

axa88, 2024-05-04:
> It sounds like the event is not properly unsubscribed when disconnecting.

Ya this issue seems to be multifaceted...
The n+1 as well as the failure to unsub.

closed 2024-05-07 by axa88