dotnet-bluetooth-le/dotnet-bluetooth-le #846: ISSUE: DeviceAdvertised event on Android

Auto72, 2024-05-06:
On Android if we setup for the DeviceAdvertised event, while the scan is active, it is generated on the MainThread and the Advertisements that arrive are countless and very frequent, more then 100 per second. The GUI stops responding, i.e. if you press the Stop Scan button, it responds after about 10 seconds. I also tried to launch a task every time this event is generated, but the result does NOT change, i.e. the GUI continues to be NOT responsive. When it eventually responds (like after 10 seconds I pressed the button) and stops the scan, this event is still generated a bunch more times (like for another 30 seconds). 
At the moment the solution (for me) for ANDROID is NOT listen for this event.

On iOS this event is generated on the MainThread (like for Android), but the Advertisements that arrive are a thousand times lower than those that arrive on Android and the GUI responds without problems.

On Windows this event is NOT raised on the MainThread.

How can we fix it?
Can we try to generate that event not in the MainThread for Android?

axa88, 2024-05-06:
I wonder if the `DeviceAdvertised` `event` is effected by the `Adapter.ScanMatchMode`. I wouldn't think so but you might try setting `Adapter.ScanMatchMode = STICKY` and see if that avoids your device from processing duplicate packets from the same device as it does during a Scan.

Otherwise you might try setting Adapter.ScanMode = Passive, but i really doubt that is going to have any impact on the `DeviceAdvertised` `event` either.

That said i don't believe `DeviceAdvertised` is called back on the UI thread, so you might look at other ways you might be tying up the UI thread unwittingly, but honestly I'm not confident in any of the advice im giving you here will help so...

Auto72, 2024-05-06:
Thanks for your tips, I will try them and report here the result.

NOTE: On Android and iOS MainThread.IsMainThread = true inside the event DeviceAdvertised, but not on Windows.
I tried this.

```
private void Adapter_DeviceAdvertised(object? sender, Plugin.BLE.Abstractions.EventArgs.DeviceEventArgs e)
{
    Debub.WriteLine($"IsMainThread: {MainThread.IsMainThread}");    // returns True on Android and iOS, but not on Windows.
}
```

Auto72, 2024-05-06:
I used to set Adapter.ScanMode = ScanMode.LowLatency in Xamarin with no problem.
In .NET MAUI I have the issue reported above on Android.
Adapter.ScanMatchMode is and has been always STICKY.
I solved the problem setting Adapter.ScanMode = ScanMode.Balanced.
For sure setting it Passive would be even better to mitigate the issue.

Thanks @axa88

axa88, 2024-05-07:
> e is and has been always STICKY.
> Using Adapter.ScanMode = ScanMode.Balanced is better, but sometimes the behaviour is similar to ScanMode.LowLatency
> With Adapter.ScanMode = Passive I don't receive any scan device.

Unsure. im able to get good response both on windows and Android
Perhaps post a gist and someone might be able to comment...

janusw, 2024-05-07:
How to handle this situation probably depends a bit on your kind of setup and what you're trying to achieve.

First question: Are all of these advertisements even relevant for you? If not, you should probably use a scan filter to reduce the number of advertisements you have to process.

Then, of course, you should do as little work as possible in response to an advertisement (but you're probably already doing that, I hope).

And finally you can try to do the processing in the background, e.g. via a `ThreadPool` (which re-uses the threads in the pool and avoids the overhead of creating a new thread for each task).

AFAICS the library does nothing specific to put these events on the main thread, but just passes them on as they come from the OS.

Auto72, 2024-05-08:
> How to handle this situation probably depends a bit on your kind of setup and what you're trying to achieve.
>
> First question: Are all of these advertisements even relevant for you? If not, you [...]

First answer: 
No, I'm interested only in my devices, that as the name that start with a prefix.
How can I setup a scan filter ?

I use the DeviceAdvertised event for two reasons:
a) I get the device before the event DeviceDiscovered is raised, so I can show it soon.
b) I get the value of RSSI updated.

Sure, I tried with Task.Run() but it doesn't work.
I think the library put all the advertisements are put in a queue and raise the event until the queue is empty, even if the scan has been stopped.

Auto72, 2024-05-08:
I reviewed the my code and now I used Task.Run() also in the DeviceDiscovered event and other adjustments.
I set Adapter.ScanMode = Balanced for Android.
Now seems fine.
Thanks.

closed 2024-05-08 by Auto72