dotnet-bluetooth-le/dotnet-bluetooth-le #847: Need to be able to register for notification from characteristics that is only INDICATE

CodeMusher, 2024-05-06:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Try to use a peripheral with a characteristic with only INDICATE

2.

3.

## Expected behavior
Registration for updates should succeed

## Actual behavior
Registration is not performed. The Windows implementation of the StartUpdatesNativeAsync only register in case of Notify.
It could be solved this way
            GattWriteResult result;
            if (NativeCharacteristic.CharacteristicProperties.HasFlag(GattCharacteristicProperties.Indicate))
                result = await NativeCharacteristic.WriteClientCharacteristicConfigurationDescriptorWithResultAsync(GattClientCharacteristicConfigurationDescriptorValue.Indicate);
            else
                result = await NativeCharacteristic.WriteClientCharacteristicConfigurationDescriptorWithResultAsync(GattClientCharacteristicConfigurationDescriptorValue.Notify);

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3
3.1.0-rc.1

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
UWP/Windows 10

**Device:** e.g. HTC Sensation /i Phone 7 ...

janusw, 2024-05-19:
> ## Actual behavior
>
> Registration is not performed. The Windows implementation of the StartUpdatesNativeAsync only register in case of Notify. It could be solved this way GattWriteResult result; if [...]

This fix is supposed to go into `Characteristic.StartUpdatesNativeAsync`, right? It currently looks like this:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/1e19fa858cded5eeb3c0ddd54a1aee05db14d4c1/Source/Plugin.BLE/Windows/Characteristic.cs#L57-L64

@CodeMusher  Can you open a PR for it?

AskBojesen, 2024-05-19:
I do not understand the issue. In our project  we have a device with a lot of characteristics with only INDICATE.
and even one where the device decides if the characteristic should be send as NOTIFY or INDICATE.
And that works for iOS, Android and Windows

Remember: 

- NOTIFY: Unacknowledged
- INDICATE: Acknowledged

CodeMusher, 2024-05-19:
@janusw I can try. What is the correct procedure?

AskBojesen, 2024-05-19:
But It could be related to how the device handle the requests.

I see the same in the current implementation for Windows, which is also what @janusw points out.
@CodeMusher : Should I make a PR you can test with the solution @janusw  suggests? 

  var result = await NativeCharacteristic.WriteClientCharacteristicConfigurationDescriptorWithResultAsync(GattClientCharacteristicConfigurationDescriptorValue.Notify);

CodeMusher, 2024-05-19:
@AskBojesen Do you refer to my suggestion in the original issue? 
That works for me but my peripheral has a characteristic with only INDICATE. I'm not sure what happens in case of both INDICATE and NOTIFY. Is one registration enough or do you need to register for both INDICATE and NOTIFY? I feel my knowledge ending here.

AskBojesen, 2024-05-19:
I actually see now, that our device is _lying_ in the characteristic properties - reporting NOTIFY, but it is indicate.

CodeMusher, 2024-05-19:
Ok, so then you can understand my issue?

AskBojesen, 2024-05-19:
> Ok, so then you can understand my issue?

Yes I can imagine there could be an issue - I cannot reproduce.
I have created this PR: #858

PR #858 (2024-05-20): Fixed StartUpdatesNativeAsync for indicate
To fix #847
closed 2024-05-20 by janusw