dotnet-bluetooth-le/dotnet-bluetooth-le #850: WinUI Call to `DisconnectDeviceAsync` may not complete.

axa88, 2024-05-09:
Call to `DisconnectDeviceAsync` may never complete.
The call doesn't take a `CancelationToken`, perhaps it should.

It might take several Connect/Disconnect cycles, but eventually `DisconnectDeviceAsync` will hang.

Hadn't noticed this behavior in v3.0.0

## Steps to reproduce

1. Connect/Disconnect

2. Observe eventually a call to `DisconnectDeviceAsync` will not complete

3.

## Expected behavior
the call completes

## Actual behavior
it doesn't complete

### Crashlog
none

## Configuration

**Version of the Plugin:** 
3.1 rc1

**Platform:** 
[image]

**Device:** 
```
Processor	Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz   1.80 GHz
Device ID	2F7EBCF5-2024-4778-9E85-988D5FD862D6
System type	64-bit operating system, x64-based processor

Intel Dual Band Wireless-AC 8265, Wi-Fi 2x2 802.11ac + **Bluetooth 4.1**, M.2 card
```

axa88, 2024-05-09:
Hmm I see this issue has some history perhaps #788

AskBojesen, 2024-05-13:
> Hmm I see this issue has some history perhaps #788

Yes I agree #788 is the same... It is quite annoying Windows API doesn't have a disconnect method.
In my desktop app I actually turn on / off the Bluotooth - and that works as a workaround. Turning Bluetooth off disconnects :-)

I can give it yet another shot before next release.

axa88, 2024-05-13:
@AskBojesen 
I can imagine how a lack of Disconnect functionality could be problematic here... that seems fundamentally flawed.

During [initial observations](#842) of this issue, I note a catastrophic symptom in that after `DisconnectDeviceAsync` failure, `IDevice.NativeDevice` is then null and thus unusable.
 My current workaround is basically appending `DisconnectDeviceAsync` with `WaitAsync` which takes a `CancellationToken`.  Then ignoring any `IDevice` that is found having `NativeDevice = null`, but this can be brittle.

Hopefully you find a proper solution.
Perhaps that includes overloading `DisconnectDeviceAsync` to take a `CancellationToken` and return `Task<bool>` : _success or not_

MarkKananen, 2024-06-27:
Has anyone found a any work arounds for this?  So far the only reliable way to clear it up is to close and restart the app.  Turning of/on bluetooth will disconnect the device, but you cannot reconnect to it as something is cached.

axa88, 2024-06-27:
> Has anyone found a any work arounds for this? So far the only reliable way to clear it up is to close and restart the app. Turning of/on bluetooth will disconnect the device, but you cannot reconnect to it as something is cached.

There's this:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/864#issuecomment-2159804792

I haven't tried it as ive basically given up on the WinUI implementation for now as it has become unusable

sfrase59, 2024-10-10:
I have been working on a MAUI BLE app for a while.  I have also found that the code I had to put in due to switching pages in an Android app destroys the BLE  objects, does not work with Windows because the DisconnectDeviceAsync does not work.
Looking for a work-around.

axa88, 2024-10-11:
There is no work around. 
The windows implementation has been wrecked. 
You might try never disconnecting...

That said i promise you switching pages does/should not destroy your BLE objects, you're doing something wrong.

sfrase59, 2024-10-11:
Thank you for the reply.

axa88, 2024-11-02:
fwiw the API `GetBondState()` might be better served by not referencing the `NativeDevice` in the `catch` block as its consistently `null` when Bonding on Windows blows up

[image]