dotnet-bluetooth-le/dotnet-bluetooth-le #852: Windows: Check OS version runtime

AskBojesen, 2024-05-14:
Windows: Check OS version number.
Currently a runtime on the Windows client needs > 22000
Eg. to call this: https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.bluetoothlepreferredconnectionparameters
Unfortunately Windows allow executing code linked to > 22000 - only at runtime calling new functionality causes error.

1.  Make special ILab release without checking > 22000 (in order to allow installing)
2. Install on a Windows 10
3. Connect to an instrument

Error: Cannot connect (error)
Expected: Connected without errors

## Expected behavior
Tell us what should happen

## Actual behavior
Cannot connect

Version = 3.1.0-rc1

janusw, 2024-05-15:
I think this is a regression, which has been introduced in 3.1.0-beta.2 via #801. It should be fixed before we release the final version 3.1.0.

janusw, 2024-05-15:
See also the discussion in #849 ...

AskBojesen, 2024-05-16:
Ok, I will make a draft PR today to fix this

AskBojesen, 2024-05-16:
PR: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/853

closed 2024-05-18 by janusw