dotnet-bluetooth-le/dotnet-bluetooth-le #861: It doesn't work with android 14

yj320696, 2024-05-23:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. call "StartScanningForDevicesAsync" to scan BLE device

2.

3.

## Expected behavior
It should list all the devices discovered.
And it worked on old versions before Android 14.

## Actual behavior
Nothing was found
It only happens on Android 14

### Crashlog

## Configuration

**Version of the Plugin:** 2.1.3

**Platform:** Android 14

**Device:** Samsung Tablet

axa88, 2024-05-23:
There appears 2 changes to bluetooth in version 14
Not saying this is your issue, i wouldnt know as i dont currently have access to a 14 device, but this is what changed:

https://developer.android.com/about/versions/14/behavior-changes-all#mtu-set-to-517

janusw, 2024-05-25:
Scanning on Android 14 works well for me, and AFAIK there are no relevant changes between Android 13 and 14 in that area.

A few questions:
1) Are you using the latest version 3.1.0 of the library?
2) Which older Android versions did you try? (There were some changes regarding permissions in Android 12 that you should be aware of: https://developer.android.com/develop/connectivity/bluetooth/bt-permissions#declare-android12-or-higher)
3) Could you please try our sample client BLE.Client.Maui and let us know if that works on Android 14 with your device?

yj320696, 2024-05-27:
Hi @janusw, thank you for the reply.
I have updated the library to v3.1.0.
The older version should be Android 13 or maybe Android 12. I don't have chance to check. We have installed 6 devices from Android 10 to Android 13(or 12). I never check the version number
I will use your sample client to test and let you know the result.

yj320696, 2024-05-31:
I did factory reset and no issues anymore

closed 2024-05-31 by yj320696

BenWoodford, 2025-01-09:
Sorry to bump this, but is there a way to get 2.x working with Android 14? I can't upgrade to 3.0 as I get NotImplementedException errors (presumably because our Xamarin is too old?)

janusw, 2025-01-09:
> Sorry to bump this, but is there a way to get 2.x working with Android 14?

What exactly is not working for you? Is it working with Android 13?

> I can't upgrade to 3.0 as I get NotImplementedException errors (presumably because our Xamarin is too old?)

Yes, your Xamarin is too old. It's called MAUI now ;)

BenWoodford, 2025-01-11:
> > Sorry to bump this, but is there a way to get 2.x working with Android 14?
> 
> What exactly is not working for you? Is it working with Android 13?

Android 14 device reads every Bluetooth device as disconnected even when the OS and device show connected. It _could_ be the device though so I’m just trying to rule out causes right now. I can get a list of pairings, but the state is already disconnected…

> > I can't upgrade to 3.0 as I get NotImplementedException errors (presumably because our Xamarin is too old?)
> 
> Yes, your Xamarin is too old. It's called MAUI now ;)

I’m aware, but upgrading a massive project like that isn’t a small task unfortunately. The docs said Xamarin support is still maintained though…?

DeanZhuo, 2025-02-14:
I'm using android 14 and still having this issue. why is it closed?

axa88, 2025-02-15:
> I'm using android 14 and still having this issue. why is it closed?

If you read OPs [response ](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/861#issuecomment-2143117758), they found it an invalid issue.
 if you have the same or similar problem, it would really help to file a new issue with as much pertinent information as you can gather.