dotnet-bluetooth-le/dotnet-bluetooth-le #864: Windows - Run Discover Scan twice cause Disconnection fail

andrenoel84, 2024-05-31:
On Windows, When I start scan twice and try connect/disconnect, The disconnection event is never triggered and the device stays connected to the system

## Steps to reproduce

Running BLE.Client.WinConsole Application for tests

1. Enter D1 and select device

2. Enter D1 Again

3. Enter D3 Connect-Disconnect

## Expected behavior
We should receive disconnection notification.

If you try to connect/disconnect without scan it works. 
If you run scan and connect/disconnect after it works.
BUT, if you run scan, wait result, run scan again, wait result and connect/disconnect, we never receive disconnection event

## Actual behavior
The connection is made but we never receive the disconnection notification. The device stays connected to the system

### Crashlog

No crash

## Configuration

**Version of the Plugin:** last source code

**Platform:** Win11Pro / 23H2 / 22631.3672

axa88, 2024-06-01:
The Windows implementation unfortunately doesn't currently work well. [maybe related](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/849#issuecomment-2111009167)

andrenoel84, 2024-06-04:
Is there anyone able to reproduce this bug ?
Just to know if i'm the only one :)
I spent a lot of time try to figure where the problem is with no result :/

axa88, 2024-06-05:
> Is there anyone able to reproduce this bug ? Just to know if i'm the only one :) I spent a lot of time try to figure where the problem is with no result :/

Had you a look at the link of issues provided?
Did you notice issue #850?
Might it be plausible it or a similar issue is affecting you?
In my estimation the windows implementation isn't currently usable as disconnection is important for my purposes. 

In the case that there might be something else to uncover, might i suggest clarifying what you mean by _Enter D1_, _Enter D3_, surely no one will be able to reproduce your workflow not understanding what you are doing... or am i out of the loop with something?

andrenoel84, 2024-06-07:
Yes I checked all issues on disconnect part but nothing seems to match directly to my problem.

To be more clear, I add a test into PluginDemo.cs with the following code:
```  
public async Task DisconnectAfterScanTest()
  {
      //Hard set MAC address for tests ...
      //string bleaddress = BleAddressSelector.GetBleAddress();
      string bleaddress = "FF0BCAC1DA4B";
      var id = bleaddress.ToBleDeviceGuid();
      var connectParameters = new ConnectParameters(connectionParameterSet: ConnectionParameterSet.ThroughputOptimized);

      //Run scan once ...
      Console.WriteLine("Scan start");
      await DoTheScanningTimeout();
      await Task.Delay(200);
      Console.WriteLine("Scan end");

      IDevice dev = await Adapter.ConnectToKnownDeviceAsync(id, connectParameters);
      Write("Waiting 2 secs");
      await Task.Delay(2000);
      Write("Disconnecting");
      await Adapter.DisconnectDeviceAsync(dev);
      dev.Dispose();
      Write("Test_Connect_Disconnect done");

      //Run scan twice ...
      Console.WriteLine("Scan (1) start");
      await DoTheScanningTimeout();
      await Task.Delay(200);
      Console.WriteLine("Scan (1) end");

      Console.WriteLine("Scan (2) start");
      await DoTheScanningTimeout();
      await Task.Delay(200);
      Console.WriteLine("Scan (2) end");

      dev = await Adapter.ConnectToKnownDeviceAsync(id, connectParameters);
      Write("Waiting 2 secs");
      await Task.Delay(2000);
      Write("Disconnecting");
      await Adapter.DisconnectDeviceAsync(dev);
      dev.Dispose();
      Write("Test_Connect_Disconnect done");
```

Running it I have the following output : 
```
Environment.OSVersion.Version.Build: 22631

Using BLE Address: FF0BCAC1DA4B
List of tests to run for key:

Escape: Quit!
B: Turn Bluetooth ON
N: Turn Bluetooth OFF
D0: Discover only
D1: Discover and set the BleAddress
D2: Set the BleAddress
D3: Connect -> Disconnect
D4: Pair -> Connect -> Disconnect
D5: Connect -> Change Parameters -> Disconnect
D6: Run GetSystemConnectedOrPairedDevices
D7: Loop: Connect -> Read services -> Disconnect
D8: Loop: Connect -> Read services -> Dispose
D9: Connect -> Loop: ConnectionLost -> Connect
Q: Adapter.BondAsync
W: Adapter.BondedDevices
E: Device.BondState
A: Pure Windows: Connect -> Disconnect
[... 358 lines cut ...]
07:21:40.577 Plugin.BLE - AdvReceived - Old: FF0BCAC1DA4B, DoKey_0437, Public, -63, ScanResponse, Connectable, Scannable, ScanResponse
07:21:40.580 Plugin.BLE - AdvReceived - Old: 6D2A8E57407C, Bluetooth 6d:2a:8e:57:40:7c, Random, -74, ConnectableUndirected, Connectable, Scannable
07:21:40.918 Plugin.BLE - AdvReceived - Old: FF0BCAC1DA4B, DoKey_0437, Public, -63, ConnectableUndirected, Connectable,Scannable
07:21:40.920 Plugin.BLE - Adapter: Scan timeout has elapsed.
07:21:40.920 Plugin.BLE - Adapter: Stopping the scan for devices.
07:21:40.920 Plugin.BLE - Stopping the scan for devices
07:21:40.929 Plugin.BLE - AdvReceived - Old: FF0BCAC1DA4B, DoKey_0437, Public, -63, ScanResponse, Connectable, Scannable, ScanResponse
Scan (2) end
07:21:41.124 Plugin.BLE - ConnectToDeviceNativeAsync FF0BCAC1DA4B Named: DoKey_0437 Connected: Disconnected
07:21:41.125 Plugin.BLE - RequestPreferredConnectionParameters(ThroughputOptimized): Success
07:21:41.200 Plugin.BLE - Device_ConnectionStatusChanged FF0BCAC1DA4B DoKey_0437 Connected
07:21:41.200 Plugin.BLE - Connected with Latency = 0, Interval = 12, Timeout = 200
07:21:41.200       DEMO - Adapter_DeviceConnected FF0BCAC1DA4B with name:  DoKey_0437
07:21:41.200 Plugin.BLE - ConnectToDeviceAsync Connected: 00000000-0000-0000-0000-ff0bcac1da4b DoKey_0437
07:21:41.200       DEMO - Waiting 2 secs
07:21:41.210 Plugin.BLE - GattSession_SessionStatusChanged: Active
07:21:41.428 Plugin.BLE - GattSession_MaxPduSizeChanged: 350
07:21:43.203       DEMO - Disconnecting
07:21:43.203 Plugin.BLE - DisconnectDeviceNative from device with ID:  FF0BCAC1DA4B
07:21:43.204 Plugin.BLE - DisconnectInternal()-> DisposeGattSession DONE
07:21:43.204 Plugin.BLE - DisconnectInternal()-> ClearServices DONE
07:21:43.205 Plugin.BLE - DisconnectInternal()-> DisposeNativeDevice DONE
```

In it you can see that if I run two scan before connect/disconnect, I never receive again the disconnection notification and the ble device stays connected to the system until I kill the App...

axa88, 2024-06-07:
Firstly understand that you are using an implementation on top of the underlying API, so frankly while it might not "directly match" it is difficult to disseminate how the underlying issue will manifest in any particular implementation.

Basically what i am saying is I dont think much that can be learned or addressed from the report that an implementation (demo in this case) is malfunctioning , if the underlying API is broken.

andrenoel84, 2024-06-07:
I'm not sure to understand well.
I use demo app only in order to simplify my tests.

It simply use basic functions like StartScanningForDevicesAsync, ConnectToKnownDeviceAsync, DisconnectDeviceAsync...
I'm not telling that the problem comes from demo app but in the underlying API and the fact that problem appears only if I run 2 consecutives scans.

axa88, 2024-06-10:
> I'm not telling that the problem comes from demo app but in the underlying API and the fact that problem appears only if I run 2 consecutives scans.

What I find is that you don't need to run any more than once scan to have the Connect/Disconnect stop working.  Though I think i can concede additional scans (ie additional interaction with the API) does help expedite the failure

andrenoel84, 2024-06-11:
On my side the problem appears when there are 2 consecutive scans. 
After a lots of research i found something.. 
In AdapterBase.cs file, it seems that all devices added to DiscoveredDevicesRegistry are not disposed but the array is simply cleared. I made one modification to fix this and it works

```
      public async Task StartScanningForDevicesAsync(ScanFilterOptions scanFilterOptions,
          Func<IDevice, bool> deviceFilter = null,
          bool allowDuplicatesKey = false,
          CancellationToken cancellationToken = default)
      {
          if (IsScanning)
          {
              Trace.Message("Adapter: Already scanning!");
              return;
          }

          IsScanning = true;
          _currentScanDeviceFilter = deviceFilter ?? (d => true);
          _scanCancellationTokenSource = new CancellationTokenSource();

          try
          {
              // Add Dispose for all devices 
              foreach (var discoveredDeviceEntry in DiscoveredDevicesRegistry)
              {
                  ((IDevice)(discoveredDeviceEntry.Value)).Dispose();

              }
              DiscoveredDevicesRegistry.Clear();

              using (cancellationToken.Register(() => _scanCancellationTokenSource?.Cancel()))
              {
                  await StartScanningForDevicesNativeAsync(scanFilterOptions, allowDuplicatesKey, _scanCancellationTokenSource.Token);
                  await Task.Delay(ScanTimeout, _scanCancellationTokenSource.Token);
                  Trace.Message("Adapter: Scan timeout has elapsed.");
                  CleanupScan();
                  ScanTimeoutElapsed?.Invoke(this, new System.EventArgs());
              }
          }
          catch (TaskCanceledException)
          {
              CleanupScan();
              Trace.Message("Adapter: Scan was cancelled.");
          }
      }
```

M0n7y5, 2024-09-25:
@andrenoel84 you should make PR for this

OomJan, 2024-10-21:
The workaround described by [andrenoel84](https://github.com/andrenoel84) did not work for me. I ended up implementing a fail safe on the disconnect that toggles the radio off and back on if the disconnect takes too long. It's a pretty bad solution and I blame Microsoft for creating this mess.

axa88, 2024-10-22:
> I blame Microsoft for creating this mess.

How so? I've other BT devices that interact well with windows, the problem here _seems_ to unfortunately be the API