dotnet-bluetooth-le/dotnet-bluetooth-le #867: Feature Request: Bonding/Pairing on ios and Android

electro-logic, 2024-06-19:
Would be great to have a way to pair a device from the code without manually doing it from the system settings page.

axa88, 2024-06-21:
public abstract Task BondAsync(IDevice device);

electro-logic, 2024-06-21:
Hello, BondAsync is not working on Android

axa88, 2024-06-21:
> Would be great to have a way to pair a device from the code without manually doing it from the system settings page.

are you using the latest version?

janusw, 2024-08-03:
> Hello, BondAsync is not working on Android

It should. What are you trying to do? And what happens when you do it? Do you get any error messages or exceptions?

electro-logic, 2024-08-03:
I tried again today

```
await Permissions.RequestAsync<Permissions.Bluetooth>();
var adapter = CrossBluetoothLE.Current.Adapter;
adapter.DeviceDiscovered += Adapter_DeviceDiscovered;
await adapter.StartScanningForDevicesAsync();
// ...
await adapter.BondAsync(e.Device);
await adapter.ConnectToDeviceAsync(e.Device);
```

and seems working fine. Thank you anyway.

closed 2024-08-03 by electro-logic