dotnet-bluetooth-le/dotnet-bluetooth-le #868: ios Bluetooth Permission

electro-logic, 2024-06-19:
Is there a way to know if user granted permission with MAUI/Plugin.Ble?

Thank you

axa88, 2024-06-21:
> Is there a way to know if user granted permission with MAUI/Plugin.Ble?
> 
> Thank you

yes

electro-logic, 2024-06-21:
Hi @axa88 do you have any example working on iOS? BLE State property is unknown on iOS

axa88, 2024-06-21:
you need to use google better
https://learn.microsoft.com/en-us/dotnet/maui/platform-integration/appmodel/permissions?view=net-maui-8.0&tabs=android

electro-logic, 2024-06-21:
I have seen that page, on iOS Permissions.CheckStatusAsync always return Granted and so I can't use it.
This library is using iOS native code so should be able to avoid this MAUI limitations someway.

axa88, 2024-06-21:
> I have seen that page, on iOS Permissions.CheckStatusAsync always return Granted and so I can't use it. This library is using iOS native code so should be able to avoid this MAUI limitations someway.

you should add pertinent information to your report, such as what you have tried, what was the result, what version you are using, sample code, etc. then perhaps you will get adequate help

janusw, 2024-08-03:
> I have seen that page, on iOS Permissions.CheckStatusAsync always return Granted and so I can't use it.

If `Microsoft.Maui.ApplicationModel.Permissions` does not work for you, please open an issue at https://github.com/dotnet/maui ....

electro-logic, 2024-08-03:
> > I have seen that page, on iOS Permissions.CheckStatusAsync always return Granted and so I can't use it.
> 
> If `Microsoft.Maui.ApplicationModel.Permissions` does not work for you, please open an issue at https://github.com/dotnet/maui 

Thanks @janusw for the reply, would be great to add this behavior in the documentation