dotnet-bluetooth-le/dotnet-bluetooth-le #869: User Description Empty / Characteristic Name Unknown

cdbenjamin, 2024-06-20:
## Overview

Howdy! I am relatively new to BLE development. I'm currently trying to do some basic wireless communication using a P-Nucleo-WB55 board along with Plugin.BLE and .NET MAUI. I have successfully created a simple service and characteristic on the Nucleo board. I have been able to connect to the device using STBLE Toolbox on my phone as well as using Plugin.BLE on my desktop. The user description for the characteristic shows up on STBLE Toolbox, but not on Plugin.BLE. I need the name of the characteristic for my application. I have looked for people having similar issues, but I have been unable to find any. Any guidance would be greatly appreciated. I've included some screenshots of my code and the results at the bottom of this post.

## Steps to reproduce

1. Add user description to characteristic in peripheral firmware

2. Connect to peripheral using STBLE Toolbox and view user description of characteristic

3. Connect to peripheral using Plugin.BLE on Windows and check user description of characteristic

## Expected behavior
I expect there to be a value for the characteristic user description containing the byte array for the string I specified in the firmware code.

## Actual behavior
The characteristic user description is empty.

## Configuration

**Version of the Plugin:** 3.1.0

**Platform:** Windows 10 Enterprise (.NET MAUI using .Net 7)

**Device:** Dell desktop with ASUS USB-BT400 USB 2.0 Bluetooth 4.0 Adapter

Sample of firmware code:
[image]

Connected to STBLE Toolbox:
[image]

Trying to read user description using Plugin.BLE:
[image]

Result Plugin.BLE:
[image]

axa88, 2024-06-21:
Am i missing something? Are you only looking at the descriptors, from the first characteristic, in the last service which was received?
Doesn't this "firmware" you're using add a Generic Access and Generic Attribute or other default services for you?  i would imagine most hobby platforms do.  Therefore how can you know this is the service, let alone the characteristic, you are looking to interrogate.  A bit presumptuous to be filing a bug?

cdbenjamin, 2024-06-21:
> Doesn't this "firmware" you're using add a Generic Access and Generic Attribute or other default services for you?  i would imagine most hobby platforms do.  Therefore how can you know this is the characteristic, let alone the service, you are looking to interrogate. 

Yes, it does add the Generic Access and Generic Attribute default services, but as far as I can tell, this is definitely the correct characteristic. When I subscribe to the ValueUpdated event of the characteristic, I get the behavior I defined (looping counting from 0-255). To be honest, maybe there isn't a bug. I may just be doing something wrong, but the user description is present in BLE debugging apps like STBLE Toolbox or LightBlue.

cdbenjamin, 2024-06-21:
> Doesn't this "firmware" you're using add a Generic Access and Generic Attribute or other default services for you?  i would imagine most hobby platforms do.  Therefore how can you know this is the characteristic, let alone the service, you are looking to interrogate.  A bit presumptuous to be filing a bug?

Yes, it does add the Generic Access and Generic Attributes services, but as far as I can tell, I am looking at the correct characteristic because I see the behavior I defined when I subscribe to the ValueUpdated event (looping counting from 0-255). Maybe there is no bug. Maybe I am just doing something totally wrong, but the user description is present when I use BLE debugging applications like STBLE Toolbox or LightBlue.

axa88, 2024-06-23:
> Maybe there is no bug. Maybe I am just doing something totally wrong, but the user description is present when I use BLE debugging applications like STBLE Toolbox or LightBlue.

Well there may be a bug, especially on windows, its a mess.  but when someone at a distance can find fault in what is presented, its hard to suggest any solution