dotnet-bluetooth-le/dotnet-bluetooth-le #874: Is Plugin.BLE only scan LE device,not suit Classic Blue Device?

deyangwangyu, 2024-07-06:
target: android 12.0 API31
I follow the guide make a test project. The scan result only contain smart ring,not contain my phone and bluetooth module(HC-06)

BTW, I didn't put
'uses-feature android:name="android.hardware.bluetooth_le" android:required="true" ' into AndroidManifest.xml

deyangwangyu, 2024-07-06:
邮件已收到，谢谢！

axa88, 2024-07-07:
> target: android 12.0 API31 I follow the guide make a test project. The scan result only contain smart ring,not contain my phone and bluetooth module(HC-06)
> 
> BTW, I didn't put 'uses-feature android:name="android.hardware.bluetooth_le" android:required="true" ' into AndroidManifest.xml

Well, that is what the "le" in dotnet-bluetooth-**_le_** means....

janusw, 2024-08-03:
Yes, BLE only, no classic Bluetooth ...

closed 2024-08-03 by janusw

deyangwangyu, 2024-08-03:
邮件已收到，谢谢！