dotnet-bluetooth-le/dotnet-bluetooth-le #877: Bug: [Adapter] Device {..} does not support BLE

BierDav, 2024-07-28:
We have recently updated to `3.1.0`, but unfortunately this version broke our code, because of a new check that has been introduced.

Our usage looks something like that:
```c#
var adapter = CrossBluetoothLE.Current.Adapter;
var device = await adapter.ConnectToKnownDeviceAsync(new Guid(guid), new ConnectParameters(forceBleTransport: true), tokenSource.Token);
``´
More specifically the problem is that on newer android devices the `BluetoothDevice.Type` is `BluetoothDeviceType.Unkown`, and this is also true if the ble device isn't even in Range. This causes the exception you can find in this snipped
[https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/2c7972e4ec3f41654b6422d7cec6b8ca72694d38/Source/Plugin.BLE/Android/Adapter.cs#L310](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/2c7972e4ec3f41654b6422d7cec6b8ca72694d38/Source/Plugin.BLE/Android/Adapter.cs#L310)

Here the implementation of SupportsBLE which uses the `Type` property:
[https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/2c7972e4ec3f41654b6422d7cec6b8ca72694d38/Source/Plugin.BLE/Android/Extensions/BluetoothDeviceExtension.cs#L7](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/2c7972e4ec3f41654b6422d7cec6b8ca72694d38/Source/Plugin.BLE/Android/Extensions/BluetoothDeviceExtension.cs#L7)

We need this check to be removed, instead we cannot upgrade to the next version.

janusw, 2024-08-03:
> We have recently updated to `3.1.0`, but unfortunately this version broke our code, because of a new check that has been introduced.

Ouch, that's a regression then. You're updating from 3.0.0, I assume?

> More specifically the problem is that on newer android devices the `BluetoothDevice.Type` is `BluetoothDeviceType.Unkown`, and this is also true if the ble device isn't even in Range.

On which host devices (and with which Android version) do you observe this? Which kind of peripheral does it occur with?

> https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/2c7972e4ec3f41654b6422d7cec6b8ca72694d38/Source/Plugin.BLE/Android/Adapter.cs#L310
>
> Here the implementation of SupportsBLE which uses [...]

This code is from #855. If this is the problem, then version `3.1.0-rc.1` should still work for you. Is that correct?

> Actually one interesting thing we noticed is that, on some devices you can fix this behavior by starting a simple BLE Scanner app from the playstore, if the app has found our BLE Device suddendly android returns the correct `BluetoothDeviceType.Le`. I could be that the fix is not removing the check but adding a step before that makes this `Type` property available.

Interesting observation. Sounds like the device type may not be initialized properly yet ...

IchoTolot, 2024-08-03:
I have the same problem, but haven't had time to take a closer look.
The whole thing occurs on two devices with API 34, API 33 works.
In addition, the device name does not seem to be returned.
As @Bierdav wrote, the problem can be solved by a previous scan. After that, the device name and type are also correct in API 34

alex3696, 2024-09-05:
> Ouch, that's a regression then

I think there is no regression here. If you try to connect to BluetoothDeviceType.Unkown you will get a GATT 133 error or completely break the device's Bluetooth stack, as was the case in my case.
It is enough to add a check in the application code, as was said.

> I could be that the fix is not removing the check but adding a step before that makes this `Type` property available.

If the device is unknown, perform a scan, I suggest you do that when connecting, something like this:
```
...
var adapter = CrossBluetoothLE.Current.Adapter;
IDevice? device = null;
try
{
  device = await adapter.ConnectToKnownDeviceAsync(new Guid(guid), new ConnectParameters(forceBleTransport: true), tokenSource.Token);
}
catch (Exception ex) // or you can catch only DeviceConnectionException exception
{
  // here you understand that the device is out of range or the bluetooth cache of the phone’s devices is not ready
    device = null;
}
if (null == device)
  device = await ConnectToScannedAsync(_Adapter, guid, conn_param, ct);
if (null == device)
    throw new Exception("Can`t connect  to device ")
...

static async Task<IDevice?> ConnectToScannedAsync(IAdapter adapter, Guid guid, ConnectParameters conn_param, CancellationToken ct)
{
    IDevice? btDevice = null;
    try
    {
        if (!IsLocationEnabled())
            throw new Exception("Loacation enable required");
        btDevice = await adapter.DiscoverDeviceAsync(guid, ct);
        if (btDevice != null)
            await adapter.ConnectToDeviceAsync(btDevice, conn_param, ct);
    }
    catch (Exception ex)
    {
        btDevice?.Dispose();
        btDevice = null;
    }
    return btDevice;
}

```
In theory, this logic could be moved into the ConnectToKnownDeviceNativeAsync function. 
But I think it is wrong to add this behavior to the dotnet-bluetooth-le library, it is better to use device verification and scanning in the application.

KevinJosephMorin, 2026-04-18:
Janusw,

I am able to connect fine via 3.0.0 without a scan.  In 3.1.0+, a scan is necessary to the same device.  I have tried rolling back and forth between 3.0.0 and 3.1.0 (and 3.2.0).  The issue happens 100% of the time on 3.1.0+ if there isn't a scan first.

I am using the "Known" device variant of the Connect routine, which didn't require a scan before. 

Can you add in a parameter that ignores the new check added in 3.1.0+ to handle situations like this?

alex3696, 2026-04-21:
> So, maybe we should better get rid of the new check (and in particular the exception) altogether, and perhaps just log a warning instead (or something like that) ...

For me this is a bad idea, because...
1. ConnectToKnownDeviceAsync connection implies that the device is KNOWN. 
2. There is no legal way or API to check the device type before attempting to connect (object NativeDevice { get; } is hidden since 3 version).  
3. Remove the check could potentially break the device stack (I was lucky to find the problem, but others may search for a long time and have a pain :-) )
4. I'll have to go back to this ugly patch:
```
   var _bluetoothAdapter = BtHelper.GetDefaultAdapter();
   var nativeDevice = _bluetoothAdapter?.GetRemoteDevice(macBytes);
   if (null == nativeDevice)
       return null;
   using (nativeDevice)
   {
       if (!nativeDevice.Type.IsLe())
           return null;
   }
   btDevice = await ad.ConnectToKnownDeviceNativeAsync(guid, conn_param, ct);
```
 > That might be an option

For this option I have two proposals to choose from:
1) add the parameter bool CheckIsLeDeviceType { get; } = true; and in ConnectToKnownDeviceNativeAsync check if (connectParameters.CheckIsLeDeviceType && !nativeDevice.SupportsBLE())... (preferably)
2) add to interface IDevice the ability to find out or check the device type, something like bool IsLe { get; }

Windows and iOS do not allow such connections. I think that connecting to DEVICE_TYPE_UNKNOWN should be disabled by default.  It should be a conscious choice by the developer: ignore the type and try to connect with the chance of breaking the stack.