dotnet-bluetooth-le/dotnet-bluetooth-le #882: Read error after reconnect due to Bluetooth toggle on Android

jspfister-sep, 2024-08-12:
I suspect the potential bug described below is due to a native device object being cached on Android across a Bluetooth off/on cycle. Looking at the source code, I can see that the cache is cleared when Bluetooth is toggled on iOS but not on Android. Interestingly, Android seems to allow the connection to be reestablished using the stale object, but any attempt to read from the device will fail.

## Steps to reproduce

1. Build and deploy [minimal reproducible sample](https://github.com/jspfister-sep/MinimalBleApp) app to an Android device. (note: all relevant code is in `MainViewModel.cs`)

2. Tap "Start Scan".

3. Select a BLE device with at least one readable characteristic.

4. Tap "Connect".

5. Wait for status to say "Connected" and for alert dialog to show the characteristic was read successfully.

6. From the Android swipe down menu, toggle Bluetooth off and back on. Return to app. (it will automatically try to reconnect)

7. Wait for status to transition from "Connecting..." back to "Connected". 

## Expected behavior
Popup indicating successful characteristic read should be shown.

## Actual behavior
Android 13: Device will reconnect but any attempt to read a characteristic will hang forever
Android 14: Device will reconnect but any attempt to read a characteristic will throw an exception (see below)

### Crashlog
```
Plugin.BLE.Abstractions.Exceptions.CharacteristicReadException: BluetoothGattCharacteristic.readCharacteristic returned FALSE
	at Plugin.BLE.Android.Characteristic.ReadInternal()
	at Plugin.BLE.Abstractions.Utils.TaskBuilder.<>c__DisplayClass11_0\`1[[System.ValueTuple\`2[[System.Byte[], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.Int32, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].<SafeEnqueueAndExecute>b__0()
--- End of stack trace from previous location ---
	at Plugin.BLE.Abstractions.Utils.TaskBuilder.<SafeEnqueueAndExecute>d__11\`1[[System.ValueTuple\`2[[System.Byte[], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.Int32, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
	at Plugin.BLE.Abstractions.Utils.TaskBuilder.<FromEvent>d__9\`3[[System.ValueTuple\`2[[System.Byte[], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.Int32, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler\`1[[Plugin.BLE.Android.CallbackEventArgs.CharacteristicReadCallbackEventArgs, Plugin.BLE, Version=3.1.0.0, Culture=neutral, PublicKeyToken=null]], System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e],[System.EventHandler, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext()
	at Plugin.BLE.Android.Characteristic.ReadNativeAsync()
	at Plugin.BLE.Abstractions.CharacteristicBase\`1.<ReadAsync>d__33[[Android.Bluetooth.BluetoothGattCharacteristic, Mono.Android, Version=0.0.0.0, Culture=neutral, PublicKeyToken=84e04ff9cfb79065]].MoveNext()
	at MinimalBleApp.MainViewModel.OnDeviceConnected(Object sender, DeviceEventArgs args)
```
## Configuration

**Version of the Plugin:** 3.1.0 (`git bisect` indicates the potential bug was introduced in 3.0.0, specifically commit 89f7f7b4f889a57ab355f96b506c6b52bcd8a37e)

**Platform:** Android 13 or 14

**Device:** Google Pixel 5a

janusw, 2024-08-13:
Thanks for the report and the analysis. This is a regression then?

> **Version of the Plugin:** 3.1.0 (`git bisect` indicates the potential bug was introduced in 3.0.0, specifically commit [89f7f7b](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/commit/89f7f7b4f889a57ab355f96b506c6b52bcd8a37e))

If this commit by @TravisRo is the culprit indeed (see PR #735), then `3.0.0-beta.5` should still work, while `3.0.0-beta.6` should fail. Is that what you see?

janusw, 2025-09-27:
Fixed via #958.

closed 2025-09-27 by janusw