dotnet-bluetooth-le/dotnet-bluetooth-le #885: iOS 18.0 still displays [?] in many scripts

Ashtontameirao, 2024-08-13:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## New Noto Fonts for iOS 18 (New System Fonts apple adds to display on every supported iOS 18 device)
- Noto Sans Vithkuqi
- Noto Sans Old Sogdian
- Noto Sans Sogdian
- Noto Serif Old Uyghur
- Noto Sans Chorasmian
- Noto Sans Elymaic
- Noto Serif Dogra
- Noto Serif Dives Akuru
- Noto Sans Nandinagari
- Noto Sans Zanabazar Square
- Noto Sans Soyombo
- Noto Serif Makasar
- Noto Sans Kawi
- Noto Sans Cypro Minoan
- Noto Sans Anatolian Hieroglyphs
- Noto Sans Tangsa
- Noto Sans Medefaidrin
- Noto Serif Tangut
- Noto Serif Khitan Small
- Noto Sans Nushu
- Noto Znamenny Notation
- Noto Music
- Noto Sans SignWriting
- Noto Serif Toto
- Noto Sans Nag Mundari
- Noto Sans Indic Siyaq Numbers
- Noto Serif Ottoman Siyaq

## Steps to reproduce

1.

2.

3.

## Expected behavior
Apple needs to add new noto fonts to iOS 18 so it can display on every iOS 18 device.

## Actual behavior
No new unicode characters and scripts get added to display on websites and apps.

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** e.g. 1.0.0 / 1.0.1-alpha3

**Platform:** e.g. iOS 10.1 / Android 4.4 / ... (including version!!! e.g. Android 5.1 / i0S 10)
iOS 18.0

**Device:** e.g. HTC Sensation /i Phone 7 ...
iPhone 13 Pro
[image]

axa88, 2024-08-14:
what?

janusw, 2024-08-14:
Obviously not related to dotnet-bluetooth-le (or to BLE in general). Seems like you somehow picked the wrong project to open this issue ...

closed 2024-08-14 by janusw