dotnet-bluetooth-le/dotnet-bluetooth-le #889: Error with GetCharacteristicsAsync method

Nick-Blair19, 2024-09-24:
I am using ble.plugin 3.1.0 and ios ipad 17.7 
I have error in visual studio with code var sendCharacteristic = service.GetCharacteristicsAsync(GattIdentifiers.UartGattCharacteristicSendId);
I get error 
'Task<IService>' does not contain a definition for 'GetCharacteristicsAsync' and no accessible extension method 'GetCharacteristicsAsync' accepting a first argument of type 'Task<IService>' could be found (are you missing a using directive or an assembly reference?)
This is the code snippet
   public  void Connect(IDevice _gattDevice)
   {
       this._adapter.ConnectToDeviceAsync(_gattDevice);
       var ble = CrossBluetoothLE.Current;
       var adapter = CrossBluetoothLE.Current.Adapter;
       try
       {
           var service = _gattDevice.GetServiceAsync(Guid.Parse("fefb"));
           if (service != null)
           {
             
                var sendCharacteristic = service.GetCharacteristicsAsync(GattIdentifiers.UartGattCharacteristicSendId); 

This should be working code and I get similar errors with read and write characteristics methods?
Regards Nick.

Nick-Blair19, 2024-09-24:
[bleplugin tme-5.zip](https://github.com/user-attachments/files/17120929/bleplugin.tme-5.zip)

Nick-Blair19, 2024-09-24:
Just added sample project zip

M0n7y5, 2024-09-25:
This repo is dead... You different library.

axa88, 2024-09-26:
> This repo is dead... Use different library.

afraid you may be right

janusw, 2024-10-10:
> This repo is dead... Use different library.

Nah, it's not quite dead yet, it only smells a bit funny 😝

janusw, 2024-10-10:
> var service = _gattDevice.GetServiceAsync(Guid.Parse("fefb")); 

The compile error you see is basically due to an `await` missing here.

This is a C# beginner question, not a bug report! (Please use 'Discussions' instead of 'Issues' for such things in GitLab, if at all.)

Nick-Blair19, 2024-10-10:
no the error is not with
var service = _gattDevice.GetServiceAsync(Guid.Parse("fefb"));
but with
            var sendCharacteristic = service.GetCharacteristicsAsync(GattIdentifiers.UartGattCharacteristicSendId); 
Also get similar errors with read and write characteristics ant StartUpdates

???

Nick-Blair19, 2024-10-10:
[image]
This is the error

janusw, 2024-10-10:
That is, as I said, because you're missing an `await` on the assignment of your service (so that it's of type `Task` instead of `IService`). Trust me ;)

(and please do some reading on asynchronous operations in C#, before posting further comments here)

Nick-Blair19, 2024-10-10:
oh by the way janus while youre there can I ask you a few more questions?
1. How do I notify characteristics? Before when using xamarin corebluetooth I had SetNotifyValue but cant see anything like this here?
2. how do I send Rx and Tx credits to 'Credit' characteristics (I have a Telit BLE module with a GATT service)?
Before I had corebluetooth code
 this.PendingLocalUARTCreditsCount = this.MaxLocalUARTCreditsCount - this.LocalUARTCreditsCount;
 Byte[] byteData = new Byte[1];
 byteData[0] = (Byte)((this.PendingLocalUARTCreditsCount & 0xFF));
 NSData valueData = (NSData.FromArray(byteData));
 peripheral.WriteValue(valueData, UartRxCreditsCharacteristic, CBCharacteristicWriteType.WithoutResponse);

3. Lastly do you have the ability to discover bluetooth device adresses with ble.plugin?

Very many thanks.
Regards Nick