dotnet-bluetooth-le/dotnet-bluetooth-le #890: .NET 9 Support + Trimming

bcaceiro, 2024-10-10:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.

## Steps to reproduce

1. Run a .NET MAUI with net9 ( rc2) with this package in release mode with Trimming/AOT

2. Multiple errors for properties not defined

3.

## Expected behavior
No errors

## Actual behavior
Tell us what happens instead

### Crashlog

If something causes an exception paste full stack trace + Exception here

## Configuration

**Version of the Plugin:** 3.1.0

**Platform:** Android

janusw, 2024-10-10:
Maybe it's a good start to add `net9.0` targets to our sample app, so that this sort of thing can be reproduced.

I'll need to check what it takes to support Trimming/AOT in .NET 9 ...

axa88, 2024-10-17:
fwiw with recent SDK updates targeting future .net9 AOT changes, building in .net8 will yield new warnings/errors.
After researching it, this will help:

	<PropertyGroup>
	  <CsWinRTAotOptimizerEnabled>false</CsWinRTAotOptimizerEnabled>
	  <AllowUnsafeBlocks>true</AllowUnsafeBlocks>
	</PropertyGroup>

janusw, 2025-03-30:
Just released [v3.2.0-beta.1](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.2.0-beta.1) with .NET 9 support.

@bcaceiro, does this release work for you (with full trimming/AOT), or do you still get errors?

bcaceiro, 2025-03-31:
@janusw I have experimented in release mode ( without linking) and it works okay. Atm in my app I have the linking disabled because of other stuff

janusw, 2025-04-01:
Alright, so we can close this ticket?

bcaceiro, 2025-04-01:
I think so!

janusw, 2025-04-01:
Awesome! 👍

closed 2025-04-01 by janusw