dotnet-bluetooth-le/dotnet-bluetooth-le #894: NullReferenceException when using IAdapter adapter in new methods

FedeLovera, 2024-10-23:
Problem Description:

I am using the Plugin.BLE in a .NET MAUI project to manage BLE connections. However, I'm encountering a problem when trying to use the adapter in new methods of my BLELoginServiceAndroid class. The issue is that I'm getting a NullReferenceException when attempting to access the adapter inside the new methods, even though the adapter works fine in older methods where it is also used. I have tried initializing the adapter in different ways and ensuring it is not null before usage, but the problem persists.

Technical Details:
BLE Plugin: Plugin.BLE v3.0.0
Platform: Android
Framework: .NET MAUI 8 with Blazor

Relevant Code:
internal class BLELoginServiceAndroid : IBLELoginService
{
    private IAdapter adapter = CrossBluetoothLE.Current.Adapter;
    private bool adapterInitialized = false;
    private IMemoryLog log;

        public async void Initialize()
        {
            var state = ble.State;
            ble.StateChanged += (s, e) =>
            {
                Console.WriteLine($"The bluetooth state changed to {e.NewState}");
            };

            adapter.ScanTimeout = 1300; 
            adapter.DeviceDiscovered += (s, a) =>
            {
                log.AddLog($"DeviceDiscovered");
                byte[] manufacturer = Array.Empty<byte>();
            };
            adapter.DeviceConnected += async (s, d) => await LoginWithCredentials(d.Device);
        }

        public async Task DoLogin(int appId, int[] accesosIds, byte[] cryptedPinPwd)
        {
            //Method working correctly
            this.appId = appId;
            this.cryptedPinPwd = cryptedPinPwd;
            this.idAccesos = accesosIds;

            await adapter.StartScanningForDevicesAsync();
            await Connect(null, null);
        }
        public async Task<Dictionary<string, object>> ScanAndGetBeaconInfo(int appId, int[] accesosIds, byte[] cryptedPinPwd)
            {
                this.appId = appId;
                this.cryptedPinPwd = cryptedPinPwd;
                this.idAccesos = accesosIds;
                var beaconInfo = new Dictionary<string, object>();
  
                await adapter.StartScanningForDevicesAsync();
                //NullReferenceException here
                await Task.Delay(1000);
        }

}

Additionally, in my MauiProgram.cs, I have configured the services as follows:
builder.Services.AddSingleton<IBLELoginService>(bleLoginServiceAndroid); 
builder.Services.AddScoped<IBluetoothService, BluetoothService>();

Steps to Reproduce:
Initialize CrossBluetoothLE.Current.Adapter in a BLE service class.
Attempt to scan for Bluetooth devices with the ScanAndGetBeaconInfo method.
Receive a NullReferenceException when accessing the adapter inside this method.

Expected Behavior:
The adapter should be properly initialized and available for scanning devices, as it works in other methods.

Actual Behavior:
A NullReferenceException occurs when trying to access the IAdapter in the ScanAndGetBeaconInfo method, while it doesn't happen in other parts of the class.

Solutions Tried:
Verified that the adapter is not null before using it.
Explicitly initialized the adapter in the class constructor.
Checked that the adapter is working correctly in other methods without issues.

Development Environment:
IDE: Visual Studio 2022
Platform: Android
Framework: .NET MAUI with Blazor

Possible Cause (Hypothesis):
The issue might be related to the asynchronous initialization of the adapter. It seems like the adapter may not be fully initialized when I attempt to use it in the ScanAndGetBeaconInfo method, even though it works correctly in other methods.

Question:
Is there a recommended way to handle the initialization of CrossBluetoothLE.Current.Adapter in asynchronous contexts or during device scanning to avoid these types of errors? Any suggestions on how to improve the initialization or lifecycle management of the adapter would be very helpful.

janusw, 2024-10-30:
> ```
>     public async Task<Dictionary<string, object>> ScanAndGetBeaconInfo(int appId, int[] accesosIds, byte[] cryptedPinPwd)
>         {
>             this.appId = appId;
>             this.cryptedPinPwd [...]

So, you get the exception upon calling `adapter.StartScanningForDevicesAsync`, right? Is the adapter itself `null` or is the exception thrown somewhere inside of `StartScanningForDevicesAsync`? If the latter, do you have a backtrace?

FedeLovera, 2024-11-11:
The method StartScanningForDevicesAsync is the one causing the error, and the call is as follows:
System.NullReferenceException
Mensaje = Object reference not set to an instance of an object.