dotnet-bluetooth-le/dotnet-bluetooth-le #898: Method not found: Java.Lang.Object Android.Content.Intent.GetParcelableExtra(string,Java.Lang.Class)

stsln, 2024-11-08:
Hi! I'm using net6.0 in my project and I'm having trouble connecting to a device that requires a passcode (PIN). When I try to use the vanilla library, I get an exception. I also want to note that I use Avalonia.

## Steps to reproduce

1. `<PackageReference Include="Plugin.BLE" Version="3.1.0" />`
2.
```
await App.BleService.Adapter.BondAsync(DeviceBle);
await App.BleService.Adapter.ConnectToDeviceAsync(DeviceBle, parameters, _tokenSource.Token);
```

## Expected behavior
Must make a pairing request and connect to the device.

## Actual behavior
An exception occurs during connection.

### Crashlog
```
System.MissingMethodException: Method not found: Java.Lang.Object Android.Content.Intent.GetParcelableExtra(string,Java.Lang.Class)
   at Android.Content.BroadcastReceiver.n_OnReceive_Landroid_content_Context_Landroid_content_Intent_(IntPtr jnienv, IntPtr native__this, IntPtr native_context, IntPtr native_intent) in /Users/runner/work/1/s/xamarin-android/src/Mono.Android/obj/Release/net6.0/android-32/mcw/Android.Content.BroadcastReceiver.cs:line 406
   at Android.Runtime.JNINativeWrapper.Wrap_JniMarshal_PPLL_V(_JniMarshal_PPLL_V callback, IntPtr jnienv, IntPtr klazz, IntPtr p0, IntPtr p1) in /Users/runner/work/1/s/xamarin-android/src/Mono.Android/Android.Runtime/JNINativeWrapper.g.cs:line 187
```

## Configuration

**Version of the Plugin:** 3.1.0

**Platform:** Android 7.1.2 and 12

**Device:** Xiaomi Redmi 4A and POCO X6

janusw, 2024-11-09:
Hm, strange.

Inside Plugin.BLE, the mentioned method is used here:

https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/ba3f06644925c3f6895ace703dfc51acce964e03/Source/Plugin.BLE/Android/BroadcastReceivers/BondStatusBroadcastReceiver.cs#L33-L44

Which Android API level are you targeting? (Since you mention net6.0, that's probably API level 31, right?)

stsln, 2024-11-09:
> Hm, strange.
>
> Inside Plugin.BLE, the mentioned method is used here: [...]

Yes, I'm targeting on Android API 31 and 25.

janusw, 2024-11-09:
> > Which Android API level are you targeting? (Since you mention net6.0, that's probably API level 31, right?)
> 
> Yes, I'm targeting on Android API 31 and 25.

No, those are the versions you actually run on, I think.

I was asking about the version that you _target_. That is the API level that you are building for (aka the SDK that you are using), i.e. the highest API level supported by your app. Your stack trace seems to show that this is API level 32:

> Mono.Android/obj/Release/net6.0/android-32/mcw/Android.Content.BroadcastReceiver.cs:line 406

Note: The `GetParcelableExtra` method with two parameters (which your exception refers to) is supported from API level 33 on, and as you can see in the code I linked above, it is guarded by a corresponding version check, therefore I don't quite understand how it can be called in your case ... 🤔

janusw, 2024-11-09:
Also note that version 3.1.0 does not support net6.0 directly, so probably the netstandard2.0 / Xamarin targets will be used in your case, and maybe that is somehow the reason for the error you see (although I still don't fully understand it).

Is there anything keeping you from updating to net8.0?

stsln, 2024-11-14:
> Note: The `GetParcelableExtra` method with two parameters (which your exception refers to) is supported from API level 33 on, and as you can see in the code I linked above, it is guarded by a corresponding version check, therefore I don't quite understand how it can be called in your case ... 🤔

Yes, I have already seen that there is no net6.0. I myself recompiled the library for net6.0-android by removing the GetParcelableExtra call with two parameters. And I got the connection working correctly)
There is no way to upgrade to net8.0 yet due to the dependency of other internal libraries in my project.

stsln, 2024-11-14:
> Also note that version 3.1.0 does not support net6.0 directly, so probably the netstandard2.0 / Xamarin targets will be used in your case, and maybe that is somehow the reason for the error you see (although I still don't fully understand it).
> 
> Is there anything keeping you from updating to net8.0?

Yes, I have already seen that there is no net6.0. I myself recompiled the library for net6.0-android by removing the GetParcelableExtra call with two parameters. And I got the connection working correctly)
There is no way to upgrade to net8.0 yet due to the dependency of other internal libraries in my project.

janusw, 2024-11-15:
Well, ok. We might have been a bit too quick with removing net6.0 support. And if this request had come a bit earlier, I'd have no problem with keeping or re-adding it, but by now .NET 6 is officially out of support by Microsoft. So I'm afraid there's no way we're going back now.

If you are stuck with net6.0, I guess your options for now are:
* Build a custom variant of Plugin.BLE yourself (as you apparently did already).
* Use an older version. 3.0.0 should still have net6.0 TFMs.

In the long run you should really update to net8.0.