dotnet-bluetooth-le/dotnet-bluetooth-le #901: Bluetooth LE Scan Limitations?

JaronrH, 2024-11-11:
I'm trying to write a MAUI application that uses Plugin.BLE to detect a non-paired bluetooth device that it advertising and get that advertised data.  

Initially, I prototyped the BTLE logic using Python and Bleak to find all devices (before filtering by device address for the one I wanted):
```
import asyncio
import logging
import sys
import binascii

from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData

logger = logging.getLogger(__name__)

def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    print("--------------------------------------------------------------------------")
    print(f"Device name: {device.name}")
    print(f"Device address: {device.address}")
    print(f"RSSI: {device.rssi}")
    print("Advertisement data:")
    for key, value in advertisement_data.manufacturer_data.items():
        print(f"Manufacturer ID: {key}, Data: {value}")
    for service in advertisement_data.service_uuids:
        print(f"Service UUID: {service}")

async def main(service_uuids):
    scanner = BleakScanner(simple_callback, service_uuids)

    while True:
        print("(re)starting scanner")
        await scanner.start()
        await asyncio.sleep(5.0)
        await scanner.stop()

if __name__ == "__main__":
    logging.basicConfig(
        level=logging.INFO,
        format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
    )
    service_uuids = sys.argv[1:]
    asyncio.run(main(service_uuids))

```

I ported this code to MAUI and found that it did not detect the same devices during a scan!  This included the device I needed.  

Device as displayed by Python code when found:
```
--------------------------------------------------------------------------
Device name: None
Device address: ED:67:3B:19:4C:AA
RSSI: -64
Advertisement data:
Manufacturer ID: 65535, Data: b'\xaa\xbb\xedg;\x19L\xaa\x17-2g\xff\xff\xff\x04`$\tL\x05\x03Ug'
```
(The advertised data changes every ~140ms as it is sending out sensor readings.)

For reference, I did the Python and MAUI scans on the same Windows computer.  Also, I loaded the app onto my Android Phone (with working permissions) and had the same issue with seeing all available devices (again, I can always see some devices -- Phone, Computer, Headphones -- in a combination of connected/paired and disconnected/unpaired states so I know the scanning is working at some level!).

Any ideas?

Rough, example MAUI code:
```
private void StartScanning()
        {
            var adapter = CrossBluetoothLE.Current.Adapter;
            adapter.ScanMatchMode = ScanMatchMode.AGRESSIVE;
            adapter.ScanMode = ScanMode.LowLatency;
            adapter.ScanTimeout = 1000;
            adapter.DeviceDiscovered += OnDeviceDiscovered;
            Task.Run(async () =>
            {
                await Task.Delay(TimeSpan.FromSeconds(1));
                while (!stopScanningBluetoothDevices)
                {
                    try
                    {
                        if (!adapter.IsScanning)
                        {
                            Debug.WriteLine("--------------START SCANNING BTLE------------------");
                            await adapter.StartScanningForDevicesAsync();
                        }
                        else
                            await Task.Delay(TimeSpan.FromSeconds(1));
                    }
                    catch (SecurityException)
                    {
                        Debug.WriteLine("Missing Bluetooth Scan Permission");
                        await Task.Delay(TimeSpan.FromSeconds(15));
                    }
                    catch (Exception e)
                    {
                        Debug.WriteLine($"CRAP!  {e}");
                    }
                }
                adapter.DeviceDiscovered -= OnDeviceDiscovered;
            });
        }

        private void OnDeviceDiscovered(object sender, DeviceEventArgs e)
        {
            var device = e.Device;
            Debug.WriteLine($"Device: {device.Name}");
            if (device.Name.ToLower().Contains("ED:67:3B:19:4C:AA".ToLower()))
                Debug.WriteLine("FOUND!!!!!!!!!!!!!!!!!!!!!!!!!!!"); // Never happens but I do find other devices!
            // Get Advertisement Data
            foreach (var manufacturerData in device
                         .AdvertisementRecords
                         .Where(record =>
                             record.Type == AdvertisementRecordType.ManufacturerSpecificData && record.Data != null
                         )
                         .Select(i => i.Data)
                    )
            {
                var hexString = BitConverter.ToString(manufacturerData).Replace("-", "");
                Debug.WriteLine(hexString);
            }
        }
```

axa88, 2024-11-12:
too much trouble to use the template and let us know all the pertinent info eh?

Ok ill assume this is Windows cuz i already know what your problem is... you cant scan for _any_ device that doesn't have a name... ammirite?
if that's your issue and you want to file a bug ill vouch for you.

JaronrH, 2024-11-12:
Sorry - Didn't realize the Template was such a strict requirement here.  I'll close this issue.

(And, yes, the device does not have an advertised name.)

axa88, 2024-11-12:
Relax, I'm not the boys of you and didn't mean to scare you away. But providing info really helps to isolate problems. 

I take it you're not going to file that bug ?

janusw, 2024-11-15:
> Sorry - Didn't realize the Template was such a strict requirement here.

It's not.

And, yes, don't get scared away by some random dude, who hasn't even contributed to this repo (code-wise) 😉 

Reopening ...

janusw, 2024-11-15:
> ```
> private void StartScanning()
>         {
>             var adapter = CrossBluetoothLE.Current.Adapter;
>             adapter.ScanMatchMode = ScanMatchMode.AGRESSIVE;
>             adapter.ScanMode = [...]

Your problems might be due to the fact that you use the `DeviceDiscovered` event. If you really want to catch all advertisements, you should use `DeviceAdvertised`.

JaronrH, 2024-11-15:
> Your problems might be due to the fact that you use the `DeviceDiscovered` event. If you really want to catch all advertisements, you should use `DeviceAdvertised`.

I had tried both DeviceDiscovered and DeviceAdvertised - neither  ever saw the Mac address I was looking for much less return the advertised data.

axa88, 2024-11-15:
> And, yes, don't get scared away by some random dude, who hasn't even contributed to this repo (code-wise) 😉

So your saying if i contribute code i can scare people away?
And identifying the _actual_ issue this other random dude is having, unnamed device, is contributing enough for me...

axa88, 2024-11-15:
> I had tried both DeviceDiscovered and DeviceAdvertised - neither ever saw the Mac address I was looking for much less return the advertised data.

Thats cuz i told you what the problem was already....

janusw, 2024-11-16:
> > And, yes, don't get scared away by some random dude, who hasn't even contributed to this repo (code-wise) 😉
> 
> So your saying if i contribute code i can scare people away?

Totally. 😆 

Seriously, though: I'm sorry about the stupid joke and the sloppy language. No offense intended. Testing, reporting bugs and contributing to discussions is a kind of contribution as well, and an important one.

Scaring off other people is not a recommended behavior for anyone, including maintainers, of course. (Btw, I've seen our issue-checker bot do it as well, although that is not really its purpose.)

> And identifying the _actual_ issue this other random dude is having, unnamed device, is contributing enough for me...

If you did identify the issue, maybe you did not express yourself precise enough ...?

janusw, 2024-11-16:
> ```
>         private void OnDeviceDiscovered(object sender, DeviceEventArgs e)
>         {
>             var device = e.Device;
>             Debug.WriteLine($"Device: {device.Name}");
>             if [...]

Second try: It seems like you're expecting the device name to contain the address here. As your Python output shows, the device that you're looking for does not have a name. You should probably check `device.Id` instead. (Note that this may not work on other platforms, like iOS, though).

If you're only looking for advertisements from a particular kind of device, it may be a good idea to use scan filters, but first you should verify that you see the desired device at all ...

janusw, 2024-11-16:
> Ok ill assume this is Windows cuz i already know what your problem is...

The original report mentions that the testing was done on Windows (and Android).

> you cant scan for _any_ device that doesn't have a name... ammirite?

It should be perfectly possible to scan for arbitrary devices (including ones without a names).

(Possibly I misunderstood this sentence?)

> if that's your issue and you want to file a bug ill vouch for you.

No need to open a second issue. And no need to "vouch". We already have a "bug report" here, we just need to clarify what the actual bug is (and if there really is one).

axa88, 2024-11-16:
> > So your saying if i contribute code i can scare people away?
> 
> Totally. 😆
Deal.
 
> Seriously, though: I'm sorry about the stupid joke and the sloppy language. No offense intended. Testing, reporting bugs and contributing to discussions is a kind of contribution as well, and an [...]

No need apologize, people have way too thin skin online, believing they can judge the intonation of the written word and then always judge it as malicious. Go back to read what i wrote imagining i have a smile on my face, it will come across differently...

IMHO the template should be a requirement (unless you enjoy assuming randos thoughts), and scrap the bot.
 
> > And identifying the _actual_ issue this other random dude is having, unnamed device, is contributing enough for me...
> If you did identify the issue, maybe you did not express yourself precise enough ...?

ok i can see where you're going with this, though ironic _I'm_ the one here being called out for not being descriptive enough (•_•)  but i aint mad... 
 
> > you cant scan for _any_ device that doesn't have a name... ammirite?
> It should be perfectly possible to scan for arbitrary devices (including ones without a names).
> (Possibly I misunderstood this sentence?)

Yes, that's exactly what im saying _doesn't_ happen, no matter what ive tried, the API as is, does not return any device in `DiscoveredDevices` that does not have `IDevice.Name` property populated on Windows.  I don't recall what `DeviceAdvertised` returns but i don't think does either.
  But ive not investigated thoroughly, ive found so many issues with the API windows implementation as well as issues with Bluetooth on Windows OS itself and rather not file an issue in the wrong place.

> The original report mentions that the testing was done on Windows (and Android).
I dont see any mention of this issue reported on Android.  The template wasn't used so im genuinely curious to know how you knew this?

janusw, 2024-11-16:
> Yes, that's exactly what im saying _doesn't_ happen, no matter what ive tried, the API as is, does not return any device in `DiscoveredDevices` that does not have `IDevice.Name` property populated [...]

@JaronrH Could you please check this hypothesis? Do you see any advertisements without device name on Windows? If not, this should be different from Android (I see lots of devices without name there, with the BLE.Client.Maui app).

janusw, 2024-11-16:
> > The original report mentions that the testing was done on Windows (and Android).
> 
> I dont see any mention of this issue reported on Android. The template wasn't used so im genuinely curious to know how you knew this?

By carefully reading the fine print in the report: 😉

> Also, I loaded the app onto my Android Phone (with working permissions) and had the same issue with seeing all available devices

axa88, 2024-11-16:
> By carefully reading the fine print in the report: 😉
> 
> > Also, I loaded the app onto my Android Phone (with working permissions) and had the same issue with seeing all available devices

ah touché ... i quickly read that as they _could_ see all available devices.

janusw, 2024-11-16:
> @JaronrH Could you please check this hypothesis? Do you see any advertisements without device name on Windows? If not, this should be different from Android (I see lots of devices without name [...]

So, I just dusted off an old Win10 laptop that I use _very_ rarely and ran BLE.Client.Maui on Windows.

And indeed I see no devices that have an empty `Device.Name`. But that does not mean that no-name devices are not discovered, but instead it looks like devices the have no name obtain a name like "Bluetooth ab:cd:ef:00:00:00", where the latter part is the device address.

There are two conclusions from this observation:

1. @axa88, I'm afraid your hypothesis (that devices with no name are not discovered on Windows) is wrong.
2. @JaronrH, your method of looking for the device address in the device name can indeed work, but only on Windows (not on Android), and only for devices that have no name set.

At this point, I'm completely clueless why there might be advertisements missing, in particular if this happens both on Android and Windows. @JaronrH if you have any further clues what might be happening, please let us know.

axa88, 2024-11-16:
So i finally dug in to it as well.

It appears to me the Android SDK presents a whole bunch of additional bluetooth adverts in the air than what the windows SDK chooses to present. Including many Apple/Samsung Find My Device and many other adverts that dont have names, but are not then renamed by the OS/SDK and made available to the API.
Where in the case windows decides to expose a nameless device, it does appear to rename it to a named device before its presented by the SDK.  Its confusing cuz i dont know what these rules for presentation are.

This is what lead me, wrongly, to believe there were many lost devices which all happened to have no name, regardless of where they were being lost, as I never looked into it enough to consider it an API issue until now.

Who knows what the next version/OS/hardware/chipset might do so here's mine that hides alot of what Android presents:
```
Version of the Plugin: e.g. 3.0.1
Platform: WinUI 10.0.19041

Device:
Processor Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz 1.80 GHz Device ID 2F7EBCF5-2024-4778-9E85-988D5FD862D6 System type 64-bit operating system, x64-based processor
Intel Dual Band Wireless-AC 8265, Wi-Fi 2x2 802.11ac + Bluetooth 4.1, M.2 card
```

pretty old hardware

JaronrH, 2024-11-16:
I've worked with things a bit on my side and I don't have anything new to share. 
 - My code was doing a case-insensitive search for the MAC address within the device.Name so it should have found any variation of the name with the MAC address in it!
  - I was logging all the names found (most being "Bluetooth xx:xx....") which means that, even if I didn't catch it in code, I would have seen it in the debug log!
   - I tried the test again on Windows and Android.  Windows MAUI can't see it and today Android decided to crash when the scan started (didn't get that worked out).
    - Confirmed Python could still see the device on Windows and nRF still found it on Android.
    - I used both DeviceDiscovered and DeviceAdvertised events (could see the double logging for found devices/broadcasts)
    
 Here is the nRF snapshot of the found device:   
[image]

janusw, 2024-11-17:
Thanks for the feedback, @JaronrH.

Some general suggestions:

1. Increase the `ScanTimeout` from 1s to something longer (it seems you use 5s in Python), just to make sure you're not missing anything due to the short scanning duration.
2. Check if you see any advertisements carrying `ManufacturerSpecificData` using Plugin.BLE. On Windows? On Android?
3. Maybe try to use the `BLE.Client.Maui` sample app to cross-check if you see your expected device in the list of advertisements? If it is transmitting at an interval of 140ms, it should show up quickly there.

Plus some more comments ...
 
> * My code was doing a case-insensitive search for the MAC address within the device.Name so it should have found any variation of the name with the MAC address in it!
>
>     * I was logging all the [...]

As an alternative to your method of finding the device via its address (which anyway is not portable, i.e. will not work on all platforms), maybe you can try to find the device via the data it is carrying?

I.e. in you received AdvertisementRecords, look for ones that have the type `ManufacturerSpecificData`. Do you see any of those? If yes (and if you get a lot of them), check that (at least the beginning of) the data array matches what you expect.

>     * I tried the test again on Windows and Android.  Windows MAUI can't see it and today Android decided to crash when the scan started (didn't get that worked out).

What's the cause of the crash? Do you have a stack trace, or at least an error message/exception?

>       - Confirmed Python could still see the device on Windows and nRF still found it on Android.

That's an important piece of information, which I take as a strong indication that you should also be able to see the device with Plugin.BLE.

>       - I used both DeviceDiscovered and DeviceAdvertised events (could see the double logging for found devices/broadcasts)

Using `DeviceAdvertised` should be enough. `DeviceDiscovered` is not useful for your use case.

JaronrH, 2024-11-18:
> 1. Increase the `ScanTimeout` from 1s to something longer (it seems you use 5s in Python), just to make sure you're not missing anything due to the short scanning duration.

Been running with 10s ScanTimeout -- no change for my test app.

> 3. Maybe try to use the `BLE.Client.Maui` sample app to cross-check if you see your expected device in the list of advertisements? If it is transmitting at an interval of 140ms, it should show up quickly there.

I just tried it out (sorry for the delay!).  On windows, it did not see it HOWEVER it **did on Android** (through both the Advertised and Discovered event handlers.).

janusw, 2024-11-19:
> I just tried it out (sorry for the delay!). On windows, it did not see it HOWEVER it **did on Android** (through both the Advertised and Discovered event handlers.).

Thanks for checking. So this is a problem on Windows only.

decriptor, 2025-01-07:
I think I'm running into a similar issue. I have noticed that the advertisements are showing up in this app: https://github.com/microsoft/BluetoothLEExplorer (Advertisement Monitor) but not in my little sample Plugin.BLE I'm running in LINQPad.

axa88, 2025-01-09:
@JaronrH 
i think i have found the source of this issue.
In the windows implementation there is a process of obtaining the underlying device during a scan.
The current implementation is never "registering" per say, devices which this underlying device cannot be created.

I dont yet know under what conditions windows wont allow this, but there are may devices in the air that wont be... 
A quick test the devices as found using NRf for android, which i couldn't get a device created in windows seemed to be ones that have no name as i earlier suggested, and sort of advertisement beacons such the "FindMy" service from Apple, and adverts from Apple and Windows which i also suspect are some kind of Advertising service Beacons.

I would suspect your device might have similar properties and while the advertisements are received they were not registered by the library and you wouldn't ever know they were indeed discovered.

Looking at the code, the windows implementation, to be kind, is not at all done well and I wont be look into investigating a quick fix for this as there are far too many issues and im committed to a complete rewrite.

But if you would like to try to work around it you could look at line 205 of the Windows Adapter implementation.
Where the comment mentions to make sure advertisements actually returns a device, remove that as it will not likely ever return a device for these types of devices.  At least they then should show up in your DiscoveredDevices listing.
 

```
       private void AdvertisementReceived(BluetoothLEAdvertisementWatcher watcher, BluetoothLEAdvertisementReceivedEventArgs ad)
	{
		var deviceId = ad.BluetoothAddress.ParseDeviceId();

		if (DiscoveredDevicesRegistry.TryGetValue(deviceId, out var device))
		{
			// This deviceId has been discovered
			Trace.Message($"{nameof(AdvertisementReceived)} - Old: {0}", ad.ToDetailedString(device.Name));
			(device as Device)?.Update(ad.RawSignalStrengthInDBm, ParseAdvertisementData(ad.Advertisement));
			HandleDiscoveredDevice(device);
		}
		else
		{
			var bluetoothLeDevice = BluetoothLEDevice.FromBluetoothAddressAsync(ad.BluetoothAddress).AsTask().Result;
			if (bluetoothLeDevice != null) //make sure advertisement bluetooth address actually returns a device
			{
				#pragma warning disable CA1416
				device = new Device(this, bluetoothLeDevice, ad.RawSignalStrengthInDBm, deviceId, ParseAdvertisementData(ad.Advertisement), ad.IsConnectable);
				#pragma warning restore CA1416
				Trace.Message("AdvReceived - New: {0}", ad.ToDetailedString(device.Name));
				HandleDiscoveredDevice(device);
			}
		}
	}
```

JaronrH, 2025-01-09:
@axa88 Thanks for all the work you put into this!   I may look into getting the .NET implementation working but, for now, since the sensors I needed to read were in one area, I just added an edge device that can handle what I need and push the data to my MQTT network.  While not as convenient, it has provided a good work around for my proof-of-concept project.

linjmeyer, 2025-01-17:
[image]

Hey @axa88 thanks for the input, I just hit this issue and found this thread and I can confirm the same.  The above screenshot is running my MAUI app from my Mac, and I can see the device and advertisements.  It shows as not connectable/bondable.

When I run in Windows, the same code never sees the device (despite other BLE apps on the Windows machine seeing it just fine).  I suspect it is because of this problem where the Windows implementation filters unregistered devices out? 

You mentioned a rewrite, any tbd. on that?  If I fork this and try to remove the if condition you pointed out and it fixes the problem, would a PR be accepted for a quick fix?

Thanks!

axa88, 2025-01-18:
> Hey [@axa88](https://github.com/axa88) thanks for the input, I just hit this issue and found this thread and I can confirm the same. The above screenshot is running my MAUI app from my Mac, and I can see the device and advertisements. It shows as not connectable/bondable.

> 
> When I run in Windows, the same code never sees the device (despite other BLE apps on the Windows machine seeing it just fine). I suspect it is because of this problem where the Windows implementation filters unregistered devices out?
> 

Yea i know it could work as everything else on my windows machine worked fine...
I never ran on mac/ios but the same results on Android.  Android receives and returns 'all' bluetooth advertised traffic, while windows will not "register", so to speak, any device that Windows OS will not create an underlying (aka native) object for. 

> You mentioned a rewrite, any tbd. on that? If I fork this and try to remove the if condition you pointed out and it fixes the problem, would a PR be accepted for a quick fix?
> 

Yes i think that _could_ be a quick fix, but I believe you will also likely have to avoid "using" those new found devices in anyway.  I mean just display the info, try to bond/connect it may crash your app.

Ya ive rewritten the windows implementation and it works great.
But it wouldnt get merged, you can read more about that here #917

janusw, 2025-01-18:
> You mentioned a rewrite, any tbd. on that? If I fork this and try to remove the if condition you pointed out and it fixes the problem, would a PR be accepted for a quick fix?

If an appropriate fix for this issue can be found in this way, it certainly has good chances of getting accepted. It would probably involve setting up an `IDevice` that has no corresponding `NativeDevice`, right? If so, one also needs to make sure that the condition of `NativeDevice == null` is properly handled everywhere and will not lead to any unwanted exceptions / crashes.

A complete rewrite is a totally different question, but I hope that is not necessary to fix this issue.

axa88, 2025-01-18:
> If an appropriate fix for this issue can be found in this way, it certainly has good chances of getting accepted. It would probably involve setting up an `IDevice` that has no corresponding [...]

All i was stating was that _my_ implementation does not have a chance of being merged, as along the way im no longer holding back from making architectural improvements which require changes in other implementations... I mean we are debating the use of white space for petes sake`

If someone goes into the fix understanding that the reason the windows implementation doesn't work is because these devices that are not getting underlying devices arent being included; And that including these as IDevices which dont have an underlying device will only make the current implementation less stable..... Then perhaps they can fix all of these issues not feeling the need to rewrite the entire implementation... 
Though for the record, there is absolutely no doubt in my mind rewriting the entire thing is by far the best thing to do. even the parts of current implementation that work were not done right...

axa88, 2025-01-18:
@linjmeyer 

fwiw my fixed Maui app , you can see the unnamed , non connectable devices listed, usually these are apple find my , samsung or google similar services... anything that is the roll of a Beacon for sure. Its close but I now find windows more stable than even android... only you cant get rssi updates when connected... differnt hardware architecture 

[image]