dotnet-bluetooth-le/dotnet-bluetooth-le #913: Pull #899 prevents the built in Windows UI pairing , removing any security the pairing process intends to provide

axa88, 2025-01-04:
So it turns out my [original assessment](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/issues/899#issuecomment-2466663466) was correct, and pull #899 breaks built in Windows support for pair handling (on WinUI 3 and perhaps others like UWP)

TL/DR
In summary, the pull forces pairing to try and complete without any authentication or user interaction. But if you have properly configured your windowed based programming to use authentication, then the built in pairing mechanism which worked prior to this pull #899, ive now confirmed doesn't any longer.

 That in and of itself is what this issues is about, _not_ that Windows pairing hasn't been fully supported (available in #911), but that this pull breaks a very functional component in order to hack a less functional work around.

Details:

Windows or at least some windowed frameworks like WinUI, provides a built in Bluetooth pairing interface, just how the Android OS provides, but you need to make the simple standard Bond request rather than for use of the "custom" request as was introduced in the offending pull.

So when you _used to_ make a bond request BLE peripheral which supports user authentication, the 2 devices would negotiate the most secure process available based on their shared capabilities. Then WinUI would spawn all necessary user controls to complete any of the various pairing processes that was choosen.

This was a very functional feature… For example if a peripheral has configured its capabilities indicating it has both ability to Display a PIN and input to confirm a matching pin, then without any further programming, WinUI spawns the popup accordingly:

[image]

If the peripheral indicates it has capabilities to only enter a PIN, then WinUI spawns that UI accordingly:

[image]

If the peripheral indicates it only has a Display capable of showing a PIN it spawns that UI:
[image]

Or if the peripheral was configured to not have any capability to either display, enter or do anything at all, then it properly handles that as well:

[image]

All that is now prevented/suppressed essentially removing any security the pairing process had intended to provide.

I speculate anyone using a peripheral that had intended to encourage user authentication, and they initiating a pair from any non complaint UI framework or windows console, or initiated a pair from a supported UI framework but on a Windows machine not supporting at least BT v4.2 with `LE Secure Connections`, then they needed this hack allowing them to circumvent security.
But its a regressive breaking change.
 Basically this pull says, i dont know how to use the security my setup demands, so lets break it for anyone else who does... 

## Steps to reproduce

1. Try pair using a complaint windows pairing framework like WinUI 3 or perhaps UWP or other modern ones supporting built in windows paring, a machine supporting BT 4.2, with the library before and after this pull

2. Observe advanced built in pairing used to work, and now it doesnt.

3.

## Expected behavior
Windows is allowed to handle the pairing process automatically with user authentication

## Actual behavior
It prevents this built in functionality introducing a breaking change.

## Configuration

**Version of the Plugin:** 
#899

**Platform:** 
Windows WinUI3

**Device:** e.g. HTC Sensation /i Phone 7 ...
with BT 4.2+

l0gaw, 2025-01-31:
I've been facing this issue trying to use the BLE Pin.  @axa88 Did you find a workaround for that?

axa88, 2025-01-31:
You'll need to provide me a bit more info on exactly what you're running into before I can recommend a work around.

Basically 'le secure connections' ie built-in pair handling, was broken with this pull.  Is that what you're reporting used to work but now doesn't? 

That said I don't think there are any work around

l0gaw, 2025-01-31:
That's exactly my case. Pin/code confirmation is not working. BTW I didn't find the #899 in the 3.1.0 release notes.

[image]

axa88, 2025-01-31:
> That's exactly my case. Pin/code confirmation is not working. BTW I didn't find the #899 in the 3.1.0 release notes.
> 
> [image]

Honestly Im no longer sure what features get into which release as I've divested myself from this library. maybe 899 didn't make it in yet.

Also I'm not at all sure this issue is reflected in the exception you posted, but anyway... Pairing, let alone the entire windows implementation, didn't work prior to 3.1 and isn't going to work after.
You can read my rant about it in discussions if interested.

l0gaw, 2025-01-31:
If I disable the ble security pin the pairing works. I'm thinking about saying to the users that BLE security is not supported by the Windows version, only by Android and iOS.

axa88, 2025-02-01:
> If I disable the ble security pin the pairing works. I'm thinking about saying to the users that BLE security is not supported by the Windows version, only by Android and iOS. 

Sure depends on your requirements. Disable or regress a feature, seemingly authentication in this case, then you will no longer have a problem with it being broken. 

What I'm finding a bit surprising is that using this library you have users on Windows at all... Its extremely fragile

janusw, 2025-02-01:
> That's exactly my case. Pin/code confirmation is not working. BTW I didn't find the [#899](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/899) in the 3.1.0 release notes.

PR #899 is not part of release 3.1.0, and there are some very simple ways to check that:
1) As you noticed correctly, it's not part of the release notes: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/releases/tag/3.1.0
2) PR #899 was merged in December 2024, but release 3.1.0 was tagged and published in May.

Since you tested with 3.1.0, @l0gaw, that also indicates that #899 did not break the "pairing with pin" feature on Windows (as claimed by @axa88), since it was apparently not working before either? Could you guys please try to figure out which of these two claims is correct? (Unfortunately I don't have any proper Windows-based device here to check this with.)

Note: If @axa88 is indeed right that #899 breaks something that was previously working (i.e. it caused a regression), then this should of course be fixed quickly. However, #906 is obviously much too big of a hammer for a regression fix. I will provide a much simpler fix in this case.

@l0gaw If you want to test a version with #889 included, you can download the nuget package from https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/12390041162#artifacts (which is built from current master). If you have downloaded it, put it into a local folder on your build machine and set up a local package source for VS to find it. The package will identify itself as `Plugin.BLE.3.1.0-14-gc9261fc`. If you have tested this version, please tell us if it behaves any different than 3.1.0 for your use case.

axa88, 2025-02-01:
Whether you call it a break or regression I don't think matters at this point. But I'll describe this in the least number of words as possible as I've obviously failed miserably in doing so already.

Pairing never fully worked in the Windows implementation.
The first implementation allowed, with a compatible device, what is called 'le secure connections' which manifests itself as built-in pairing as described in the OP.

 The latest edit broke that to allow what could be described as 'just works' pairing, which manifest itself with the least amount of security, basically none.

you can see it all working in my PR that fully support Windows pairing, or continue to question it.

l0gaw, 2025-02-01:
Hey @janusw, the issue is still happening in the version 3.1.0-14-gc9261fc. That's the exception: 

System.Exception: '[StartUpdatesNativeAsync] failed with status: ProtocolError and protocol error  Insufficient Authentication'

[image]
 
It doesn't ask to pair or to put the pin. It fails immediately. I'm able to connect to the device without the pin enabled.  This is only happening on Windows, the other platforms are working fine.

janusw, 2025-02-01:
> Hey [@janusw](https://github.com/janusw), the issue is still happening in the version 3.1.0-14-gc9261fc. That's the exception:
> 
> System.Exception: '[StartUpdatesNativeAsync] failed with status: ProtocolError and protocol error Insufficient Authentication'

Thanks for the feedback, @l0gaw. So you're saying that both 3.1.0 as well as 3.1.0-14 fail to bond with pin on Windows? But apparently with a slightly different exception? Or maybe your code or test setup has changed in between?

First of all that is good news to me, because it means that #899 did not introduce a regression for the use case that you're investigating.

Second, it's still not very good news to you, I'm sure, because stuff does not work. That's not very nice, but it's also not too surprising, since a) the Windows implementation in this library is the youngest of all four OS and b) bonding on Windows was really only implemented in the latest release 3.1.0 (via PR #820) and received a yet-unreleased bugfix in #899. Apparently not all bonding scenarios are working yet on Windows.

Note that there is PR #906 from @axa88, which goes exactly in this direction. You could try this one and give us feedback whether it works any better for you than 3.1.0, and if it fixes the bonding-with-pin workflow for your scenario. The CI-built package for this PR can be found here: https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/actions/runs/12576850525#artifacts

If you try it, please let us know how if performs for you.

> It doesn't ask to pair or to put the pin. It fails immediately. I'm able to connect to the device without the pin enabled. This is only happening on Windows, the other platforms are working fine.

What other platforms have you tested? Bonding on Android should be quite robust since version 3.0.0, see #730.

Maybe you could check whether you see any messages like this in your trace output (with 3.1.0-14)?

```
PairingKind  ... not supported
BondAsync pairing result was ... with: ...
```

Do you see those? And what exactly do they say?

They come from here:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/c9261fc147a65553092df2c5b52c4542035c5dca/Source/Plugin.BLE/Windows/Adapter.cs#L56

and here:
https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/blob/c9261fc147a65553092df2c5b52c4542035c5dca/Source/Plugin.BLE/Windows/Adapter.cs#L68

janusw, 2025-02-01:
> Bonding on Android should be quite robust since version 3.0.0, see [#730](https://github.com/dotnet-bluetooth-le/dotnet-bluetooth-le/pull/730).

Btw, @axa88, in your direction: This is a very nice example of people successfully contributing extensive and non-trivial changes (even including coding-style changes and cleanup, and indeed concerning very similar functionality as your PR #906, just on a different OS). So, believe me, it is possible.

This PR even includes contributions from three different people, who were able to build upon each other's results and also managed to react to my review comments (something that you haven't even considered so far).

l0gaw, 2025-02-01:
The PR #906 didn't fix my issue too:

ConnectToDeviceNativeAsync 5CC7C1E3D65F Named: RM510C 501001321 Connected: Disconnected
Device_ConnectionStatusChanged 5CC7C1E3D65F RM510C 501001321 Connected
GattSession_SessionStatusChanged: Active
[20250201-135858.953] - Connected: RM510C 501001321
GattSession_MaxPduSizeChanged: 247
ConnectToDeviceAsync Connected: 00000000-0000-0000-0000-5cc7c1e3d65f RM510C 501001321
[13:59:07 - App - VRB] BLE:INFO - [BleComms] - Connected to 'RM510C 501001321'.
[13:59:07 - App - VRB] BLE:VERBOSE - [BleComms] - Negotiated MTU of 247
[13:59:07 - App - VRB] BLE:VERBOSE - [BleComms] - Querying service...
[GetServicesNativeAsync] success.
[13:59:13 - App - VRB] BLE:VERBOSE - [BleComms] - Connecting to characteristics...
[GetCharacteristicsNativeAsync] success.
[13:59:13 - App - VRB] BLE:VERBOSE - [BleComms] - Start Updates on TxCharacteristics...
Characteristic.StartUpdates
Exception thrown: 'System.Exception' in System.Private.CoreLib.dll
[StartUpdatesNativeAsync] failed with status: ProtocolError and protocol error  Insufficient Authentication

The thread 53584 has exited with code 0 (0x0).

It's happening the same thing. I also have a version of this App for Android and iOS and they are working fine. The pin input pops up and after the user fills in the pin, the connection succeeds.

l0gaw, 2025-02-01:
Perhaps I'm not connecting to the device correctly. I'm using Adapter.ConnectToDeviceAsync(_device, cancellationToken: ct) to connect and you mentioned a BondAsync method. Would you tell me the right way to connect to the device?

l0gaw, 2025-02-01:
I tried to call the BondAsync method and I saw this in the logs:
BondAsync pairing result was OperationAlreadyInProgress with: RM510C 501001321: 00000000-0000-0000-0000-5cc7c1e3d65f

janusw, 2025-02-01:
> Perhaps I'm not connecting to the device correctly. I'm using Adapter.ConnectToDeviceAsync(_device, cancellationToken: ct) to connect and you mentioned a BondAsync method. Would you tell me the right way to connect to the device?

Well, in order to use the pin-based bonding process, you'll need to call `BondAsync` at least once, I think. After the two devices have been bonded successfully, it should be enough to call `ConnectToDeviceAsync` for connecting to the device.

If you did not use `BondAsync` before, could you please try that with 3.1.0, 3.1.0-14 and the packet from #899? Does any of the three work? (If yes, you may have to "un-bond" the device in the system, in order to test the other versions.)

l0gaw, 2025-02-01:
I did what you said but it keeps failing in both versions. I can find this message in the logs:

BondAsync pairing result was OperationAlreadyInProgress with: RM510C 501001321: 00000000-0000-0000-0000-5cc7c1e3d65f
 
and the exception is still the same:
System.Exception: '[StartUpdatesNativeAsync] failed with status: ProtocolError and protocol error  Insufficient Authentication'

The pin dialog doesn't appear like on the other platforms.

janusw, 2025-02-01:
> I did what you said but it keeps failing in both versions.

I'm not surprised it doesn't work with 3.1.0 or 3.1.0-14, but #906 should fix exactly this case IIUC. Does this also fail?
(Sorry, I meant #906, not #899 in my last comment.)

> I can find this message in the logs:
> 
> BondAsync pairing result was OperationAlreadyInProgress with: RM510C 501001321: 00000000-0000-0000-0000-5cc7c1e3d65f

That's weird. I don't understand why you would get such an error. Sounds like you might be triggering the pairing process twice or so ...?

axa88, 2025-02-02:
I shouldn't looked closer. It appears commenting is running into their issue when authentication is triggered explicitly by accessing protected data from the gatt.  Not from a call to Bond 

Certainly not directly related to the OP, but exposes a related insufficiency in the current implementation.

Eventually commentor might want to open a new issue, or the title of this one might be changed to something like 'windows pairing and authentication isn't fully supported'

axa88, 2025-02-02:
> Btw, @axa88, in your direction: This is a very nice example of people successfully contributing extensive and non-trivial changes (even including coding-style changes and cleanup, and indeed [...]

Yes certainly possible to do additional work in order to find a compromise on functionality and form.  I suppose my point was I wasn't satisfied in compromising on either.

axa88, 2025-02-02:
> The PR #906 didn't fix my issue too:

Ah, so you tried this PR? 

_How_ did you implement the Bond? 

Did you _explicitly_ call the overloaded BondAsync function? 
If not then you are using the old broken pairing method. I didn't remove access to the old method as my original intention was a PR review allowing users to compare both before submitting it officially and removing the old method. 

Dare I estimate you didn't access the new method?

janusw, 2025-02-04:
> Whether you call it a break or regression I don't think matters at this point. 

Well, let's not split hairs and let's not talk about language here. Something that was working before, but is then broken, is a regression.

The important point is to distinguish this from something that never worked. That is also not nice, but much less severe than a regression.

> Pairing never fully worked in the Windows implementation. The first implementation allowed, with a compatible device, what is called 'le secure connections' which manifests itself as built-in [...]

AFAIU, the "last edit" aka #899 was supposed to fix exactly the case of 'just works' pairing, no?

So, my simple question is: Is there a use case that #899 breaks, which was working before? If yes, which case is that exactly? And I'm asking here about a case that you have **tested**, not something that you _believe_ may potentially be broken (by reading the code).

axa88, 2025-02-04:
> AFAIU, the \"last edit\" aka #899 was supposed to fix exactly the case of 'just works' pairing, no?

Yes.  See my last comment explaining to commenter why it was most likely not implemented correctly.

axa88, 2025-02-04:
> ## Steps to reproduce

1. Try pair using a complaint windows pairing framework like WinUI 3 or perhaps UWP or other modern ones supporting built in windows paring, a machine supporting BT 4.2, with the library before and after this pull

2. Observe advanced built in pairing used to work, and now it doesn't.

Unclear?

axa88, 2025-02-04:
Fwiw anyone wanting to implement or understand pairing themselves, here is a BLE Peripheral i whipped up to support both, in layman's terms, automatic and manual/custom pairing.  I even used arduino so it can easily be modified.

You can change settings in run time to support all but OOB authentication methods, as well as encryption and key strategies. You will likely need something like this to truly understand how pairing truly works or have a vast array of devices that each support all the different authentication methods.

https://github.com/axa88/Esp32-SSP