dotnet-bluetooth-le/dotnet-bluetooth-le #916: Service  characteristic discovery not enacted with a 4 character UUID

Nick-Blair19, 2025-01-09:
To help us fix your issue, please provide the information in the below template. If something causes a crash, provide as much information as you can gather.
Just imagine: we do not know what you are doing! 

> Note: There is often little we can do without a minimal reproducible sample of the issue, so please provide that in a **standalone git repository** and link it here.
https://github.com/[Nick-Blair19/TME](https://github.com/Nick-Blair19/TME)

## Steps to reproduce

1. I  have code:
public async  void  Connect(IDevice _gattDevice)
{
 await   this._adapter.ConnectToDeviceAsync(_gattDevice);
  
    try
    {
        var service = await _gattDevice.GetServiceAsync(Guid.Parse("fefb"));
        if (service != null)
        ...... (other code)
       ........
           }
    catch
    {
       
    }
}
2. I build and launch and run on Ipad from a mac with latest preview vs studio for windows community edition 

    
## Expected behavior
The service characteristic should be discovered

## Actual behavior
The debug jumps over the line of code and goes straight to the 'Catch' exception though it does connect to the module but no communication

### Crashlog

Doesnt crash

## Configuration

PLugin 3.1.0

iOS 17.7.1

Apple Ipad 9th edition

janusw, 2025-01-11:
> ## Actual behavior
> 
> The debug jumps over the line of code and goes straight to the 'Catch' exception though it does connect to the module but no communication

What kind of exception do you get?

Nick-Blair19, 2025-01-11:
Hi Janus,
Noticed a while back when I had the 4 character service characteristic accessed in GattIdentifiers.cs file with a sample Ble.plugin project on the web it threw a null exception. With my current project I define the characteristic directly with Guid.Parse("fefb") instead of the GattIdentifiers.cs file and there is no exception as in a 'try' loop. 
Thanks and regards.
Nick.

Nick-Blair19, 2025-01-11:
Sorry meant accessed the characteristic UUID

Nick-Blair19, 2025-01-11:
I have a Telit ble module with the Telit TIO proprietary service by the way

Nick-Blair19, 2025-01-12:
its ok Janus I now have
var service = await _gattDevice.GetServiceAsync(GattIdentifiers.Service);
and 
 public class GattIdentifiers
 {
     public static Guid Service = Guid.Parse("fefb");

........
}
And no exception.
But do you know if ble.plugin can handle 4 character Characteristic UUIDs as still goes to 'Catch'?
Just want to check with you.
Regards Nick.
[bleplugin tme-5.zip](https://github.com/user-attachments/files/18390332/bleplugin.tme-5.zip)
Ps. Here is the project before the change just now.

Nick-Blair19, 2025-01-12:
I mean it worked in corebluetooth

Nick-Blair19, 2025-01-12:
[image]
Here the exception without the try loop Janus

Nick-Blair19, 2025-01-12:
Tried
var service = await _gattDevice.GetServiceAsync(Guid.Parse("00000000-0000-0000-0000-000000000fefb"));
But exception thrown was
[image]

Nick-Blair19, 2025-01-12:
Sorry Janus had extra zero in last chunk. Now have 
var service = await _gattDevice.GetServiceAsync(Guid.Parse("00000000-0000-0000-0000-00000000fefb"));
if (service != null)

But the "fefb" service is null.
As I said "fefb" was discovered with old xamarin corebluetooth code?
Any suggestions?
Nick.

Nick-Blair19, 2025-01-12:
[image]
Here are UUIDs Janus

janusw, 2025-01-17:
> Now have var service = await _gattDevice.GetServiceAsync(Guid.Parse("00000000-0000-0000-0000-00000000fefb")); if (service != null) { ...
> 
> }
> 
> But the "fefb" service is null.

Which services does `_gattDevice.GetServicesAsync()` return?

> The service characteristic UUID "fefb" was discovered with old xamarin corebluetooth code?

What exactly do you mean by "old xamarin corebluetooth code"? Are you saying you got the expected service with an old version of Plugin.BLE? If yes, which one?

Nick-Blair19, 2025-01-18:
Hi Janus, tried to discover all the services with connected module using this code implementation-

[image]

But didnt work with this exception

[image]

Re the (Xamarin) Corebluetooth it is an Apple Bluetooth Api.

This code with Corebluetooth worked before with a Xamarin project but the Api was outdated with Net 8.0 Maui

 internal void Connect(CBPeripheral peripheral)
 {
   
     peripheral.DiscoveredService += (sender, e) =>
     {
         if (peripheral.Services == null) { return; }
         foreach (var s in peripheral.Services)
         {
             if (s.UUID.ToString().Contains("fefb"))
           
                 {
                 service = s;
                 TIOservice = s;
                 
                 peripheral.DiscoverCharacteristics(service);
             };
         }
     };
Obviously you know nothing about this but I'm just saying that the Api accepted a 4 character service characteristic UUID.
I had a similar problem with this with I think Ble.plugin 2.0 though

janusw, 2025-01-19:
Well, I guess you need to check where your exceptions come from. You have shown two different ones now, but none of them tells me much. You should have a look at the call stack to check where they come from and do some debugging to find out why they occur.

Nick-Blair19, 2025-01-20:
Actually Janus 'fefb" is not a valid Guid. Here code with only 7 characters in the first block of the UUID.

[image]

And here the system format exeption

[image]

Nick-Blair19, 2025-01-20:
See post above

Nick-Blair19, 2025-01-22:
HI Janus 
I have code
 var allServices = await _gattDevice.GetServicesAsync();
 {
    
     var stringservices = allServices.Count;
     var options = new JsonSerializerOptions
     {
         WriteIndented = true
     };

     var jsonString = JsonSerializer.Serialize(stringservices, options);
     Console.WriteLine(jsonString);
     return;
 }
And the count is **2** discovered services. I tried to show services by puttting  var stringservices = allServices.ToString();
but there was nothing in the console window. I have no other viable fields to try.
See screenshot

[image]

I suspect the services are 0xFEFB and 0x180A (see nRF Connect app screenshot)
Anyway the fefb is an invalid guid
See screenshot

[image]

janusw, 2025-01-22:
> HI Janus I have code var allServices = await _gattDevice.GetServicesAsync(); {
>
> ```
>  var stringservices = allServices.Count;
>  var options = new JsonSerializerOptions
>  {
>      WriteIndented = true [...]

Try something like this:

```
var allServices = await _gattDevice.GetServicesAsync(); 
foreach (var service in allServices)
{
    Console.WriteLine(service.Id);
}
```
That should show you the GUIDs of all services found on the device.

Nick-Blair19, 2025-01-23:
By the way janus i know this should really be a separate issue but just a quick question 

[image]
Can you help with this warning? Googled warning and couldn't find anything on the web?
Regards nick

janusw, 2025-01-23:
> This is really interesting Janus. Got service Ids 0000fefb-0000-1000-8000-00805f9b34fb And 0000180a-0000-1000-8000-00805f9b34fb

That makes a lot of sense. These are exactly the services 0xFEFB and 0x180A that you are looking, but instead of the partial 4-char (32bit) UUID, they have the corresponding "full" 256bit Bluetooth UUID. You can easily convert between both forms via the helper methods`GuidExtension.UuidFromPartial` and `GuidExtension.PartialFromUuid`.

janusw, 2025-01-23:
Btw, this is more of a question than a bugreport, so I'll convert it from an issue to a discussion. Please use the discussion section for such questions in the future.