dotnet-bluetooth-le/dotnet-bluetooth-le #918: .net maui connection with esp32

cosmaflorin, 2025-01-20:
Hello.
I'm new in .net maui programming.
I try a simple .net maui app to connect to an esp32 device.
Although there are no errors, the connection cannot be established.
CancellationToken cancel = new CancellationToken();
await adapter.ConnectToKnownDeviceAsync(Guid.Parse(deviceId), new ConnectParameters(true), cancel);  
always returns adapter=null.
Please help me.

axa88, 2025-01-23:
> Hello.
> I'm new in .net maui programming.
> I try a simple .net maui app to connect to an esp32 device.
> Although there are no errors, the connection cannot be established.
> CancellationToken cancel = [...]

Check out the Maui demo in this repo for proper useage.

CinzV, 2025-09-19:
> Hello. I'm new in .net maui programming. I try a simple .net maui app to connect to an esp32 device. Although there are no errors, the connection cannot be established. CancellationToken cancel = [...]

Hello! For work, I had to interface with an esp32-c3-mini module with AT firmware loaded, and I used this code (you will need to adapt it a little to your case; I am only showing you the functions I use).

```
using Microsoft.Extensions.Configuration;
using Plugin.BLE;
using Plugin.BLE.Abstractions;
using Plugin.BLE.Abstractions.Contracts;
using Plugin.BLE.Abstractions.EventArgs;
using Plugin.BLE.Abstractions.Extensions;
using System.Text;

namespace MyService.BLEComunication
{
    public class BLEService
    {
        #region PROPERTIES

        /// <summary>
        /// IAdapter instance
        /// </summary>
        public static IAdapter? Adapter;

        /// <summary>
        /// IBluetoothLE instance
        /// </summary>
        public static IBluetoothLE? Ble;

        /// <summary>
        /// Variable that continents the bluetooth device to which the phone is connected
        /// </summary>
        public static IDevice? ConnectedDevice;

        /// <summary>
        /// Variable containing the list of scanned BLE devices
        /// </summary>
        public List<IDevice> Devices;

        /// <summary>
        /// Variable containing the byte stream being sent by the BLE
        /// </summary>
        private static List<byte[]>? dataBuffer;

        /// <summary>
        /// Variable containing the event of message reception by the BLE
        /// </summary>
        private static EventHandler<CharacteristicUpdatedEventArgs>? bleValueUpdatedHandler;

        /// <summary>
        /// Variable containing the characteristic used for receiving messages from the BLE
        /// </summary>
        private static ICharacteristic? activeCharacteristic;

        #endregion

        public BLEService()
        {
            Ble = CrossBluetoothLE.Current;
            Adapter = CrossBluetoothLE.Current.Adapter;
            Adapter.ScanMode = ScanMode.LowLatency;
            Devices = new List<IDevice>();
            Ble.StateChanged += (s, e) => { };
            dataBuffer = new List<byte[]>();
        }

        /// <summary>
        /// Function allowing connection to bluetooth device
        /// </summary>
        /// <param name="device"></param>
[... 206 lines cut ...]
                            characteristic.ValueUpdated -= bleValueUpdatedHandler;

                        bleValueUpdatedHandler = (sender, args) =>
                        {
                            //Here, enter the code to be executed when the form responds to you.
                        };

                        characteristic.ValueUpdated += bleValueUpdatedHandler;

                        activeCharacteristic = characteristic;

                        //Start receiving updates
                        await characteristic.StartUpdatesAsync();
                    }
                }
            }

            return string.Empty;
        }
    }
}

```
I hope this helps, good luck!!
Bye bye