dotnet-bluetooth-le/dotnet-bluetooth-le #924: DeviceConnectionLost works on Windows but not on Android

TimoKinnunen, 2025-02-09:
// works on Windows but not on Android
            bleAdapter.DeviceConnectionLost += (sender, e) => Debug.WriteLine("🟢 DeviceConnectionLost event fired!");

axa88, 2025-02-10:
> ```
>         // works on Windows but not on Android

works on Android, and on windows as long as a solicited disconnection request to the same device hadn't failed prior.

>         bleAdapter.DeviceConnectionLost += (sender, e) => Debug.WriteLine("🟢 DeviceConnectionLost event fired!");
> ```

Are you sure on Android Debug.WriteLine does what you think it does...

TimoKinnunen, 2025-02-10:
Debug.WriteLine is not a problem. It works well. I have more code to perform when event triggers.

axa88, 2025-02-10:
> I have more code to perform when event triggers.

Then maybe the more performing is throwing and silently aborting the callback thread... Without fooling with adequate information it's going to be difficult to help as Android is known to work well in this area.

TimoKinnunen, 2025-02-10:
This happens on Android. [BluetoothGatt] onClientConnectionState() - status=19 clientIf=6 device=XX:XX:XX:XX:18:3A
[0:] 🔴 DeviceDisconnected event fired! Does not trigger DeviceConnectionLost on Android. Plugin.BLE must have a bug in Android implementation. I use Microsoft Visual Studio and build MAUI Hybeid app.

Senso4sLab, 2025-10-03:
I came across a similar problem.  If BLE is turned off on the phone, I would expect that the ConnectionLost event will be triggered. This is not the case on some Android phones (Samsung phones).

janusw, 2025-10-04:
> I came across a similar problem. If BLE is turned off on the phone, I would expect that the ConnectionLost event will be triggered. This is not the case on some Android phones (Samsung phones).

This case may actually be fixed by the recent PR #958.