dotnet-bluetooth-le/dotnet-bluetooth-le #931: Issue: Device Not Found After Disconnection Until App Restart

AbdullahGaw, 2025-02-27:
I have been using this library mainly to scan and discover nearby devices, and it worked great. Recently, I have been connecting to my IoT devices, disconnecting, exchanging data, and all that works great. However, I have been struggling to understand why, once I connect to device then I disconnect from a device, then run a scan, it does not find it, i waited up to 10 minutes, and the device still does not show. However, if I terminate the app then launch it, run a scan, now I can see the device. I am wondering if there is a configuration to fix this ? 

**Description**
I have been using this library primarily for scanning and discovering nearby devices, and it has worked great. Recently, I started implementing connections, disconnections, and data exchanges with my IoT devices, and everything functions as expected.

However, I encountered an issue where, after connecting to a device and then disconnecting, the device no longer appears in subsequent scans. Even after waiting for up to 10 minutes, the device does not show up in the scan results. However, if I terminate the app and relaunch it, the device becomes visible again when I run a scan.

**Expected Behavior**
After disconnecting from a device, running a new scan should detect the device again without requiring an app restart.

**Actual Behavior**
After disconnecting from the device, running a scan does not find it.
The device remains undetectable even after waiting for several minutes.
Restarting the app resolves the issue, and the device appears in the scan.

**Questions**
Is there a configuration setting or a known issue that could be causing this behavior?
Is there a way to refresh the scanning state after disconnection to detect the device again?

**Additional Information**
Library Version: 3.1.0
Platform: NET MAUI (Android, iOS)
Device Model: Google Pixel 8a (Issue is not device-specific)

**Reproduction Steps:**
Scan for nearby devices.
Connect to a discovered device.
Disconnect from the device.
Run a scan again.
Observe that the device is no longer found.
Restart the app and scan again—now the device appears.

**Code setup**:

private readonly IAdapter _bluetoothAdapter;

public BLEManager()
{
    _bluetoothAdapter = CrossBluetoothLE.Current.Adapter;
}

public async Task<IReadOnlyCollection<IDevice>> ScanAsync()
{
    await _sem.WaitAsync();
    try
    {
        await _bluetoothAdapter.StartScanningForDevicesAsync(new ScanFilterOptions
        {
            ServiceUuids = [MY_SERVICE]
        });
        var discoveredDevices = new List<IDevice>(_bluetoothAdapter.DiscoveredDevices);
        return discoveredDevices;
    }
    finally
    {
        _sem.Release();
    }
}

public async Task DisconnectAsync()
{
    await _sem.WaitAsync();
    try
    {
        await _bluetoothAdapter.DisconnectDeviceAsync(_device);
    }
    finally
    {
        _sem.Release();
    }
}

public async Task ConnectAsync(IDevice device)
{
    await _sem.WaitAsync();
    try
    {
            _device = await _bluetoothAdapter.ConnectToKnownDeviceAsync(device.Id);
    }
    finally
    {
        _sem.Release();
    }
}

Any insights or recommendations would be greatly appreciated. Thanks in advance!

Lmariotti95, 2025-02-27:
Hi @AbdullahGaw,

I'm using that library with multiple connections/disconnections with multiple devices and beside #926 I've never faced any truble.
- What about `adapter.ScanTimeout` ? Have you tried to use different configuration?
- Have you tried to run stan without filtering for `ServiceUuids`?

AbdullahGaw, 2025-02-28:
Hi @Lmariotti95 ,

I have a timeout set to 5 seconds (_bluetoothAdapter.ScanTimeout = 5000;). I tried scanning without a ScanFilter, but the issue persists.

I also have a BLE scanner app installed from the Play Store. Whenever my app fails to detect the device after a disconnection, I open the BLE scanner app, start a scan, wait for it to complete and display my device, then switch back to my app and re-scan. Only then does my app detect the device again.

Lmariotti95, 2025-02-28:
Hi @AbdullahGaw 

I mean `adapter.ScanMode` 😅 don't know why I wrote ScanTimeout 

https://developer.android.com/reference/android/bluetooth/le/ScanSettings.html